ros2/examples #23: Can't compile

facontidavide, 2015-05-20:
I followed the instructions provided but I get the following error:

/home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp: In function ‘int main(int, char**)’:
/home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:23:45: error: ‘AsyncParametersClient’ is not a member of ‘rclcpp::parameter’
   auto parameters_client = std::make_shared<rclcpp::parameter::AsyncParametersClient>(node);
                                             ^
/home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:23:45: note: suggested alternative:
In file included from /home/shadylady/ros2_ws/install/include/rclcpp/rclcpp.hpp:23:0,
                 from /home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:15:
/home/shadylady/ros2_ws/install/include/rclcpp/parameter_client.hpp:43:7: note:   ‘rclcpp::parameter_client::AsyncParametersClient’
 class AsyncParametersClient
       ^
/home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:23:45: error: ‘AsyncParametersClient’ is not a member of ‘rclcpp::parameter’
   auto parameters_client = std::make_shared<rclcpp::parameter::AsyncParametersClient>(node);
                                             ^
/home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:23:45: note: suggested alternative:
In file included from /home/shadylady/ros2_ws/install/include/rclcpp/rclcpp.hpp:23:0,
                 from /home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:15:
/home/shadylady/ros2_ws/install/include/rclcpp/parameter_client.hpp:43:7: note:   ‘rclcpp::parameter_client::AsyncParametersClient’
 class AsyncParametersClient
       ^
/home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:23:91: error: no matching function for call to ‘make_shared(std::shared_ptr<rclcpp::node::Node>&)’
   auto parameters_client = std::make_shared<rclcpp::parameter::AsyncParametersClient>(node);
                                                                                           ^
/home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:23:91: note: candidate is:
In file included from /usr/include/c++/4.8/memory:82:0,
                 from /home/shadylady/ros2_ws/install/include/rclcpp/rclcpp.hpp:19,
                 from /home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:15:
/usr/include/c++/4.8/bits/shared_ptr.h:610:5: note: template<class _Tp, class ... _Args> std::shared_ptr<_Tp1> std::make_shared(_Args&& ...)
     make_shared(_Args&&... __args)
     ^
/usr/include/c++/4.8/bits/shared_ptr.h:610:5: note:   template argument deduction/substitution failed:
/home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:23:91: error: template argument 1 is invalid
   auto parameters_client = std::make_shared<rclcpp::parameter::AsyncParametersClient>(node);
                                                                                           ^
/home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:47:17: error: unable to deduce ‘auto&&’ from ‘values_f2’
   for (auto p : values_f2) {
                 ^
/home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:53:72: error: expected primary-expression before ‘)’ token
         (p.get_value<rclcpp::parameter::ParameterType::PARAMETER_BOOL>() ? "true" : "false");
                                                                        ^
/home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:57:74: error: expected primary-expression before ‘)’ token
         p.get_value<rclcpp::parameter::ParameterType::PARAMETER_INTEGER>();
                                                                          ^
/home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:61:73: error: expected primary-expression before ‘)’ token
         p.get_value<rclcpp::parameter::ParameterType::PARAMETER_DOUBLE>();
                                                                         ^
/home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:65:73: error: expected primary-expression before ‘)’ token
         p.get_value<rclcpp::parameter::ParameterType::PARAMETER_STRING>();
                                                                         ^
In file included from /home/shadylady/ros2_ws/install/include/rclcpp/node.hpp:276:0,
                 from /home/shadylady/ros2_ws/install/include/rclcpp/rclcpp.hpp:21,
                 from /home/shadylady/ros2_ws/src/ros2/examples/userland/src/prototype_intraprocesses.cpp:19:
/home/shadylady/ros2_ws/install/include/rclcpp/node_impl.hpp: In lambda function:
/home/shadylady/ros2_ws/install/include/rclcpp/node_impl.hpp:318:64: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         return std::count(substr.begin(), substr.end(), '.') < depth;
                                                                ^
In file included from /home/shadylady/ros2_ws/install/include/rclcpp/node.hpp:276:0,
                 from /home/shadylady/ros2_ws/install/include/rclcpp/rclcpp.hpp:21,
                 from /home/shadylady/ros2_ws/src/ros2/examples/userland/src/prototype_intraprocesses.cpp:19:
/home/shadylady/ros2_ws/install/include/rclcpp/node_impl.hpp: In lambda function:
/home/shadylady/ros2_ws/install/include/rclcpp/node_impl.hpp:318:64: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         return std::count(substr.begin(), substr.end(), '.') < depth;
                                                                ^
In file included from /home/shadylady/ros2_ws/install/include/rclcpp/node.hpp:276:0,
                 from /home/shadylady/ros2_ws/install/include/rclcpp/rclcpp.hpp:21,
                 from /home/shadylady/ros2_ws/src/ros2/examples/userland/src/imu/rosimu/publisher.cpp:22:
/home/shadylady/ros2_ws/install/include/rclcpp/node_impl.hpp: In lambda function:
/home/shadylady/ros2_ws/install/include/rclcpp/node_impl.hpp:318:64: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         return std::count(substr.begin(), substr.end(), '.') < depth;
[... 26 lines cut ...]
                 from /home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:15:
/usr/include/c++/4.8/bits/shared_ptr.h:610:5: note: template<class _Tp, class ... _Args> std::shared_ptr<_Tp1> std::make_shared(_Args&& ...)
     make_shared(_Args&&... __args)
     ^
/usr/include/c++/4.8/bits/shared_ptr.h:610:5: note:   template argument deduction/substitution failed:
/home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:23:91: error: template argument 1 is invalid
   auto parameters_client = std::make_shared<rclcpp::parameter::AsyncParametersClient>(node);
                                                                                           ^
/home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:47:17: error: unable to deduce ‘auto&&’ from ‘values_f2’
   for (auto p : values_f2) {
                 ^
/home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:53:72: error: expected primary-expression before ‘)’ token
         (p.get_value<rclcpp::parameter::ParameterType::PARAMETER_BOOL>() ? "true" : "false");
                                                                        ^
/home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:57:74: error: expected primary-expression before ‘)’ token
         p.get_value<rclcpp::parameter::ParameterType::PARAMETER_INTEGER>();
                                                                          ^
/home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:61:73: error: expected primary-expression before ‘)’ token
         p.get_value<rclcpp::parameter::ParameterType::PARAMETER_DOUBLE>();
                                                                         ^
/home/shadylady/ros2_ws/src/ros2/examples/userland/src/async_parameters.cpp:65:73: error: expected primary-expression before ‘)’ token
         p.get_value<rclcpp::parameter::ParameterType::PARAMETER_STRING>();
                                                                         ^
make[2]: *** [CMakeFiles/async_parameters__rmw_opensplice_cpp.dir/src/async_parameters.cpp.o] Error 1
make[1]: *** [CMakeFiles/async_parameters__rmw_opensplice_cpp.dir/all] Error 2

dirk-thomas, 2015-05-20:
This seems to be related to a recently merged PR (ros2/rclcpp#30). For a temporary workaround you can revert the `rclcpp` repository to the hash ros2/rclcpp@198b40a0bf74fe5b6fe76766ccc1666dfe7f4abd until it will be fixed on the master.

facontidavide, 2015-05-20:
I am not in a hurry, I will wait for you to fix it with the latest version.  ;)

facontidavide, 2015-05-20:
just pulled the latest changes and it compiles just fine.

closed 2015-05-20 by facontidavide