ros2/examples #25: Example error

shreyasr89, 2015-05-22:
I am getting an error when i run the sync_parameters, async_parameters and prototype_intraprocess which are in the examples.

For sync and async errors are:
hiya@hiya-ThinkPad-T530:~/ros2_ws$ source install/setup.bashhiya@hiya-ThinkPad-T530:~/ros2_ws$ ./install/bin/sync_parameters
  rcl_interfaces::type_support::register_type__GetParameters()
  rcl_interfaces::type_support::register_type__GetParameterTypes()
  rcl_interfaces::type_support::register_type__SetParameters()
  rcl_interfaces::type_support::register_type__ListParameters()
  rcl_interfaces::type_support::register_type__DescribeParameters()
terminate called after throwing an instance of 'std::future_error'
  what():  No associated state
Aborted (core dumped)

For intraprocess errors are:

hiya@hiya-ThinkPad-T530:~/ros2_ws$ source install/setup.bash
hiya@hiya-ThinkPad-T530:~/ros2_ws$ ./install/bin/prototype_intraprocess
------ Creating subscriber:
------ Creating publisher:
------ monitor string:Message sent through ROS
------ monitor string:Message sent through ROS
  simple_msgs::type_support::register_type__Intraprocess()
  simple_msgs::type_support::register_type__Intraprocess()
------ Launching subscriber:
------ monitor string:Message sent through ROS
------ monitor string:Message sent through ROS
Segmentation fault (core dumped)
Running it in gdb I got:
“Program received signal SIGSEGV error
May I know the reasons for these errors.

wjwwood, 2015-05-22:
Thanks for the report. Those examples are a little stale; it is on my todo list to clean the examples up to remove some of the examples which are really tests like `publisher` and `subscriber` and update any out of date examples and remove some which are actually just prototypes.

The sync/async parameter examples should be working, maybe @esteve can have a look at the error you're getting. I'm not surprised, however, that the `prototype_intraprocess` example isn't working anymore. That was an early experiment that we haven't touched in a while.

esteve, 2015-05-22:
@shreyasr89 thanks for trying out ROS 2.0, it's still in early stages and many things may break, though.

The Parameters API is not complete yet, the `sync_parameters` and `async_parameters` examples are there just to have a grasp of what the client-side Parameters API will look like. We're still working on the service side of the API, which will bring support for parameters in ROS 2.0

closed 2015-06-30 by wjwwood