ros2/examples #34: Error compiling ROS2 under windows.

bslava, 2015-06-24:
Hey,

I'm absolutely new to ROS/ROS2 and I'm trying to compile ROS2 under windows. I did all the necessary installations described in "ROS 2.0 setup on Windows". However when I run:
   python src\ament\ament_tools\scripts\ament.py build
I receive the following error message:

[image]

Please help!

wjwwood, 2015-06-24:
This happens when one of your end variables that we are extending contains quotes somewhere in the path. Can you post the output of `set`?

Also please copy and paste the error, nobody can search text in an image :smile:

bslava, 2015-06-24:
Thank you very much for help. PATH variable definitely contained quotes. I appreciate you help very much.

closed 2015-06-24 by bslava

tfoote, 2015-06-24:
@bslava If you're new to ROS entirely, I'd recommend getting started with http://wiki.ros.org/ROS/Tutorials 

ROS2 is in a very active early development stage. Be aware that there will be large changes happening throughout and that it's not entirely implemented yet either. With that said if you're still interested in trying it out and contributing that would be great.

wjwwood, 2015-06-24:
Np, I've found this one line batch command addresses the problem:

```
set "PATH=%PATH:"=%"
```

bslava, 2015-06-25:
**wjwwood**, **tfoote**,

Thank you very much for your help.

I would like to explain the reasons why I decided to start from ROS2 and not from the ROS:
* Most of the OS drivers that I expect to use exist for Windows only. There will be the Linux versions in the future, but I believe the Windows ones might be more up-to-day.
* At least the least 2 ROS version do not have Windows binding. Is it correct?
* I found a great explanation how to setup ROS2 on Windows.
* I read somewhere in the blogs that ROS2 is pretty mature and the next ROS version will be based on ROS2 codebase.
* I decided if something went wrong with ROS2/Windows I’d switch to ROS/Ubuntu to gather some experience and then would return to ROS2/Win. 

I’d be very grateful for your thoughts & comments.

Regards,
Slava

wjwwood, 2015-06-26:
@bslava I would not suggest that you use ROS 2 yet. You can look at using http://wiki.ros.org/win_ros or http://wiki.ros.org/rosjava for what you need to do on Windows.

You could also potentially use http://www.mathworks.com/hardware-support/robot-operating-system.html if your sensors have drivers in Matlab.

> At least the least 2 ROS version do not have Windows binding. Is it correct?

I don't understand your question. ROS 1 has win_ros and rosjava and ROS for Matlab, but no official support for Windows. ROS 2 will have official support for Windows, but it's not ready right now.

> I read somewhere in the blogs that ROS2 is pretty mature and the next ROS version will be based on ROS2 codebase.

ROS 2 is not mature yet, and ROS 1 will continue to be the platform of choice for the near term.

bslava, 2015-06-28:
**@wjwwood** Thanks a lot!