ros2/examples #36: build src/ament/uncrustify fails

spiked3, 2015-06-29:
Uncrustify_Win32_2011.vcxproj: The application which this project type is based on was not found.

Looks like project was not migrated?

wjwwood, 2015-06-29:
@spiked3 Our build script tries to upgrade the project in place, see:

https://github.com/ament/uncrustify/blob/master/CMakeLists.txt#L74-L82

It's not clear to me why you get that message. We've talked about upgrading it and committing the upgrade, but we haven't yet. You can try to open it in VS manually and have it upgrade it, then try again.

Unfortunately, I think you'll run into other issues a little further down the line, it's currently failing on our CI:

http://ci.ros2.org/job/ros2_batch_ci_windows/5/

I haven't gotten around to fixing it this time.

spiked3, 2015-06-29:
Yeah, it almost look as though 'doesnt need upgrading' is the error that is stopping the build.

I will see if I can figure it out. I (as a windows person) was looking forward to ROS2. Thanks.

dirk-thomas, 2015-07-15:
Our build farm is able to get beyond that point with the current code base: http://ci.ros2.org/job/ros2_batch_ci_windows/12/

@spiked3 Are you still having this problem or can this be closed?

spiked3, 2015-07-15:
according to the current status, it is still failing.  But since I was told no one was working on it, I have not really tried since then.  Close it.

dirk-thomas, 2015-07-15:
The current job fails after `uncrustify` was build successfully at a different package. It would be great if you could confirm that the problem is also solved for you.

spiked3, 2015-07-15:
I no longer have a lot of the software installed.  I will put this on my ToDo list, but it may be a while.

dirk-thomas, 2015-07-15:
I will close the ticket for now since it seems to work for us now. But once you had a chance to try it again please comment again to either confirm that its fixed or to reopen the ticket. Thanks.

spiked3, 2015-07-17:
still fails, same place and message;

       CustomBuild:
         Building Custom Rule C:/dev/ros2/src/ament/uncrustify/CMakeLists.txt
         CMake does not need to re-run because C:\dev\ros2\build\uncrustify\CMa
         keFiles\generate.stamp is up-to-date.
         Generating C:/dev/ros2/src/ament/uncrustify/src/uncrustify_version.h
         Could Not Find C:\dev\ros2\src\ament\uncrustify\src\uncrustify_version
         .h
         Generating C:/dev/ros2/src/ament/uncrustify/win32/Debug/Uncrustify.exe

         Some errors occurred during migration. For more information, see the m
         igration report:
         C:\dev\ros2\src\ament\uncrustify\win32\UpgradeLog.htm

         Information:
         This project/solution does not require migration. It will open without
          modification.
     5>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppComm
       on.targets(170,5): error MSB6006: "cmd.exe" exited with code 1. [C:\dev\
       ros2\build\uncrustify\external_uncrustify.vcxproj]
     5>Done Building Project "C:\dev\ros2\build\uncrustify\external_uncrustify.
       vcxproj" (default targets) -- FAILED.
     3>Done Building Project "C:\dev\ros2\build\uncrustify\external_uncrustify.
       vcxproj.metaproj" (default targets) -- FAILED.
     2>Done Building Project "C:\dev\ros2\build\uncrustify\ALL_BUILD.vcxproj.me
       taproj" (default targets) -- FAILED.
     1>Done Building Project "C:\dev\ros2\build\uncrustify\uncrustify.sln" (def
       ault targets) -- FAILED.

Build FAILED.

       "C:\dev\ros2\build\uncrustify\uncrustify.sln" (default target) (1) ->
       "C:\dev\ros2\build\uncrustify\external_uncrustify.vcxproj.metaproj" (def
       ault target) (3) ->
       "C:\dev\ros2\build\uncrustify\external_uncrustify.vcxproj" (default targ
       et) (5) ->
       (CustomBuild target) ->
         C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCo
       mmon.targets(170,5): error MSB6006: "cmd.exe" exited with code 1. [C:\de
       v\ros2\build\uncrustify\external_uncrustify.vcxproj]

    0 Warning(s)
    1 Error(s)

Time Elapsed 00:00:03.24

<== Command 'C:\dev\ros2\build\uncrustify\cmake__build.bat C:\Program Files (x86
)\MSBuild\14.0\bin\amd64\msbuild.EXE /m C:\dev\ros2\build\uncrustify\uncrustify.
sln' failed with exit code '1'

C:\dev\ros2>

dirk-thomas, 2015-07-17:
I will reopen the ticket since it still happens for you.
But since it works for us and we don't know a way to reproduce you problem we can't help much in trouble shooting it.

I assume you use the latest state from all repositories as well as Visual Studio 2015. May be it helps if you post more information about your exact environment and versions as well as the command sequence you use.

spiked3, 2015-07-17:
I am usually up to date on everything and that might be part of the problem. I noticed one of the recommended packages would not install. cppcheck is in a choco 'pending' state so I downloaded the newer .msi and installed manually. Other than that nothing unusual I can think of.

dirk-thomas, 2015-07-17:
The console output of our CI job (http://ci.ros2.org/job/ros2_batch_ci_windows/13/consoleFull) only indicates warnings during the migration of the project file:

    Migration completed successfully, but some warnings were detected during migration.

while your contains errors:

    Some errors occurred during migration. For more information, see the migration report:
    C:\dev\ros2\src\ament\uncrustify\win32\UpgradeLog.htm

Can you please check the content of that upgrade log.

spiked3, 2015-07-17:
It points to the generic migration page;  
https://msdn.microsoft.com/en-US/library/hh266747(v=vs.120).aspx

But I notice a comment;
You can use Visual Studio 2013 to open a C++ project that was created in Visual Studio 2012 or Visual Studio 2010 SP1. If you want to use the Visual Studio 2013 build environment to build a project that was created in Visual Studio 2012, you must have both versions of Visual Studio installed on the same computer.

I'm wondering if I would need to have vs2010 installed? (which I do not)

dirk-thomas, 2015-07-17:
We certainly have only VS 2015 on our build slave.

Is the message you posted a 1-to-1 copy from the upgrade log? If yes, are you using Visual Studio 2013? For our CI job the log looks like this: http://ci.ros2.org/job/ros2_batch_ci_windows/ws/workspace/src/ament/uncrustify/win32/UpgradeLog.htm

spiked3, 2015-07-17:
<!DOCTYPE html>

 <html xmlns:msxsl="urn:schemas-microsoft-com:xslt"><head><meta content="en-us" http-equiv="Content-Language" /><meta content="text/html; charset=utf-16" http-equiv="Content-Type" /><title _locID="ConversionReport0">
          Migration Report
        </title><style> 
                    /* Body style, for the entire document */
                    body
                    {
                        background: #F3F3F4;
                        color: #1E1E1F;
                        font-family: "Segoe UI", Tahoma, Geneva, Verdana, sans-serif;
                        padding: 0;
                        margin: 0;
                    }

                    /* Header1 style, used for the main title */
                    h1
                    {
                        padding: 10px 0px 10px 10px;
                        font-size: 21pt;
                        background-color: #E2E2E2;
                        border-bottom: 1px #C1C1C2 solid; 
                        color: #201F20;
                        margin: 0;
                        font-weight: normal;
                    }

                    /* Header2 style, used for "Overview" and other sections */
                    h2
                    {
                        font-size: 18pt;
                        font-weight: normal;
                        padding: 15px 0 5px 0;
                        margin: 0;
                    }

                    /* Header3 style, used for sub-sections, such as project name */
                    h3
                    {
                        font-weight: normal;
                        font-size: 15pt;
                        margin: 0;
                        padding: 15px 0 5px 0;
                        background-color: transparent;
                    }

                    /* Color all hyperlinks one color */
                    a
                    {
                        color: #1382CE;
                    }

                    /* Table styles */ 
                    table
                    {
                        border-spacing: 0 0;
                        border-collapse: collapse;
                        font-size: 10pt;
                    }

                    table th
                    {
                        background: #E7E7E8;
                        text-align: left;
                        text-decoration: none;
                        font-weight: normal;
                        padding: 3px 6px 3px 6px;
                    }

                    table td
[... 178 lines cut ...]
              toggleTableRowsByName('MessageRowHeaderHide' + projectName); 
            }
            
            function ScrollToFirstVisibleMessage(projectName)
            {
              if(!projectName || 0 === projectName.length)
              {
                return; 
              }
              
              // First try the 'Show messages' row
              if(!scrollToFirstVisibleRow('MessageRowHeaderShow' + projectName))
              {
                // Failed to find a visible row for 'Show messages', try an actual message row 
                scrollToFirstVisibleRow('MessageRowClass' + projectName); 
              }
            }
           </script></head><body><h1 _locID="ConversionReport">
          Migration Report - </h1><div id="content"><h2 _locID="OverviewTitle">Overview</h2><div id="overview"><table><tr><th></th><th _locID="ProjectTableHeader">Project</th><th _locID="PathTableHeader">Path</th><th _locID="ErrorsTableHeader">Errors</th><th _locID="WarningsTableHeader">Warnings</th><th _locID="MessagesTableHeader">Messages</th></tr><tr><td class="IconErrorEncoded" /><td><strong><a href="#Uncrustify">Uncrustify</a></strong></td><td>Uncrustify_Win32_2011.vcxproj</td><td class="textCentered"><a href="#UncrustifyError">1</a></td><td class="textCentered"><a>0</a></td><td class="textCentered"><a href="#">0</a></td></tr><tr><td class="IconSuccessEncoded" /><td><strong><a href="#Solution"><span _locID="OverviewSolutionSpan">Solution</span></a></strong></td><td>Uncrustify_Win32_2011.sln</td><td class="textCentered"><a>0</a></td><td class="textCentered"><a>0</a></td><td class="textCentered"><a href="#" onclick="ScrollToFirstVisibleMessage('Solution'); return false;">1</a></td></tr></table></div><h2 _locID="SolutionAndProjectsTitle">Solution and projects</h2><div id="messages"><a name="Uncrustify" /><h3>Uncrustify</h3><table><tr id="UncrustifyHeaderRow"><th></th><th class="messageCell" _locID="MessageTableHeader">Message</th></tr><tr name="ErrorRowClassUncrustify"><td class="IconErrorEncoded"><a name="UncrustifyError" /></td><td class="messageCell"><strong>Uncrustify_Win32_2011.vcxproj:
        </strong><span>The application which this project type is based on was not found. Please try this link for further information: http://go.microsoft.com/fwlink/?LinkId=500718&amp;projecttype=8BC9CEB8-8B4A-11D0-8D11-00A0C91BC942</span></td></tr></table><a name="Solution" /><h3 _locID="ProjectDisplayNameHeader">Solution</h3><table><tr id="SolutionHeaderRow"><th></th><th class="messageCell" _locID="MessageTableHeader">Message</th></tr><tr name="MessageRowHeaderShowSolution"><td class="IconInfoEncoded" /><td class="messageCell"><a _locID="ShowAdditionalMessages" href="#" name="SolutionMessage" onclick="ToggleMessageVisibility('Solution'); return false;">
          Show 1 additional messages
[... cut ...] onMessage" /></td><td class="messageCell"><strong>Uncrustify_Win32_2011.sln:
        </strong><span>The solution file does not require migration.</span></td></tr><tr style="display: none" name="MessageRowHeaderHideSolution"><td class="IconInfoEncoded" /><td class="messageCell"><a _locID="HideAdditionalMessages" href="#" name="SolutionMessage" onclick="ToggleMessageVisibility('Solution'); return false;">
          Hide 1 additional messages
        </a></td></tr></table></div></div></body></html>

spiked3, 2015-07-17:
well that didnt work well, and I cant post zips. 
[image]

dirk-thomas, 2015-07-17:
The error link points to VS 2013 documentation and says that it couldn't find the right project type. Do you have Visual C++ installed? Do you also have VS 2013 installed beside VS 2015?

spiked3, 2015-07-17:
I have vs2013 and vs 2015, but I have been having trouble with c++ in vs2015.  Is that the first c++ program in the build?  It ran for quite a while before it got there.

dirk-thomas, 2015-07-17:
Yes, that is the first C++ program in the build. You should make sure to only have VS 2015 in the environment and that the build is actually using VS 2015. Especially the following packages won't compile with VS 2013 anyway.

spiked3, 2015-07-17:
I definitely have vs 2015 installed - about 10 times in the last 3 days trying to resolve another c++ problem with it.  So it is probably my compiler. sigh.  Ok, I will dig more into it on this end.

spiked3, 2015-07-18:
I finally got vs 2015 straightened out (I hope for good) and it did make it past this.
It is now failing @geometry_interfaces - will look to see if that is the same point as your build system.
But you can close this one for good. Thanks for your help.

dirk-thomas, 2015-07-18:
Thanks for digging into it.

closed 2015-07-18 by dirk-thomas