ros2/examples #40: [Suggestion] Disable build for example using opencv

ruffsl, 2015-07-29:
Just a quick remark as I was following the example build from the wiki to create a onbuild Docker hub repo for ros2. The current wiki instruction had to be edited to include apt-get'ing `libopencv-dev` package for the cam2image example. It'd just be nice to add a note on how to use a flag to disable it to alevate the opencv dependance, as l libopencv is quite large. Just a suggestion

dirk-thomas, 2015-07-29:
There is no flag to disable OpenCV. The only option is to remove / ignore the `cam2image` package from the workspace.

ruffsl, 2015-07-29:
Where can I find the details on using the ament CLI to do this?
I've been looking around here so far:
https://github.com/ament/ament_tools/blob/master/doc/index.rst

Or do I just somthing simalure to [`CATKIN_IGNORE`](https://github.com/ros-infrastructure/rep/blob/master/rep-0128.rst)?

dirk-thomas, 2015-07-29:
You can either drop an `AMENT_IGNORE` file in any folder to ignore the subtree or remove the folder from the workspace.

dirk-thomas, 2015-07-30:
Is there anything more to do for this? Can you please add a comment to the instructions how to disable the package using OpenCV if that would have helped you building the code?

ruffsl, 2015-07-31:
Done, added relevant note to build section in examples wiki.

closed 2015-07-31 by ruffsl