ros2/examples #43: Examples compilation errors (related to opensplice) when using VS2015 final release.

j-rivero, 2015-08-07:
We are getting a new error, seems like it was not present on VS2015 RC (14.0.22816.0) nor other compilers, when compiling the examples with the final (both Express and Professional) VS2015.

The [error in jenkins](http://ci.ros2.org/job/ros2_batch_ci_windows/125/console):

```bash
c:\jenkins\workspace\ros2_batch_ci_windows\workspace\build\example_interfaces\rosidl_typesupport_opensplice_cpp\example_interfaces\srv\dds_opensplice\AddTwoInts_Request_Dcps_impl.h(32): 
error C2248: 'DDS::TypeSupportFactory::TypeSupportFactory': cannot access private member declared in class 'DDS::TypeSupportFactory' [C:\Jenkins\workspace\ros2_batch_ci_windows\workspace\build\example_interfaces
```

Looks like a typical error of accessing to a default constructor under insufficient permission.

Probably the high level is placed in our examples code is being generated from the [examples interfaces IDL code](https://github.com/ros2/examples/blob/master/example_interfaces/srv/AddTwoInts.srv) which generates the `Sample_AddTwoInts_Response_Dcps_impl.h` (this file appears explicitly in the error message) containing:

```c++
class ROSIDL_TYPESUPPORT_OPENSPLICE_CPP_PUBLIC_example_interfaces Sample_AddTwoInts_Response_TypeSupportFactory : public ::DDS::TypeSupportFactory_impl
            {
            public:
                Sample_AddTwoInts_Response_TypeSupportFactory() {}
                virtual ~Sample_AddTwoInts_Response_TypeSupportFactory() {}
            private:
                ::DDS::DataWriter_ptr
                create_datawriter (gapi_dataWriter handle);
           
                ::DDS::DataReader_ptr
                create_datareader (gapi_dataReader handle);
           
                ::DDS::DataReaderView_ptr
                create_view (gapi_dataReaderView handle);
}; <-- line 32
```
This class inherited [DDS::TypeSupportFactory_impl class](https://github.com/PrismTech/opensplice/blob/master/src/api/dcps/ccpp/include/ccpp_TypeSupport_impl.h#L25). All the code seems sane to me here. The error does not point to this clase but the base class, [`DDS::TypeSupportFactory`](https://github.com/PrismTech/opensplice/blob/master/src/api/dcps/sacpp/include/dds_dcps.h) which defined the default constructor to protected.

I could imaging two source of errors, if my approaching is right: a compiler bug that got introduced in the [last changes of VS 2015 final release](http://blogs.msdn.com/b/vcblog/archive/2015/07/01/c-compiler-front-end-fixes-in-vs2015.aspx) or not a bug, but a feature, in the sense of stricter rule  for visibility permissions on class inheritance that requires fixes in the code (I can not see which ones).

j-rivero, 2015-08-07:
I tried to reproduce the inheritance structure to quick check but it compiles fine (`cl /EHsc foo.ccp`).

```c++
class TypeSupportFactory
{
   protected:
      TypeSupportFactory () {};
      ~TypeSupportFactory () {};
   private:
      TypeSupportFactory (const TypeSupportFactory &) {};
      TypeSupportFactory & operator = (const TypeSupportFactory &) {};
};

class TypeSupportFactory_impl :
    public TypeSupportFactory
{
    public:
        TypeSupportFactory_impl() {};
        virtual ~TypeSupportFactory_impl() {};
};

class Interface :
    public TypeSupportFactory_impl
{

    public:
        Interface () {};
};

int main()   
{
    Interface foo;
    return 1;
}
```

dirk-thomas, 2015-08-07:
Afaik this is valid code until you try to either copy construct foo or use the assignment operator on it. You could try adding either of the fofllowing which will let the compiler generate the default versions of these functions:

    Interface bar(foo);

or

    Interface bar = foo;

j-rivero, 2015-08-07:
> Afaik this is valid code until you try to either copy construct foo or use the assignment operator on it.

I think so. Looks like other projects are [suffering from the same problem](http://stackoverflow.com/questions/31679144/how-to-build-log4cxx-in-visual-studio-2015) which was caused by a new implementation of the STL library (move semantics went into it). So not really a compiler bug. Seems like we need to patch Opensplice code and give it a try.

dirk-thomas, 2015-08-08:
Closing this - see latest comment on ros2/ros2#72.

closed 2015-08-08 by dirk-thomas