ros2/examples #52: ROS2 installation on MacOS

ainchauspe, 2015-08-25:
Hi @wjwwood,
This is Anita from Erle Robotics and I´m new with ROS. I´m trying to install ROS2 on Mac following these instructions: https://github.com/ros2/examples/wiki
However, I´m stuck in the very first command: 
```
sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu `lsb_release -cs` main" > /etc/apt/sources.list.d/ros-latest.list'
sh: lsb_release: command not found
``` 
I´ve searched it on the net but I have not been able to find a solution, I do not know how to translate those steps in Mac. Could you guide me in the installation of ROS2 on Mac?
Thanks for all,

Anita

wjwwood, 2015-08-25:
Hi Anita,

That command is Ubuntu specific and those instructions are not generalizable to OS X. I'm not sure we have OS X only instructions yet.

@tfoote is currently refactoring the installation instructions. Maybe it would be best to hold off until he's done.

jacquelinekay, 2015-09-08:
@ ainchauspe, can you confirm that the updated instructions on this page work for you?

https://github.com/ros2/ros2/wiki/OSX-Development-Setup

ainchauspe, 2015-09-09:
Hi, it works perfectly for me!

closed 2015-09-09 by wjwwood