ros2/examples #76: alpha2 build error

flobotics, 2015-11-21:
hi,
just a report: I build on rpi2B (arm) and when i run ament build --cmake-args -DCMAKE_BUILD_TYPE=Debug i get this error:

```
make[2]: *** [CMakeFiles/add_two_ints_client_async__rmw_opensplice_cpp.dir/src/services/add_two_ints_client_async.cpp.o] Error 1
make[2]: *** [CMakeFiles/add_two_ints_client_async.dir/src/services/add_two_ints_client_async.cpp.o] Error 1
make[1]: *** [CMakeFiles/add_two_ints_client_async.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/add_two_ints_client_async__rmw_opensplice_cpp.dir/all] Error 2
make[2]: *** [CMakeFiles/add_two_ints_client.dir/src/services/add_two_ints_client.cpp.o] Error 1
make[1]: *** [CMakeFiles/add_two_ints_client.dir/all] Error 2
/home/ubuntu/overlay_ws/src/examples/rclcpp_examples/src/services/add_two_ints_client.cpp: In function ‘example_interfaces::srv::AddTwoInts_Response_<std::allocator<void> >::SharedPtr send_request(rclcpp::node::Node::SharedPtr, rclcpp::client::Client<example_interfaces::srv::AddTwoInts>::SharedPtr, example_interfaces::srv::AddTwoInts_Request_<std::allocator<void> >::SharedPtr)’:
/home/ubuntu/overlay_ws/src/examples/rclcpp_examples/src/services/add_two_ints_client.cpp:36:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
make[2]: *** [CMakeFiles/add_two_ints_client__rmw_opensplice_cpp.dir/src/services/add_two_ints_client.cpp.o] Error 1
make[1]: *** [CMakeFiles/add_two_ints_client__rmw_opensplice_cpp.dir/all] Error 2
make: *** [all] Error 2

<== Command '. /home/ubuntu/overlay_ws/build/rclcpp_examples/cmake__build.sh && /usr/bin/make -j4 -l4' failed with exit code '2'
```

dirk-thomas, 2015-11-23:
The posted snippet only contains a warning. Please add the part which contains the error message.

Also the code from the tag `release-alpha2` seems to return correctly (https://github.com/ros2/examples/blob/release-alpha2/rclcpp_examples/src/services/add_two_ints_client.cpp#L27-L35). So I am not sure where the warning is coming from. Please make sure that you have checked out the correct tag.

flobotics, 2015-11-23:
ubuntu@ubuntu:~/overlay_ws/src/examples$ git describe --tags
release-alpha2-10-gf68758c

sorry for the warnings, the error is 4 times the same:
error: ‘rclcpp::executor::FutureReturnCode’ has not been declared

```
Building CXX object CMakeFiles/add_two_ints_client.dir/src/services/add_two_ints_client.cpp.o
Building CXX object CMakeFiles/add_two_ints_client_async.dir/src/services/add_two_ints_client_async.cpp.o
Building CXX object CMakeFiles/add_two_ints_client__rmw_opensplice_cpp.dir/src/services/add_two_ints_client.cpp.o
/home/ubuntu/overlay_ws/src/examples/rclcpp_examples/src/services/add_two_ints_client.cpp: In function ‘example_interfaces::srv::AddTwoInts_Response_<std::allocator<void> >::SharedPtr send_request(rclcpp::node::Node::SharedPtr, rclcpp::client::Client<example_interfaces::srv::AddTwoInts>::SharedPtr, example_interfaces::srv::AddTwoInts_Request_<std::allocator<void> >::SharedPtr)’:
/home/ubuntu/overlay_ws/src/examples/rclcpp_examples/src/services/add_two_ints_client.cpp:30:23: error: ‘rclcpp::executor::FutureReturnCode’ has not been declared
     rclcpp::executor::FutureReturnCode::SUCCESS)```

dirk-thomas, 2015-11-23:
`release-alpha2-10-gf68758c` implies that you are not using the tag but something ten commits ahead of the tag.

flobotics, 2015-11-23:
ah, on the site https://github.com/ros2/ros2/wiki/Linux-Development-Setup it uses 
```wget https://raw.githubusercontent.com/ros2/ros2/release-alpha2/ros2.repos```

and on the site https://github.com/ros2/ros2/wiki/Ament-Tutorial it uses
```wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos```

is this correct ?

dirk-thomas, 2015-11-23:
The ament tutorial is simply not specific to a single alpha release but works generically. You can either use all repos from the `release-alpha2` tag or all from `master`. But only the first one has been tested explicitly. While we do have CI jobs testing that master works we can't give the same "guarantees" as for the alpha packages.

flobotics, 2015-11-23:
ah, i used from https://github.com/ros2/ros2/wiki/Ament-Tutorial the line 
```git clone https://github.com/ros2/examples.git```

which seemed to lead me to release-alpha2-10-gf68758c. I checked out ```git checkout tags/release-alpha2``` and build again. Now it runs without any errors.

dirk-thomas, 2015-11-24:
Ok, if the problem is fixed for you please close this ticket.

closed 2015-11-24 by flobotics