ros2/examples #83: talker_listener test is flaky

gerkey, 2015-12-17:
Following the same procedure as ros2/system_tests#93:

~~~
while true; do nosetests3 -s build/rclcpp_examples/test_talker_listener__rmw_opensplice_cpp.py; done
~~~

It eventually stops with this output:

~~~
(test_executable_0) pid 28945: ['/home/gerkey/ros2_ws/build/rclcpp_examples/talker__rmw_opensplice_cpp', 'test_executable'] (all > console, InMemoryHandler: test_executable_0)
(test_executable_1) pid 28946: ['/home/gerkey/ros2_ws/build/rclcpp_examples/listener__rmw_opensplice_cpp', 'test_executable'] (all > console, InMemoryHandler: test_executable_1)
[test_executable_0] Publishing: 'Hello World: 1'
[test_executable_0] Publishing: 'Hello World: 2'
(test_executable_0) rc -15
~~~

The `talker` program has now exited, leaving behind the `listener`, who still wants something. The `listener` backtrace is:

~~~
#0  pthread_cond_wait@@GLIBC_2.3.2 () at ../nptl/sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
#1  0x00007feb1a6f109b in RTIOsapiSemaphore_take () from /usr/lib/libnddscore.so.5
#2  0x00007feb1a9013c7 in PRESWaitSet_wait () from /usr/lib/libnddscore.so.5
#3  0x00007feb1bf53e6d in DDS_WaitSet_waitI () from /usr/lib/libnddsc.so.5
#4  0x00007feb1ba23ba3 in DDSWaitSet_impl::wait(DDSConditionSeq&, DDS_Duration_t const&) () from /usr/lib/libnddscpp.so.5
#5  0x00007feb1d9e3c28 in int wait<ConnextStaticSubscriberInfo, ConnextStaticServiceInfo, ConnextStaticClientInfo>(rmw_subscriptions_t*, rmw_guard_conditions_t*, rmw_services_t*, rmw_clients_t*, rmw_time_t*) ()
   from /home/gerkey/ros2_ws/install/lib/librmw_connext_cpp.so
#6  0x00007feb1d9e040d in rmw_wait () from /home/gerkey/ros2_ws/install/lib/librmw_connext_cpp.so
#7  0x00007feb1d6e9d30 in rclcpp::executor::Executor::wait_for_work(std::chrono::duration<long, std::ratio<1l, 1000000000l> >) ()
   from /home/gerkey/ros2_ws/install/lib/librclcpp.so
#8  0x00007feb1d6eaea0 in rclcpp::executor::Executor::get_next_executable(std::chrono::duration<long, std::ratio<1l, 1000000000l> >)
    () from /home/gerkey/ros2_ws/install/lib/librclcpp.so
#9  0x00007feb1d6f2cfb in rclcpp::executors::single_threaded_executor::SingleThreadedExecutor::spin() ()
   from /home/gerkey/ros2_ws/install/lib/librclcpp.so
#10 0x00007feb1d6ee22c in rclcpp::spin(std::shared_ptr<rclcpp::node::Node>) () from /home/gerkey/ros2_ws/install/lib/librclcpp.so
#11 0x000000000043d75f in main ()
~~~

jacquelinekay, 2015-12-17:
How long does it usually take you to reproduce this? I'm trying a fix locally right now.

gerkey, 2015-12-17:
I got the deadlock after just a few iterations (also what Jenkins saw, I believe).

gerkey, 2015-12-17:
I'm seeing on the 3rd or 4th iteration pretty reliably.

jacquelinekay, 2015-12-17:
Even with my fix, which is quite braindead (adding another line of expectation to the `talker.txt` expectation file), I see deadlock in the test after running this for several minutes.

These are probably more symptoms of the nondeterministic startup behavior.

Ideally we could change this test so that the publisher gets killed when the listener receives a message. It would require breaking out of the `test_executable` template laid out in this package, however.

gerkey, 2015-12-17:
To share my understanding of what's happening here:

* The talker is running, printing output, and when that output matches, the launcher kills it.
* The listener is running, but it never receives any messages from the talker, presumably because by the time it's up and ready, the talker has already died (or perhaps the talker's published messages never even had a chance to go the wire before it was killed).

jacquelinekay, 2015-12-17:
I agree.

gerkey, 2015-12-17:
I don't see a good way to fix this problem. Options that come to mind are:

* Delay killing the child process following a match on its output (https://github.com/ros2/launch/blob/master/launch_testing/launch_testing/__init__.py#L65-L70). But delay by how much? At that point we can't do much more than sleep for a fixed time.
* Modify the talker to publish, wait, then print the message that's being matched (https://github.com/ros2/examples/blob/master/rclcpp_examples/src/topics/talker.cpp#L39-L40). But that would violate the idea of these tests, which is that they're intended to apply some minimal check to the example programs without modifying the example programs themselves.
* Extend the launch API to, for example, add to `ProcessDescriptor` a Boolean `marked_for_termination` member field that can be set and checked from the outside (https://github.com/ros2/launch/blob/master/launch/launch/__init__.py#L64). Then, where we're currently calling `terminate()` on our child, we'd instead do something like mark the child for termination, then iterate over all the other processes, and if they've all been marked, then really kill them. But that would be adding a poorly specified feature to the launch API just to support a pretty narrow test case.

jacquelinekay, 2015-12-17:
I'll take over this for the moment. I think I have a decent approach, but it requires a little bit more code.

dirk-thomas, 2015-12-17:
The unit tests already use a simple and clean approach for this: only a condition on the listener process can abort the launch file. The talker is simply not able to make that decision. See https://github.com/ros2/system_tests/blob/master/test_communication/test/test_publisher_subscriber.py.in

jacquelinekay, 2015-12-17:
Yes, currently all the unit tests in this package follow a template laid
out in test_executables_example.py.in, which is overly restrictive. My plan
is to write a different script which follows the same pattern as the
publisher/subscriber test.

https://github.com/ros2/examples/blob/master/rclcpp_examples/test/test_executables_example.py.in

gerkey, 2015-12-17:
Ah, @jacquelinekay, I just noticed that you're working on this. Fine by me.

In case it's helpful, here's my solution:

* https://github.com/ros2/examples/compare/fix_talker_listener_test?expand=1
* https://github.com/ros2/launch/compare/fix_talker_listener_test?expand=1

It seems to work for the talker listener case, and I don't expect it to have any effect on existing single-process tests. We might want to verify the order of declaration (in `CMakeLists.txt`) of the other multi-process tests to ensure that the "right" one is listed first, putting it in charge.

PR #84 (2015-12-18): Fix talker listener test
Fixes #83

I'm pull requesting this for @gerkey and queued a job that will repeat the test 20 times or until failure (simulating the conditions of the nightly job). http://ci.ros2.org/job/ci_osx/658
closed 2015-12-18 by gerkey