ros2/examples #95: Errors in parameters examples using FastRTPS

ernestmc, 2016-03-18:
I'm debugging [the errors](http://ci.ros2.org/view/nightly/job/nightly_linux_release/15/testReport/) that arise on the nightly linux release in the parameters examples using FastRTPS.
Example error:
```
$ ./list_parameters__rmw_fastrtps_cpp 
terminate called after throwing an instance of 'eprosima::fastcdr::exception::BadParamException'
  what():  Bad parameter
Aborted (core dumped)
```

What I've found so far:
* The examples fail only in the release build
* They work fine in the default build and in the debug build.
```
$ ./list_parameters__rmw_fastrtps_cpp 
Parameter name: bar
Parameter name: foo
Parameter name: foo.first
Parameter name: foo.second
Parameter prefix: foo
```

* I've tracked the exception and it's being raised [here](https://github.com/eProsima/Fast-CDR/blob/master/src/cpp/Cdr.cpp#L1259) when fast-cdr tries to deserialize a bool.

Some questions:

1. What can be changing in the release build to produce this?
1. What kind of build is the default build (using no specific parameters)?

jacquelinekay, 2016-03-18:
Perhaps there is a race condition in the serialization code in which the memory location holding the octect corresponding to a bool is written over by another thread. Then when the message gets deserialized, FastCDR reads a value greater than 1, causing the exception. If the 

The default build has a build type of "None", which compiles the code with no optimizations and no debug symbols.
"Debug" compiles with optimizations and debug symbols.
"Release" compiles with -O3 and no debug symbols.
"RelWithDebInfo" compiles with -O2 and debug symbols.

https://cmake.org/Wiki/CMake_Useful_Variables#Compilers_and_Tools

ernestmc, 2016-03-18:
Thanks @jacquelinekay . I'll keep looking into this. Maybe the higher optimization level causes some portion of the code to run faster and that creates the race condition?

jacquelinekay, 2016-03-25:
As a side note, I compiled ROS 2 with `RelWithDebInfo` recently and all the parameter tests failed. The `local_parameters` test segfaults in a different place than where you reprted, however.

jacquelinekay, 2016-03-25:
One hint might be this lint error from cppcheck that only shows up on OSX in Release mode:

http://ci.ros2.org/view/nightly/job/nightly_osx_release/29/testReport/projectroot.Users.osrf.jenkins.workspace.nightly_osx_release.ws.src.eProsima.ROS-RMW-Fast-RTPS-cpp/rmw_fastrtps_cpp/cppcheck/

ernestmc, 2016-05-25:
So the story goes on...
I'm tracing the serialization for a bool inside Cdr, specifically this function https://github.com/eProsima/Fast-CDR/blob/master/src/cpp/Cdr.cpp#L514-L531

I added code to print the stack trace and the ```bool_t``` parameter value as soon as we enter the function. I got the following output:
```
$ ./list_parameters
[CDR] bool_t=5

backtrace() returned 11 addresses
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/libfastcdr.so(_ZN8eprosima7fastcdr3Cdr9serializeEb+0x39) [0x7f0196faf5c9]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librmw_fastrtps_cpp.so(+0x1526d) [0x7f01976d526d]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librmw_fastrtps_cpp.so(+0x15258) [0x7f01976d5258]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librmw_fastrtps_cpp.so(+0x14b9d) [0x7f01976d4b9d]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librmw_fastrtps_cpp.so(+0x1765d) [0x7f01976d765d]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librmw_fastrtps_cpp.so(rmw_send_request+0x8a) [0x7f01976c734a]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librclcpp.so(_ZN6rclcpp16parameter_client21AsyncParametersClient14set_parametersERKSt6vectorINS_9parameter16ParameterVariantESaIS4_EESt8functionIFvSt13shared_futureIS2_IN14rcl_interfaces3msg20SetParametersResult_ISaIvEEESaISF_EEEEE+0x279) [0x7f0198624709]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librclcpp.so(_ZN6rclcpp16parameter_client20SyncParametersClient14set_parametersERKSt6vectorINS_9parameter16ParameterVariantESaIS4_EE+0x42) [0x7f0198624da2]
./list_parameters(main+0x1ab) [0x40397b]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf5) [0x7f0197d0eec5]
./list_parameters() [0x40405f]

-----------------

[CDR] bool_t=185

backtrace() returned 11 addresses
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/libfastcdr.so(_ZN8eprosima7fastcdr3Cdr9serializeEb+0x39) [0x7f0196faf5c9]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librmw_fastrtps_cpp.so(+0x1526d) [0x7f01976d526d]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librmw_fastrtps_cpp.so(+0x15258) [0x7f01976d5258]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librmw_fastrtps_cpp.so(+0x14b9d) [0x7f01976d4b9d]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librmw_fastrtps_cpp.so(+0x1765d) [0x7f01976d765d]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librmw_fastrtps_cpp.so(rmw_send_request+0x8a) [0x7f01976c734a]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librclcpp.so(_ZN6rclcpp16parameter_client21AsyncParametersClient14set_parametersERKSt6vectorINS_9parameter16ParameterVariantESaIS4_EESt8functionIFvSt13shared_futureIS2_IN14rcl_interfaces3msg20SetParametersResult_ISaIvEEESaISF_EEEEE+0x279) [0x7f0198624709]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librclcpp.so(_ZN6rclcpp16parameter_client20SyncParametersClient14set_parametersERKSt6vectorINS_9parameter16ParameterVariantESaIS4_EE+0x42) [0x7f0198624da2]
./list_parameters(main+0x1ab) [0x40397b]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf5) [0x7f0197d0eec5]
./list_parameters() [0x40405f]

-----------------

[CDR] bool_t=109

backtrace() returned 11 addresses
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/libfastcdr.so(_ZN8eprosima7fastcdr3Cdr9serializeEb+0x39) [0x7f0196faf5c9]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librmw_fastrtps_cpp.so(+0x1526d) [0x7f01976d526d]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librmw_fastrtps_cpp.so(+0x15258) [0x7f01976d5258]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librmw_fastrtps_cpp.so(+0x14b9d) [0x7f01976d4b9d]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librmw_fastrtps_cpp.so(+0x1765d) [0x7f01976d765d]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librmw_fastrtps_cpp.so(rmw_send_request+0x8a) [0x7f01976c734a]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librclcpp.so(_ZN6rclcpp16parameter_client21AsyncParametersClient14set_parametersERKSt6vectorINS_9parameter16ParameterVariantESaIS4_EESt8functionIFvSt13shared_futureIS2_IN14rcl_interfaces3msg20SetParametersResult_ISaIvEEESaISF_EEEEE+0x279) [0x7f0198624709]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librclcpp.so(_ZN6rclcpp16parameter_client20SyncParametersClient14set_parametersERKSt6vectorINS_9parameter16ParameterVariantESaIS4_EE+0x42) [0x7f0198624da2]
./list_parameters(main+0x1ab) [0x40397b]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf5) [0x7f0197d0eec5]
./list_parameters() [0x40405f]

-----------------

[CDR] bool_t=0

backtrace() returned 11 addresses
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/libfastcdr.so(_ZN8eprosima7fastcdr3Cdr9serializeEb+0x39) [0x7f0196faf5c9]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librmw_fastrtps_cpp.so(+0x1526d) [0x7f01976d526d]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librmw_fastrtps_cpp.so(+0x15258) [0x7f01976d5258]
/home/ecorbellini/repos/osrf/ros2_master_release_modify/install/lib/librmw_fastrtps_cpp.so(+0x14b9d) [0x7f01976d4b9d]
[... cut ...]
-----------------

terminate called after throwing an instance of 'eprosima::fastcdr::exception::BadParamException'
  what():  Bad parameter
Aborted (core dumped)
```

Basically I can notice two things:
 1. the serialize function is called 6 times when we only set one bool parameter in the example. In fact I've tested and it is called once for every parameter that we set.
 1. the ```bool_t``` parameter has random garbage when the function is called.

Any thoughts are welcomed while I continue investigating.

jacquelinekay, 2016-06-02:
https://github.com/eProsima/ROS-RMW-Fast-RTPS-cpp/blob/master/rmw_fastrtps_cpp/src/MessageTypeSupport.cpp#L17-L18
https://github.com/eProsima/ROS-RMW-Fast-RTPS-cpp/blob/master/rmw_fastrtps_cpp/src/TypeSupport.cpp#L180-L182
https://github.com/eProsima/ROS-RMW-Fast-RTPS-cpp/blob/master/rmw_fastrtps_cpp/src/TypeSupport.cpp#L327-L329

have you looked at these lines at all? 

If everything was working properly I would expect parameter events to never work (an exception would be thrown). So I can't explain the behavior you see but I suspect this might be related.

jacquelinekay, 2016-06-07:
This was fixed in a recent patch by Dirk.

closed 2016-06-07 by jacquelinekay

ernestmc, 2016-06-07:
Great! @jacquelinekay can you point to the fix?

jacquelinekay, 2016-06-07:
https://github.com/eProsima/ROS-RMW-Fast-RTPS-cpp/pull/37