ros2/examples #100: service add two ints client/service example test is flakey

wjwwood, 2016-04-01:
See:

http://ci.ros2.org/view/nightly/job/nightly_osx_debug/29/testReport/projectroot.Users.osrf.jenkins.workspace.nightly_osx_debug.ws.src.ros2.examples/rclcpp_examples/test_example_add_two_ints_server_add_two_ints_client__rmw_fastrtps_cpp/

If you search through the console output, you'll see that it is failing due to the test timeout. I can reproduce this on my machine.

If I start the server and then run the client in a loop I cannot reproduce the issue. So I'm going to look into a startup race condition again, as I believe it is still this issue: https://github.com/ros2/examples/pull/93#issuecomment-198585326

I'm not sure I can solve it, but I am going to try some things and at least we'll have an issue tracking this (I couldn't find one that is already open other than https://github.com/ros2/rclcpp/issues/201 which is tracking a related feature but not the issue). I think tracking the issue is a good idea because it took me a while to figure out we had already looked into this because I looked at the open tickets on this repository and didn't see anything.

jacquelinekay, 2016-04-04:
have we observed this as flaky on fastrtps only or on multiple middlewares?

wjwwood, 2016-04-04:
I believe more that one, but we couldn't implement the fix @gerkey originally wanted partly because the rmw layer for FastRTPS doesn't have topic listing (if I remember correctly).

wjwwood, 2016-06-09:
This should hopefully be addressed by https://github.com/ros2/examples/pull/110 (at least for opensplice) once it is merged.

dirk-thomas, 2016-10-03:
Since the referenced PR which is supposed to address this has been merged has this been seen since then or can this be closed?

wjwwood, 2016-10-03:
Yup.

closed 2016-10-03 by wjwwood