ros2/examples #102: Client shutdown causes OpenSplice errors on server

gerkey, 2016-04-05:
When testing the binary package for alpha5 on Windows (10), I found that the `add_two_ints_server` would start printing a stream of errors like the one below after a client connected and (successfully) called the service.

~~~
========================================================================================
Report      : ERROR
Date        : Tue Apr 05 14:31:01 Pacific Daylight Time 2016
Description : UDP recvmsg sock 1816: ret -1 errno 10054

Node        : DESKTOP-LM8SBG1
Process     : add_two_ints_server.exe <4532>
Thread      : recv 3364
Internals   : V6.4.140407OSS/3b83e82/3b83e82/ddsi2/ddsi_socket.c/45/0/1459891861.893614141
========================================================================================
~~~

dhood, 2016-04-05:
On Trusty (today) I found that after a delay of around 10 seconds after the client disconnected I got:

```
========================================================================================
Report      : WARNING
Date        : Tue Apr 05 14:58:44 PDT 2016
Description : Missed heartbeat for node - heartbeat id : 70813206. (DCPSPublication state 1)
Node        : esteve
Process     : add_two_ints_server__rmw_opensplice_cpp <30948>
Thread      : OSPL Garbage Collector 7f5612d0b700
Internals   : V6.4.140407OSS///v_spliced/v_spliced.c/2439/0/1459893524.384434688

```
(I've had this happen before (alpha3) on my machine at home and it led to a memory leak)

dirk-thomas, 2016-06-01:
Beside the OpenSplice error reports our code also fails to delete the participant:

```
Subscriber::delete_datareader: precondition not met: the operation is called on a different Subscriber than the one which created it, or the DataReader contains one or more ReadCondition or QueryCondition objects, or the DataReader still contains unreturned loans
DomainParticipant::delete_subscriber: precondition not met: the operation is called on a different DomainParticipant than was used to create it, or the Subscriber contains one or more DataReader objects
Error from Subscriber::delete_datareader in responder teardown
DomainParticipant::delete_topic: precondition not met: the operation is called on a different DomainParticipant than was used to create it, or the Topic is still referenced by other objects
Error from Participant::delete_subscriber in responder teardown
[rmw|error_handling.c:93] error string being overwritten: error not set, at ./ros2/rcl/rcl/src/rcl/service.c:172
[rmw|error_handling.c:93] error string being overwritten: failed to destroy responder: Error from Participant::delete_topic in responder teardown, at ./ros2/rmw_opensplice/rmw_opensplice_cpp/src/rmw_service.cpp:214
[rmw|error_handling.c:93] error string being overwritten: failed to delete contained entities of participant, at ./ros2/rmw_opensplice/rmw_opensplice_cpp/src/rmw_node.cpp:284
Error in destruction of rmw node handle: failed to delete participant, at ./ros2/rmw_opensplice/rmw_opensplice_cpp/src/rmw_node.cpp:289, at ./ros2/rcl/rcl/src/rcl/node.c:174
```

wjwwood, 2016-06-07:
I have a fix for this in my local working copy. Here's the diff until I or someone else has time to pr it:

```diff
diff --git a/rosidl_typesupport_opensplice_cpp/resource/srv__type_support.cpp.em b/rosidl_typesupport_opensplice_cpp/resource/srv__type_support.cpp.em
index 0e390b8..a64c75b 100644
--- a/rosidl_typesupport_opensplice_cpp/resource/srv__type_support.cpp.em
+++ b/rosidl_typesupport_opensplice_cpp/resource/srv__type_support.cpp.em
@@ -124,33 +124,33 @@ public:

     if (sample_infos.length() > 0 && sample_infos[0].valid_data) {
       sample = reinterpret_cast<Sample<@(__dds_msg_type_prefix)@(suffix)_> &>(dds_messages[0]);
-      status = typed_datareader->return_loan(dds_messages, sample_infos);
-      switch (status) {
-        case DDS::RETCODE_ERROR:
-          return "@(__dds_sample_type_prefix)@(suffix)_DataReader.return_loan failed with: "
-                 "an internal error has occurred";
-        case DDS::RETCODE_ALREADY_DELETED:
-          return "@(__dds_sample_type_prefix)@(suffix)_DataReader.return_loan failed with: "
-                 "this @(__dds_sample_type_prefix)@(suffix)_DataReader has already been deleted";
-        case DDS::RETCODE_OUT_OF_RESOURCES:
-          return "@(__dds_sample_type_prefix)@(suffix)_DataReader.return_loan failed with: "
-                 "out of resources";
-        case DDS::RETCODE_NOT_ENABLED:
-          return "@(__dds_sample_type_prefix)@(suffix)_DataReader.return_loan failed with: "
-                 "this @(__dds_sample_type_prefix)@(suffix)_DataReader is not enabled";
-        case DDS::RETCODE_PRECONDITION_NOT_MET:
-          return "@(__dds_sample_type_prefix)@(suffix)_DataReader.return_loan failed with: "
-                 "a precondition is not met, one of: "
-                 "the data_values and info_seq do not belong to a single related pair, or "
-                 "the data_values and info_seq were not obtained from this "
-                 "@(__dds_sample_type_prefix)@(suffix)_DataReader";
-        case DDS::RETCODE_OK:
-          *taken = true;
-          return nullptr;
-        default:
-          return "@(__dds_sample_type_prefix)@(suffix)_DataReader.return_loan failed with "
-                 "unknown return code";
-      }
+    }
+    status = typed_datareader->return_loan(dds_messages, sample_infos);
+    switch (status) {
+      case DDS::RETCODE_ERROR:
+        return "@(__dds_sample_type_prefix)@(suffix)_DataReader.return_loan failed with: "
+               "an internal error has occurred";
+      case DDS::RETCODE_ALREADY_DELETED:
+        return "@(__dds_sample_type_prefix)@(suffix)_DataReader.return_loan failed with: "
+               "this @(__dds_sample_type_prefix)@(suffix)_DataReader has already been deleted";
+      case DDS::RETCODE_OUT_OF_RESOURCES:
+        return "@(__dds_sample_type_prefix)@(suffix)_DataReader.return_loan failed with: "
+               "out of resources";
+      case DDS::RETCODE_NOT_ENABLED:
+        return "@(__dds_sample_type_prefix)@(suffix)_DataReader.return_loan failed with: "
+               "this @(__dds_sample_type_prefix)@(suffix)_DataReader is not enabled";
+      case DDS::RETCODE_PRECONDITION_NOT_MET:
+        return "@(__dds_sample_type_prefix)@(suffix)_DataReader.return_loan failed with: "
+               "a precondition is not met, one of: "
+               "the data_values and info_seq do not belong to a single related pair, or "
+               "the data_values and info_seq were not obtained from this "
+               "@(__dds_sample_type_prefix)@(suffix)_DataReader";
+      case DDS::RETCODE_OK:
+        *taken = true;
+        return nullptr;
+      default:
+        return "@(__dds_sample_type_prefix)@(suffix)_DataReader.return_loan failed with "
+               "unknown return code";
     }

     *taken = false;

```

codebot, 2016-11-16:
closed in favor of https://github.com/ros2/tutorials/issues/2

closed 2016-11-16 by codebot