ros2/examples #128: compile error

xuefengchang, 2016-10-18:
I installed ros2-alpha6 successfully from source. Then I createed the overlay workspace and git clone the ros2 examples. When I compile the examples I got the following errors:
/home/user/overlay_ws/src/examples/rclcpp_examples/src/services/add_two_ints_client_async.cpp:30:19: error: ‘class rclcpp::client::Client<example_interfaces::srv::AddTwoInts>’ has no member named ‘wait_for_service’
   while (!client->wait_for_service(1_s)) {
                   ^
/home/user/overlay_ws/src/examples/rclcpp_examples/src/services/add_two_ints_client.cpp: In function ‘int main(int, char**)’:
/home/user/overlay_ws/src/examples/rclcpp_examples/src/services/add_two_ints_client.cpp:50:19: error: ‘class rclcpp::client::Client<example_interfaces::srv::AddTwoInts>’ has no member named ‘wait_for_service’
   while (!client->wait_for_service(1_s)) {
                   ^
/home/user/overlay_ws/src/examples/rclcpp_examples/src/services/add_two_ints_client.cpp: In function ‘int main(int, char**)’:
/home/user/overlay_ws/src/examples/rclcpp_examples/src/services/add_two_ints_client.cpp:50:19: error: ‘class rclcpp::client::Client<example_interfaces::srv::AddTwoInts>’ has no member named ‘wait_for_service’
   while (!client->wait_for_service(1_s)) {
How can I solve this problem? Thanks.

dirk-thomas, 2016-10-18:
It sounds like you are mixing some ROS 2 packages from the Alpha 6 with the latest master of the example. You should stick to a consistent set of source. E.g. everything from the Alpha 8: https://github.com/ros2/ros2/blob/release-alpha8/ros2.repos

xuefengchang, 2016-10-18:
I am using Ubuntu 14.04 now, so I  choose the alpha-6 version. Are the other solutions?

dirk-thomas, 2016-10-18:
Alpha 6 is the last version supporting Ubuntu Trusty. But then you also have to stick with the Alpha 6 version of the examples. Newer versions of the examples will require newer versions of the underlying API and will therefore not compile against the Alpha 6 API (as you have noticed).

If you want to use newer versions of the examples or any of the ROS 2 packages you will have to consider moving to Ubuntu Xenial (at least within a VM or Docker.

Since the error is expected and I don't think there is a way to fix (mixing different versions on 14.04) I will go ahead and close this ticket. Please feel free to continue commenting / discussion in this ticket.

closed 2016-10-18 by dirk-thomas

xuefengchang, 2016-10-19:
Ok, I got it, Thank you very much!