ros2/examples #154: python services failing to import typesupport

dhood, 2016-12-14:
I am unable to run the rclpy server-client examples/tutorials, they all fail with:
(on OS X and linux, from source and the Dec 13 packaging job).

pub-sub working fine

```
$ add_two_ints_server_py
Traceback (most recent call last):
  File "/Users/dhood/Downloads/ros2-osx-packaging13dec/bin/add_two_ints_server_py", line 11, in <module>
    load_entry_point('rclpy-tutorials==0.0.0', 'console_scripts', 'add_two_ints_server_py')()
  File "/Users/dhood/Downloads/ros2-osx-packaging13dec/lib/python3.5/site-packages/services/add_two_ints_server_py.py", line 32, in main
    srv = node.create_service(AddTwoInts, 'add_two_ints', add_two_ints_callback)
  File "/Users/dhood/Downloads/ros2-osx-packaging13dec/lib/python3.5/site-packages/rclpy/node.py", line 84, in create_service
    raise NoTypeSupportImportedException
rclpy.exceptions.NoTypeSupportImportedException: no type_support imported
```

dirk-thomas, 2016-12-14:
The Python debug logger provides the following additional information:

```
Failed to import needed modules for type support:
Traceback (most recent call last):
  File "/home/dthomas/ros2/ws_for_bridge/install_isolated/example_interfaces/lib/python3.5/site-packages/example_interfaces/srv/_add_two_ints.py", line 20, in __import_type_support__
    'example_interfaces', rclpy_implementation)
  File "/home/dthomas/ros2/ws_for_bridge/install_isolated/rosidl_generator_py/lib/python3.5/site-packages/rosidl_generator_py/import_type_support_impl.py", line 53, in import_type_support
    return importlib.import_module(module_name, package=import_package)
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 986, in _gcd_import
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 666, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 577, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 906, in create_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
ImportError: dynamic module does not define module export function (PyInit_example_interfaces_s__rosidl_typesupport_introspection_c)
```

dirk-thomas, 2016-12-14:
The library does really not have that symbol. And looking at the code templates (https://github.com/ros2/rosidl/blob/a3074a616e437e62078dd9ba232fb8334191a20a/rosidl_generator_py/resource/_msg_support.entry_point.c.em#L132) that symbol is only generated for messages files.

Recently (#150) the only message file has been removed from `example_interfaces` which leaves only a service. That is likely the reason why the symbol is missing now.

dirk-thomas, 2016-12-14:
#155 resolves the problem temporary.

PR #155 (2016-12-14): readd dummy message
Fixes #154 for now.

Connect to #154.

For a real solution see ros2/rosidl#193
closed 2016-12-14 by dirk-thomas