ros2/examples #160: improve names of executables and mention them in the readme

dirk-thomas, 2016-12-19:
Currently the names are only discoverable by using the shell completion if you know that they start with `examples_rclcpp_minimal_`. They are not mentioned anywhere in the README files.

Also why having the suffix `_main` for `examples_rclcpp_minimal_client_main`?

codebot, 2016-12-19:
Yeah I wasn't sure how else to make sure they didn't collide with other binaries when they are installed to the global `bin` directory, so I put that long ugly prefix on them.

The idea for the `_main` suffix for `examples_rclcpp_minimal_client_main` is because it has a `main()` function, and at some point (e.g., Beta 2) I'm hoping to add more examples that do not have a `main()` but instead use the forthcoming handy macros and generic launcher to allow easy component-style coding.

mikaelarguedas, 2018-02-22:
These executables are now installed to libexec and renamed. The readme's dont refer to it yet so this is only documentation

jacobperron, 2019-01-25:
@davetcoleman My understanding is "composable" refers to the ability for a node to be used as part of a [Composition](https://index.ros.org/doc/ros2/Tutorials/Composition/) (running multiple nodes in a single process). The Action examples followed the naming convention from the minimal publisher and minimal subscriber examples. But, the executables that are not labeled as "not composable" are not composable either (not shared libraries)... Since there already exists a [composition demo](https://github.com/ros2/demos/tree/master/composition), I think that a different naming convention should be instated here. 

> The actions README should link back to its tutorial, if one exists. Perhaps that is where "composable" is defined?

Currently, a tutorial does not exist, but I think it's a good idea to link to a tutorial once it does.

davetcoleman, 2019-01-25:
Thanks for the explanation.

> Since there already exists a composition demo, I think that a different naming convention should be instated here.

You mean rename the source files, right? I don't understand all of this enough to provide input on that.

jacobperron, 2019-01-25:
> You mean rename the source files, right?

Yeah, the source files and executables.

It looks like the README's for [minimal_publisher](https://github.com/ros2/examples/blob/master/rclcpp/minimal_publisher/README.md) and [minimal_subscriber](https://github.com/ros2/examples/tree/master/rclcpp/minimal_subscriber) have explanations of "composable", so at the very least a similar description can be added to the action and service examples.

Perhaps I'm mistaken about the meaning of "composable" here, if all it means is subclassing from `Node`, then the naming scheme here might be fine. If so, we should make sure to add "composable" examples for services and actions.

davetcoleman, 2019-01-28:
I've submitted documentation as promised: https://github.com/ros2/examples/pull/224