ros2/examples #164: Add a parameter example

geoffviola, 2017-01-04:
It would be nice, if there was an example that uses the parameter service and follows the best practices from https://discourse.ros.org/t/composition-and-parameters-best-practice-suggestions/1001/2. Then, it can be tested with the ros2param tool.

wjwwood, 2017-01-04:
Yeah, we didn't get time to update things here, but in the mean time you can see "non-composition" examples that used to be here in the demos repository:

https://github.com/ros2/demos/tree/master/demo_nodes_cpp/src/parameters

geoffviola, 2017-01-04:
Thanks, but none of those examples use the register_param_change_callback node method from https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/node.hpp#L268. Those nodes setup their own server and client, which is the less desirable way to parameters.

wjwwood, 2017-01-04:
Ah, yeah that's true.

closed 2018-08-03 by mikaelarguedas