ros2/examples #170: Move example_interfaces to a separate repository

esteve, 2017-04-17:
The `example_interfaces` package is currently residing in the same repository as the `rclcpp` and `rclpy` examples, so if someone only needs the interfaces for building examples for other bindings (e.g. `rcljava` or `rclobjc`), they need to add an `AMENT_IGNORE` file to each package. It'd be a bit easier if the `example_interfaces` package could be fetched separately.

wjwwood, 2017-04-25:
Sounds good to me, I'll create the new repository. @dirk-thomas what script were you using recently to split up the rqt packages? If you send it to me I'll use it to break `example_interfaces` out of this repository into a new one.

Then we can make a pr here to remove the `example_interfaces` package and another pr to add the new repository to the repos file.

wjwwood, 2017-04-25:
New repo: https://github.com/ros2/example_interfaces

dirk-thomas, 2017-04-25:
I used this one: https://gist.github.com/dirk-thomas/62da12ccb47e455f518e156b18f4e417 (the first revision for `rqt_common_plugins`, the second revision for `rqt_robot_plugins`. Since I might remove the gist at some point here the extract:

```
export PKG=rqt_action

git clone git@github.com:ros-visualization/rqt_common_plugins.git $PKG
cd $PKG
git filter-branch --prune-empty --subdirectory-filter $PKG
git remote add new_origin git@github.com:ros-visualization/$PKG.git
git push -u new_origin master
cd ..
```

wjwwood, 2017-04-25:
Thanks @dirk-thomas.

Ok, I've moved the package to the new repository, and added `CONTRIBUTING.md`, `LICENSE`, and `README.md` to it in https://github.com/ros2/example_interfaces/commit/a7ccf7e515e7ebb98793b402ab575acc1b9665c6.

wjwwood, 2017-04-25:
CI, just doing macOS since it is just moving files around and the Linux machines are busy atm:

- macOS:
  - [image]

wjwwood, 2017-04-25:
Ok, should be good to go now.

closed 2017-04-25 by wjwwood

esteve, 2017-04-26:
@wjwwood thanks!