ros2/examples #193: all rclpy executor scripts fail on exit

dirk-thomas, 2017-12-06:
with something like:

```
Failed to fini 'rcl_publisher_t'
```

`custom_executor` also has the following error:

```
TypeError: 'tuple' object is not an iterator
```

sloretz, 2017-12-06:
**Question:** Should `executor.shutdown()` call `node.shutdown()` on each node that has been added to it? A benefit is convenience for the user. A negative is it eliminates the possibility of replacing an executor while keeping the nodes alive, though I'm not imaginative enough to think of a scenario where that would be useful.

The `Failed to fini 'rcl_publisher_t'` errors happen because the examples don't call `shutdown` on the nodes they create. At the end of the program the garbage collector calls `__del__` on the nodes, but `rclpy.shutdown()` has already been called so the rcl instance id doesn't match.

```
Exception ignored in: <bound method Node.__del__ of <examples_rclpy_executors.custom_executor.Estopper object at 0x7f2d7ab94128>>
Traceback (most recent call last):
  File "/workspace/ros2_ws/install_isolated/rclpy/lib/python3.5/site-packages/rclpy/node.py", line 300, in __del__
  File "/workspace/ros2_ws/install_isolated/rclpy/lib/python3.5/site-packages/rclpy/node.py", line 272, in destroy_node
RuntimeError: Failed to fini 'rcl_subscription_t': rcl node is invalid, rcl instance id does not match, at /workspace/ros2_ws/src/ros2/rcl/rcl/src/rcl/node.c:382
```

dirk-thomas, 2017-12-06:
I would lean toward "no".

The demo `talker` / `listener` e.g. call just:

```
node.destroy_node()
rclpy.shutdown()
```

The examples could do the same.

sloretz, 2017-12-06:
Fixed by #196

closed 2017-12-06 by sloretz