ros2/examples #199: No rule to make target '/usr/lib/libPocoFoundationd.so'

athackst, 2017-12-21:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 16.04
- Installation type:
  -binaries
- Version or commit hash:
  - 5c635c95b3ae46314fe56d446c1c36293f5ccf2f
- DDS implementation:
  - default

#### Steps to reproduce issue
Steps I took on a clean install of Ubuntu 16.04
``` bash
$ sudo apt update && sudo apt install curl
$ curl http://repo.ros2.org/repos.key | sudo apt-key add -
$ sudo sh -c 'echo "deb [arch=amd64,arm64] http://repo.ros2.org/ubuntu/main xenial main" > /etc/apt/sources.list.d/ros2-latest.list'
$ sudo apt update
$ sudo apt install `apt list "ros-ardent-*" 2> /dev/null | grep "/" | awk -F/ '{print $1}' | grep -v -e ros-ardent-ros1-bridge -e ros-ardent-turtlebot2- | tr "\n" " "`
$ sudo apt install python-argcomplete 
$ source /opt/ros/ardent/share/ros2cli/environment/ros2-argcomplete.bash
$ source /opt/ros/ardent/setup.bash
$ mkdir -p ros2_ws/src
$ cd ros2_ws/src
$ git clone https://github.com/ros2/examples.git
$ ament build --cmake-args -DCMAKE_BUILD_TYPE=Debug
```

#### Expected behavior
successful build
#### Actual behavior
```bash
Scanning dependencies of target composition_nodes
make[2]: *** No rule to make target '/usr/lib/libPocoFoundationd.so', needed by 'libcomposition_nodes.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[ 11%] Building CXX object CMakeFiles/composition_nodes.dir/src/publisher_node.cpp.o
[ 22%] Building CXX object CMakeFiles/composition_nodes.dir/src/subscriber_node.cpp.o
CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/composition_nodes.dir/all' failed
make[1]: *** [CMakeFiles/composition_nodes.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

<== Command '. /home/athackst/ros2_ws/build/examples_rclcpp_minimal_composition/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/home/athackst/ros2_ws/build/examples_rclcpp_minimal_composition' with exit code '2'
<== Command '. /home/athackst/ros2_ws/build/examples_rclcpp_minimal_composition/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/home/athackst/ros2_ws/build/examples_rclcpp_minimal_composition' with exit code '2'
```
#### Additional information
I was following the instructions here https://github.com/ros2/ros2/wiki/Ament-Tutorial minus some things that seemed related to a source build of ros2

I skipped the add some sources (https://github.com/ros2/ros2/wiki/Ament-Tutorial#add-some-sources), and went straight to the create an overlay (https://github.com/ros2/ros2/wiki/Ament-Tutorial#create-an-overlay)
If I'm doing a binary install do I still need to add those sources?
What did I miss?

dirk-thomas, 2017-12-21:
I haven't had a chance to reproduce this yet but it looks like some mismatch between what poco version is being used. It seems to look for the `libpoco-dev` package rather then the [poco_vendor](https://github.com/ros2/poco_vendor/) package. I assume the binaries only bring a release version of the vendor library and when you build in debug it falls back to the system version. A workaround is probably yo install the system dependency.

athackst, 2017-12-21:
I'm not sure, it says it finds it and I get this same error when I target Debug or Release

```bash
+++ Building 'examples_rclcpp_minimal_composition'
Running cmake because arguments have changed.
==> '. /home/athackst/ros2_ws/build/examples_rclcpp_minimal_composition/cmake__build.sh && /usr/bin/cmake /home/athackst/ros2_ws/src/examples/rclcpp/minimal_composition -DBUILD_TESTING=0 -DCMAKE_BUILD_TYPE=Debug -DCMAKE_INSTALL_PREFIX=/home/athackst/ros2_ws/install' in '/home/athackst/ros2_ws/build/examples_rclcpp_minimal_composition'
-- Found ament_cmake: 0.4.0 (/opt/ros/ardent/share/ament_cmake/cmake)
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Found class_loader: 1.0.0 (/opt/ros/ardent/share/class_loader/cmake)
-- Found poco_vendor: 1.0.0 (/opt/ros/ardent/share/poco_vendor/cmake)
-- Searching for Poco library...
--   Found Poco!
-- components found: Foundationd.
-- Found rclcpp: 0.4.0 (/opt/ros/ardent/share/rclcpp/cmake)
-- Found rmw_implementation_cmake: 0.4.0 (/opt/ros/ardent/share/rmw_implementation_cmake/cmake)
-- Found std_msgs: 0.4.0 (/opt/ros/ardent/share/std_msgs/cmake)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/athackst/ros2_ws/build/examples_rclcpp_minimal_composition
==> '. /home/athackst/ros2_ws/build/examples_rclcpp_minimal_composition/cmake__build.sh && /usr/bin/make -j4 -l4' in '/home/athackst/ros2_ws/build/examples_rclcpp_minimal_composition'
make[2]: *** No rule to make target '/usr/lib/libPocoFoundationd.so', needed by 'libcomposition_nodes.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[ 11%] Building CXX object CMakeFiles/composition_nodes.dir/src/subscriber_node.cpp.o
[ 22%] Building CXX object CMakeFiles/composition_nodes.dir/src/publisher_node.cpp.o
CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/composition_nodes.dir/all' failed
make[1]: *** [CMakeFiles/composition_nodes.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

<== Command '. /home/athackst/ros2_ws/build/examples_rclcpp_minimal_composition/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/home/athackst/ros2_ws/build/examples_rclcpp_minimal_composition' with exit code '2'
<== Command '. /home/athackst/ros2_ws/build/examples_rclcpp_minimal_composition/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/home/athackst/ros2_ws/build/examples_rclcpp_minimal_composition' with exit code '2'
```

athackst, 2017-12-21:
I could install libpocofoundation9v5 but it seems like this should come with the binaries if it's needed?

dirk-thomas, 2017-12-22:
> I could install `libpocofoundation9v5` but it seems like this should come with the binaries if it's needed?

Not necessarily. The package [poco_vendor](https://github.com/ros2/poco_vendor) should bring in the library. Can you please confirm that you have the following files which are part of the `ros-ardent-poco-vendor` Debian package:
* `/opt/ros/ardent/include/Poco/Poco.h`
* `/opt/ros/ardent/lib/libPocoFoundation.so`

If that is the case can you please try a clean release build. It is important to not reuse previous CMake caches since from a previous debug build the location in `/usr` might otherwise be cached. Please remove the build space and try a new build of the workspace with `--cmake-args -DCMAKE_BUILD_TYPE=Release`. I would expect that to find the Poco headers and libraries in `/opt/ardent`.

athackst, 2017-12-22:
The clean release build worked.

dirk-thomas, 2017-12-23:
I created ros2/poco_vendor#11 to describe the problem more precisely.

closed 2017-12-23 by dirk-thomas