ros2/examples #202: how to stop the timer in python for ros2?

lxbeyond, 2018-01-31:
if code my program in code style like publisher_member_function.py, and then I want to publish msg 100 times, and after that end the publisher, how to stop the timer, and exit the program ?

lxbeyond, 2018-01-31:
what's more, could you please provide an api manual for ros2 in c++ and python ? thanks a lot

lxbeyond, 2018-02-01:
@sloretz

dhood, 2018-02-01:
You should keep track of the iterations and then call `cancel()` on the timer.

We have generated API for C++ at docs.ros2.org (e.g.: http://docs.ros2.org/ardent/api/rclcpp/classrclcpp_1_1TimerBase.html) but not for python yet: you can check the code for `rclpy` in the meantime: https://github.com/ros2/rclpy/blob/1ef2924ef8e154c0553edf0fdba4840b08b728f8/rclpy/rclpy/timer.py

dhood, 2018-02-01:
you can take a look at this example for how to stop spinning after a number of iterations: https://github.com/ros2/demos/blob/58450594268e1540490239db31dcb413e37288b8/demo_nodes_py/demo_nodes_py/topics/talker_qos.py

e.g.:
```
ros2 run demo_nodes_py talker_qos --number_of_cycles 100
```

dhood, 2018-02-08:
@lxbeyond I'll close this issue; if there's anything else feel free to comment here or ask on answers.ros.org with the `ros2` tag.

closed 2018-02-08 by dhood

lxbeyond, 2018-02-11:
@dhood  thanks a lot