ros2/examples #208: Add an example using multi-threaded executors and callback groups

clalancette, 2018-06-05:
## Feature request

Add an example showing how to use multi-threaded executors and callback groups.

#### Feature description

ROS 2 has the ability to have callbacks executed on different threads, but there aren't any simple examples showing how to do this in the current codebase.  Adding an example something like https://github.com/clalancette/mtexec_example will show new users how to use the multi-threaded callbacks and callback groups.

mkhansen-intel, 2018-09-10:
@clalancette - I looked at your example and want to know if separate callback groups are required for this to work or can both the callbacks be in the same group? How do you make them run in parallel?

clalancette, 2018-09-11:
> @clalancette - I looked at your example and want to know if separate callback groups are required for this to work or can both the callbacks be in the same group? How do you make them run in parallel?

Both the separate callback groups and the multi-threaded executor is required.  If you just have separate callback groups, then the work gets put onto different queues, but there is only one executor available to dispatch work from queues.  If you just have a multi-threaded executor, then there is only a single queue where work gets put onto, and multiple threads just contend for the queue.  Using the combination of both of them allows work to get put onto separate queues and have multiple executors work off of those queues.  Does that make sense?

stonier, 2019-02-28:
> Does that make sense?

:+1:

jhdcs, 2019-08-29:
Good afternoon,

I believe I have an example that should work for this. See Pull Request #251 

Are there any changes that you would like to see?

clalancette, 2019-10-08:
Fixed by #251, so closing.

closed 2019-10-08 by clalancette