ros2/examples #209: subscriber_not_composable always throws an error after Ctrl-C

clalancette, 2018-06-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Bionic arm64
- Installation type:
  - Fat archive from 2018-06-21, downloaded from here: https://ci.ros2.org/view/packaging/job/packaging_linux-aarch64/475/artifact/ws/ros2-package-linux-aarch64.tar.bz2
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
$ ros2 run examples_rclcpp_minimal_subscriber subscriber_not_composable
```

Wait a few seconds, then press Ctrl-C gives:

```
^Csignal_handler(2)
[ERROR] [rclcpp]: Error in destruction of rcl subscription handle: the Node Handle was destructed too early. You will leak memory
[ERROR] [rclcpp]: Error in destruction of rcl subscription handle: the Node Handle was destructed too early. You will leak memory
```

#### Expected behavior
Program quits without error.

#### Actual behavior
Program quits with: `[ERROR] [rclcpp]: Error in destruction of rcl subscription handle: the Node Handle was destructed too early. You will leak memory`

#### Additional information
This bug may be in the underlying `rclcpp` or `rcl` layers, but since I wasn't sure, I started here.

dhood, 2018-06-21:
I have also seen this with a different executable using Connext. It's not every time so might be a race condition:

```

$ ros2 run  intra_process_demo image_pipeline_all_in_one
RTI Data Distribution Service EVAL License issued to OSRF dhood@osrfoundation.org For non-production use only.
Expires on 16-Jul-2018 See www.rti.com for more information.
RTI Data Distribution Service EVAL License issued to OSRF dhood@osrfoundation.org For non-production use only.
Expires on 16-Jul-2018 See www.rti.com for more information.
RTI Data Distribution Service EVAL License issued to OSRF dhood@osrfoundation.org For non-production use only.
Expires on 16-Jul-2018 See www.rti.com for more information.
^Csignal_handler(2)
PRESInterParticipant_removeRemoteParticipant:[Local Participant: ac1701a8 681b 1] [Remote Participant: 1c1 10f01a8 71680000 | Remote Endpoint: 0] Could not remove remote endpoint
PRESInterParticipant_removeRemoteParticipant:[Local Participant: ac1701a8 681b 1] [Remote Participant: 1c1 10f01a8 71680000 | Remote Endpoint: 0] Could not remove remote endpoint
PRESParticipant_removeRemoteParticipantFromBuiltinChannels:Failed to remove remote participant 10f01a8 71680000 0 to inter participant channel
PRESParticipant_removeRemoteParticipant:!remove remote participant from builtin channels
DISCParticipantDiscoveryPlugin_removeRemoteParticipantI:!remove remote participant: 0X10F01A8,0X71680000,0,0X1C1
DISCParticipantDiscoveryPlugin_removeRemoteParticipant:!remove remote participant: 0X10F01A8,0X71680000,0,0X1C1
DISCSimpleParticipantDiscoveryPluginReaderListener_onDataAvailable:!remove remote participant
[ERROR] [rclcpp]: Error in destruction of rcl subscription handle: the Node Handle was destructed too early. You will leak memory
```

dhood, 2018-06-22:
I see it reliably on xenial from-source for `ros2 run examples_rclcpp_minimal_subscriber subscriber_not_composable` with connext, opensplice and fastrtps.

@wjwwood that message was introduced in this PR (the behaviour, who knows): https://github.com/ros2/rclcpp/commit/9ce5aaa792e35c4d5d12ca24c0c88da804ecdb57
Can you comment on the severity? The error has been printed since that PR was merged (checked with a packaging build that it wasn't introduced later on).

wjwwood, 2018-06-22:
It's bad.

It because (for some reason) the subscription is getting destructed after the node. We should figure out why that's happening and see if we can fix it. I'm currently looking into too many things to help, but maybe tomorrow I could look into it.

clalancette, 2018-06-22:
@wjwwood And I figured out why the subscription is getting destructed after the node; it's because of https://github.com/ros2/examples/blob/master/rclcpp/minimal_subscriber/not_composable.cpp#L38, which forces the node to be destroyed before the subscription.  The easy solution would seem to be to remove line 38, but then I ran into https://github.com/eProsima/Fast-RTPS/issues/235.  A workaround for now would be to make the node scoped inside of main, which seems to work in my testing.  The downside is that:

* It really just hides the issue, since having a global node seems like it should be a valid use
* We have to duplicate the node name

Nonetheless, I will probably submit a change that scopes the node just to unblock us for now.  Note that this does not address the problem @dhood and I've seen in the image_pipeline; I'll debug that next.

clalancette, 2018-06-22:
Once I added some debugging to the `intra_process` demos, the problem went away.  @dhood is probably right that this is a race condition.  Still debugging.

clalancette, 2018-06-22:
There are really 2 separate problems here; the original that I reported (worked around in #210), and a racy problem in the intra_process_demo.  I'm going to leave this one to track the first problem, and open a new one to track the `intra_process_demo` problem.

clalancette, 2018-06-25:
I've merged https://github.com/ros2/examples/pull/210 for the workaround for this problem, but I'm actually going to leave this open to track the problem in https://github.com/eProsima/Fast-RTPS/issues/235

closed 2018-06-27 by wjwwood

ssnover95, 2019-06-12:
I tried removing nullptr assignments just now under Dashing and after Ctrl-C I found that the terminal hung until in another terminal I ran `ros2 node info /minimal_subscriber`, at which point the subscriber process crashed (in one of two ways):

```
ssnover@styrka:~/dashing$ build/examples_rclcpp_minimal_subscriber/subscriber_not_composable 
^C[INFO] [rclcpp]: signal_handler(signal_value=2)
Segmentation fault (core dumped)
ssnover@styrka:~/dashing$ build/examples_rclcpp_minimal_subscriber/subscriber_not_composable 
^C[INFO] [rclcpp]: signal_handler(signal_value=2)
double free or corruption (out)
Aborted (core dumped)
ssnover@styrka:~/dashing$ build/examples_rclcpp_minimal_subscriber/subscriber_not_composable 
^C[INFO] [rclcpp]: signal_handler(signal_value=2)
Segmentation fault (core dumped)
ssnover@styrka:~/dashing$ build/examples_rclcpp_minimal_subscriber/subscriber_not_composable 
^C[INFO] [rclcpp]: signal_handler(signal_value=2)
Segmentation fault (core dumped)
```

clalancette, 2019-06-12:
@ssnover95 This is a pretty old bug.  I'd suggest opening a new one with exact steps to reproduce, and we can debug from there.

ssnover95, 2019-06-14:
It's an old bug that was never fixed, just worked around. All I did was note the comment in the source for that node and take out the two lines assigning nullptr. I think it's better to re-open this issue, keeping the history, rather than create a completely new one.