ros2/examples #223: [launch] Propose launch-based testing API through example

hidmic, 2019-01-14:
## Feature request

Propose an API for launch-based testing through user stories.

#### Feature description

Write down testing API use examples to:

- Test an existing launch file through composition.
- Run a gtest/pytest program in test fixture context.

In addition, all legacy launch testing use cases, as can be found in [ros2/launch](https://github.com/ros2/launch) and [ros2/demos](https://github.com/ros2/demos), must be captured.

hidmic, 2019-01-17:
I believe this isn't the most appropriate place for these examples to live. I'm already PR'ing against ros2/launch. Any thoughts against moving this issue over there too?

wjwwood, 2019-01-19:
I can transfer it to launch.