ros2/examples #228: "client_main" and "service_main" failed (hang up) 3/10 times

sharronliu, 2019-02-15:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 18.04
- Installation type:
  Binaries
- Version or commit hash:
  ROS2 Crystal Release
- DDS implementation:
  the default one
- Client library (if applicable):
  rclcpp

#### Steps to reproduce issue
from terminal #1
`ros2 run examples_rclcpp_minimal_service service_main`
from terminal #2, repeat
`ros2 run examples_rclcpp_minimal_client client_main`

#### Expected behavior
On each running of the `client_main`, the terminal #2 supposed to print
`[INFO] [minimal_client]: result of 41 + 1 = 42`

#### Actual behavior
Out of 10 running, only 7 success.
3 times `client_main` hang up, **no output** from client_main, **nor output** from `service_main`.
Once hand up, only `ctrl+c` can exit the `client_main`.

#### Additional information
$ ros2 run examples_rclcpp_minimal_client client_main 
[INFO] [minimal_client]: result of 41 + 1 = 42
$ ros2 run examples_rclcpp_minimal_client client_main 
^C[INFO] [rclcpp]: signal_handler(signal_value=2)
[ERROR] [minimal_client]: service call failed :(
$ ros2 run examples_rclcpp_minimal_client client_main 
[INFO] [minimal_client]: result of 41 + 1 = 42
$ ros2 run examples_rclcpp_minimal_client client_main 
[INFO] [minimal_client]: result of 41 + 1 = 42
$ ros2 run examples_rclcpp_minimal_client client_main 
[INFO] [minimal_client]: result of 41 + 1 = 42
$ ros2 run examples_rclcpp_minimal_client client_main 
[INFO] [minimal_client]: result of 41 + 1 = 42
$ ros2 run examples_rclcpp_minimal_client client_main 
^C[INFO] [rclcpp]: signal_handler(signal_value=2)
[ERROR] [minimal_client]: service call failed :(
$ ros2 run examples_rclcpp_minimal_client client_main 
[INFO] [minimal_client]: result of 41 + 1 = 42
$ ros2 run examples_rclcpp_minimal_client client_main 
^C[INFO] [rclcpp]: signal_handler(signal_value=2)
[ERROR] [minimal_client]: service call failed :(
$ ros2 run examples_rclcpp_minimal_client client_main 
[INFO] [minimal_client]: result of 41 + 1 = 42

jacobperron, 2019-02-15:
This has been also reported with the services in the demos repo: ros2/demos#304

sharronliu, 2019-02-18:
Thanks for linking these two.

closed 2019-02-19 by jacobperron