ros2/examples #231: Can not build example using master branch

NovoG93, 2019-03-19:
## Bug report
Wanted to try ROS2 for the first time, tried the example but can not build using branch master. If I checkout origin/crystal I can build the examples.

**Required Info:**

- Operating System:
  - OS and version: Ubuntu 18.04 (VM using VirtualBox with Windows 10 as Host)
- Installation type:
  -  binaries
- Version or commit hash:
  - f4d4b14bbcb257eae4d46ef06ecf66db025580ae
- DDS implementation:
  - FastRTPS 

#### Steps to reproduce issue
1) Fresh install of ROS2 Crystal
2) Create workspace
3) clone repo 
4) Build
``` 
cd <path_to_ws>
git clone https://github.com/ros2/examples src/examples
colcon build --symlink-install
``` 
#### Expected behavior
``` 
georg@georg-ROS2:~/ros_ws$ colcon build --symlink-install
Starting >>> examples_rclcpp_minimal_action_client
Starting >>> examples_rclcpp_minimal_action_server
Finished <<< examples_rclcpp_minimal_action_server [6.46s]                                     
Starting >>> examples_rclcpp_minimal_client
Finished <<< examples_rclcpp_minimal_action_client [10.8s]                                                                               
Starting >>> examples_rclcpp_minimal_composition
Finished <<< examples_rclcpp_minimal_client [5.36s]                                                                                        
Starting >>> examples_rclcpp_minimal_publisher
Finished <<< examples_rclcpp_minimal_publisher [13.2s]                                                                                     
Starting >>> examples_rclcpp_minimal_service
Finished <<< examples_rclcpp_minimal_composition [14.5s]                                          
Starting >>> examples_rclcpp_minimal_subscriber
Finished <<< examples_rclcpp_minimal_service [6.98s]                                                                                      
Starting >>> examples_rclcpp_minimal_timer
Finished <<< examples_rclcpp_minimal_subscriber [12.3s]                                                                                    
Starting >>> examples_rclpy_executors
Finished <<< examples_rclcpp_minimal_timer [6.32s]                                                                       
Starting >>> examples_rclpy_minimal_action_client        
Finished <<< examples_rclpy_executors [1.26s]                                                                      
Starting >>> examples_rclpy_minimal_action_server
Finished <<< examples_rclpy_minimal_action_client [1.00s]                                                                      
Starting >>> examples_rclpy_minimal_client
Finished <<< examples_rclpy_minimal_action_server [1.09s]                                                                
Starting >>> examples_rclpy_minimal_publisher
Finished <<< examples_rclpy_minimal_client [1.05s]                                                                   
Starting >>> examples_rclpy_minimal_service
Finished <<< examples_rclpy_minimal_publisher [1.10s]                                                                 
Starting >>> examples_rclpy_minimal_subscriber
Finished <<< examples_rclpy_minimal_service [1.28s]                                                                    
Finished <<< examples_rclpy_minimal_subscriber [1.18s]             

Summary: 15 packages finished [42.4s]
``` 

#### Actual behavior
``` 
georg@georg-ROS2:~/ros_ws$ colcon build --symlink-install
Starting >>> examples_rclcpp_minimal_action_client
Starting >>> examples_rclcpp_minimal_action_server
--- stderr: examples_rclcpp_minimal_action_client                                              
/home/georg/ros_ws/src/examples/rclcpp/minimal_action_client/not_composable_with_cancel.cpp: In function ‘int main(int, char**)’:
/home/georg/ros_ws/src/examples/rclcpp/minimal_action_client/not_composable_with_cancel.cpp:93:47: error: ‘WrappedResult’ is not a member of ‘rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci>’
   rclcpp_action::ClientGoalHandle<Fibonacci>::WrappedResult wrapped_result = result_future.get();
                                               ^~~~~~~~~~~~~
/home/georg/ros_ws/src/examples/rclcpp/minimal_action_client/not_composable_with_cancel.cpp:94:10: error: ‘wrapped_result’ was not declared in this scope
   switch(wrapped_result.code) {
          ^~~~~~~~~~~~~~
/home/georg/ros_ws/src/examples/rclcpp/minimal_action_client/not_composable_with_cancel.cpp:94:10: note: suggested alternative: ‘wait_result’
   switch(wrapped_result.code) {
          ^~~~~~~~~~~~~~
          wait_result
/home/georg/ros_ws/src/examples/rclcpp/minimal_action_client/not_composable_with_cancel.cpp:95:37: error: could not convert ‘SUCCEEDED’ from ‘rclcpp_action::ResultCode’ to ‘<type error>’
     case rclcpp_action::ResultCode::SUCCEEDED:
                                     ^~~~~~~~~
/home/georg/ros_ws/src/examples/rclcpp/minimal_action_client/not_composable_with_cancel.cpp:97:37: error: could not convert ‘ABORTED’ from ‘rclcpp_action::ResultCode’ to ‘<type error>’
     case rclcpp_action::ResultCode::ABORTED:
                                     ^~~~~~~
/home/georg/ros_ws/src/examples/rclcpp/minimal_action_client/not_composable_with_cancel.cpp:100:37: error: could not convert ‘CANCELED’ from ‘rclcpp_action::ResultCode’ to ‘<type error>’
     case rclcpp_action::ResultCode::CANCELED:
                                     ^~~~~~~~
/home/georg/ros_ws/src/examples/rclcpp/minimal_action_client/not_composable.cpp: In function ‘int main(int, char**)’:
/home/georg/ros_ws/src/examples/rclcpp/minimal_action_client/not_composable.cpp:67:47: error: ‘WrappedResult’ is not a member of ‘rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci>’
   rclcpp_action::ClientGoalHandle<Fibonacci>::WrappedResult wrapped_result = result_future.get();
                                               ^~~~~~~~~~~~~
/home/georg/ros_ws/src/examples/rclcpp/minimal_action_client/not_composable.cpp:69:10: error: ‘wrapped_result’ was not declared in this scope
   switch(wrapped_result.code) {
          ^~~~~~~~~~~~~~
/home/georg/ros_ws/src/examples/rclcpp/minimal_action_client/not_composable.cpp:69:10: note: suggested alternative: ‘__codecvt_result’
   switch(wrapped_result.code) {
          ^~~~~~~~~~~~~~
          __codecvt_result
/home/georg/ros_ws/src/examples/rclcpp/minimal_action_client/not_composable.cpp:70:37: error: could not convert ‘SUCCEEDED’ from ‘rclcpp_action::ResultCode’ to ‘<type error>’
     case rclcpp_action::ResultCode::SUCCEEDED:
                                     ^~~~~~~~~
/home/georg/ros_ws/src/examples/rclcpp/minimal_action_client/not_composable.cpp:72:37: error: could not convert ‘ABORTED’ from ‘rclcpp_action::ResultCode’ to ‘<type error>’
     case rclcpp_action::ResultCode::ABORTED:
                                     ^~~~~~~
/home/georg/ros_ws/src/examples/rclcpp/minimal_action_client/not_composable.cpp:75:37: error: could not convert ‘CANCELED’ from ‘rclcpp_action::ResultCode’ to ‘<type error>’
     case rclcpp_action::ResultCode::CANCELED:
                                     ^~~~~~~~
/home/georg/ros_ws/src/examples/rclcpp/minimal_action_client/not_composable.cpp:84:22: error: ‘wrapped_result’ was not declared in this scope
   for (auto number : wrapped_result.result->sequence)
                      ^~~~~~~~~~~~~~
/home/georg/ros_ws/src/examples/rclcpp/minimal_action_client/not_composable.cpp:84:22: note: suggested alternative: ‘__codecvt_result’
   for (auto number : wrapped_result.result->sequence)
                      ^~~~~~~~~~~~~~
                      __codecvt_result
make[2]: *** [CMakeFiles/action_client_not_composable.dir/not_composable.cpp.o] Error 1
make[1]: *** [CMakeFiles/action_client_not_composable.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/action_client_not_composable_with_cancel.dir/not_composable_with_cancel.cpp.o] Error 1
make[1]: *** [CMakeFiles/action_client_not_composable_with_cancel.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< examples_rclcpp_minimal_action_client	[ Exited with code 2 ]
Aborted  <<< examples_rclcpp_minimal_action_server                                         

Summary: 0 packages finished [6.88s]
  1 package failed: examples_rclcpp_minimal_action_client
  1 package aborted: examples_rclcpp_minimal_action_server
  1 package had stderr output: examples_rclcpp_minimal_action_client
  13 packages not processed
``` 

#### Additional information

Expected behavior is taken using commit 62b2132f8e03b007aeb38202ce49583c331561b0

sloretz, 2019-03-19:
Hi @NovoG93, thanks for taking the time to make an issue. Please build the `crystal` branch of this repo instead. `master` branch is where development for the next ROS release (`dashing`) happens.

closed 2019-03-19 by sloretz

rukie, 2019-04-24:
It would be nice if a line was added to the Colcon build page that explicitly stated:
```
git checkout ROSDISTROHERE
```
on 
https://index.ros.org/doc/ros2/Tutorials/Colcon-Tutorial/

As I also missed the note while following commands.

sloretz, 2019-04-24:
@rukie Would you be willing to open a PR on https://github.com/ros2/ros2_documentation with the change you're proposing?

https://github.com/ros2/ros2_documentation/blob/master/source/Tutorials/Colcon-Tutorial.rst

rukie, 2019-04-24:
Sure thing, @sloretz !
I just need to figure out how to do a pull request 👍 
https://github.com/rukie/ros2_documentation/blob/master/source/Tutorials/Colcon-Tutorial.rst

sloretz, 2019-04-24:
@rukie Thanks!, These links may help.

The first is how to fork a repo. Forking gives you a copy you can make changes to. https://help.github.com/en/articles/fork-a-repo

Second is how to open a pull request. A PR is how changes are contributed back to the original repo. https://help.github.com/en/articles/creating-a-pull-request-from-a-fork