ros2/examples #233: Cannot build (rclcpp) example using crystal branch

miguelvelezmj25, 2019-03-28:
## Bug report

**Required Info:**

- Operating System: 
  -  Ubuntu 18.04.2 LTS
- Installation type:
  -  binaries
- Version or commit hash:
  -  Crystal (HEAD)
- DDS implementation:
  -  FastRTPS

#### Steps to reproduce issue

```
1. Fresh binary install of ROS2 Crystal
2. Fresh binary install of ROS1 Melodic
3. Fresh install ros1-bridge
4. Fresh install rosbag2
5. Follow instructions of [Using Colcon to build packages](https://index.ros.org/doc/ros2/Tutorials/Colcon-Tutorial/)
6. Create workspace
7. Clone repo
8. Build
```

#### Expected behavior

Built workspace

#### Actual behavior

```
mvelezce@config-crusher:~/Documents/programming/ros2/projects/ros2_example_ws$ colcon build --symlink-install
Starting >>> examples_rclcpp_minimal_action_client
Starting >>> examples_rclcpp_minimal_action_server
Starting >>> examples_rclcpp_minimal_client
Starting >>> examples_rclcpp_minimal_composition
Starting >>> examples_rclcpp_minimal_publisher
Starting >>> examples_rclcpp_minimal_service
Starting >>> examples_rclcpp_minimal_subscriber
Starting >>> examples_rclcpp_minimal_timer
--- stderr: examples_rclcpp_minimal_timer                                                                                                                                                                 
/opt/ros/crystal/lib/librclcpp.so: undefined reference to `std::thread::_State::~_State()@GLIBCXX_3.4.22'
/opt/ros/crystal/lib/librclcpp.so: undefined reference to `__cxa_init_primary_exception@CXXABI_1.3.11'
/opt/ros/crystal/lib/librclcpp.so: undefined reference to `typeinfo for std::thread::_State@GLIBCXX_3.4.22'
/opt/ros/crystal/lib/librclcpp.so: undefined reference to `std::thread::_M_start_thread(std::unique_ptr<std::thread::_State, std::default_delete<std::thread::_State> >, void (*)())@GLIBCXX_3.4.22'
/opt/ros/crystal/lib/librclcpp.so: undefined reference to `std::__exception_ptr::exception_ptr::exception_ptr(void*)@CXXABI_1.3.11'
collect2: error: ld returned 1 exit status
make[2]: *** [timer_lambda] Error 1
make[1]: *** [CMakeFiles/timer_lambda.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/opt/ros/crystal/lib/librclcpp.so: undefined reference to `std::thread::_State::~_State()@GLIBCXX_3.4.22'
/opt/ros/crystal/lib/librclcpp.so: undefined reference to `__cxa_init_primary_exception@CXXABI_1.3.11'
/opt/ros/crystal/lib/librclcpp.so: undefined reference to `typeinfo for std::thread::_State@GLIBCXX_3.4.22'
/opt/ros/crystal/lib/librclcpp.so: undefined reference to `std::thread::_M_start_thread(std::unique_ptr<std::thread::_State, std::default_delete<std::thread::_State> >, void (*)())@GLIBCXX_3.4.22'
/opt/ros/crystal/lib/librclcpp.so: undefined reference to `std::__exception_ptr::exception_ptr::exception_ptr(void*)@CXXABI_1.3.11'
collect2: error: ld returned 1 exit status
make[2]: *** [timer_member_function] Error 1
make[1]: *** [CMakeFiles/timer_member_function.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< examples_rclcpp_minimal_timer	[ Exited with code 2 ]
--- stderr: examples_rclcpp_minimal_service                                                                                                                                                                          
/opt/ros/crystal/lib/librclcpp.so: undefined reference to `std::thread::_State::~_State()@GLIBCXX_3.4.22'
/opt/ros/crystal/lib/librclcpp.so: undefined reference to `__cxa_init_primary_exception@CXXABI_1.3.11'
/opt/ros/crystal/lib/librclcpp.so: undefined reference to `typeinfo for std::thread::_State@GLIBCXX_3.4.22'
/opt/ros/crystal/lib/librclcpp.so: undefined reference to `std::thread::_M_start_thread(std::unique_ptr<std::thread::_State, std::default_delete<std::thread::_State> >, void (*)())@GLIBCXX_3.4.22'
/opt/ros/crystal/lib/librclcpp.so: undefined reference to `std::__exception_ptr::exception_ptr::exception_ptr(void*)@CXXABI_1.3.11'
collect2: error: ld returned 1 exit status
[... 153 lines cut ...]
#### Additional information
Removing ```rclcpp``` directory builds the workspace without any issues.

```
mvelezce@config-crusher:~/Documents/programming/ros2/projects/ros2_example_ws$ colcon build --symlink-install
Starting >>> examples_rclpy_executors
Starting >>> examples_rclpy_minimal_action_client
Starting >>> examples_rclpy_minimal_action_server
Starting >>> examples_rclpy_minimal_client
Starting >>> examples_rclpy_minimal_publisher
Starting >>> examples_rclpy_minimal_service
Starting >>> examples_rclpy_minimal_subscriber
Finished <<< examples_rclpy_executors [2.21s]                                                                                                                                                             
Finished <<< examples_rclpy_minimal_action_client [2.16s]                                                                                                                                                         
Finished <<< examples_rclpy_minimal_action_server [2.16s]
Finished <<< examples_rclpy_minimal_client [2.16s]
Finished <<< examples_rclpy_minimal_publisher [2.17s]
Finished <<< examples_rclpy_minimal_service [2.17s]
Finished <<< examples_rclpy_minimal_subscriber [2.44s]          

Summary: 7 packages finished [2.70s]
```

----
## Feature request

nuclearsandwich, 2019-03-28:
@miguelvelezmj25 did you check out the `crystal` branch as suggested in the tutorial?

miguelvelezmj25, 2019-03-28:
@nuclearsandwich Yes, I did.

```
mvelezce@config-crusher:~/Documents/programming/ros2/projects/ros2_example_ws/src/examples$ git status
On branch crystal
Your branch is up to date with 'origin/crystal'.

nothing to commit, working tree clean
```

wjwwood, 2019-03-28:
It looks like you might be mixing clang and gcc or different versions of the one of them. Are you using the default compiler in 18.04 that our binaries were built with?

nuclearsandwich, 2019-03-28:
>> did you check out the crystal branch as suggested in the tutorial?
> Yes, I did.

Thanks for confirming. I wasn't able to reproduce your error on on 18.04 using the latest Crystal debs and the crystal branch of ros2/examples.

In addition to @wjwwood's questions above, can you please share more about your environment.

miguelvelezmj25, 2019-03-28:
Yes, you were right. It was using the homebrew gcc

```
mvelezce@config-crusher:~$ gcc --version
gcc (Homebrew gcc 5.5.0_4) 5.5.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

instead of the default compiler. 

```
mvelezce@config-crusher:~$ /usr/bin/gcc --version
gcc (Ubuntu 7.3.0-27ubuntu1~18.04) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I just removed the homebrew compiler and it worked as expected. Thank you!

wjwwood, 2019-03-28:
I'm going to close this because I believe the issue has been addressed. Feel free to continue to comment and I can reopen if necessary.

closed 2019-03-28 by wjwwood

dirk-thomas, 2019-03-28:
@miguelvelezmj25 If your question / problem has been resolved please go ahead and close the ticket.

shanpenghui, 2019-06-04:
After checkout the crystal branch compile successfully.