ros2/examples #235: Action client crashes "no attribute 'sequence'"

sloretz, 2019-04-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - <source
- Version or commit hash:
  -  master
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

In one terminal start the server

```
ros2 run examples_rclpy_minimal_action_server server
```

In another terminal start the client

```
ros2 run examples_rclpy_minimal_action_client client
```

#### Expected behavior

Client prints feedback as it receives it

#### Actual behavior

Client crashes

```
$ ros2 run examples_rclpy_minimal_action_client client
[INFO] [minimal_action_client]: Waiting for action server...
[INFO] [minimal_action_client]: Sending goal request...
[INFO] [minimal_action_client]: Goal accepted :)
Traceback (most recent call last):
  File "/home/sloretz/ws/ros2/install/examples_rclpy_minimal_action_client/lib/examples_rclpy_minimal_action_client/client", line 11, in <module>
    load_entry_point('examples-rclpy-minimal-action-client', 'console_scripts', 'client')()
  File "/home/sloretz/ws/ros2/build/examples_rclpy_minimal_action_client/examples_rclpy_minimal_action_client/client.py", line 76, in main
    rclpy.spin(action_client)
  File "/home/sloretz/ws/ros2/install/rclpy/lib/python3.6/site-packages/rclpy/__init__.py", line 178, in spin
    executor.spin_once()
  File "/home/sloretz/ws/ros2/install/rclpy/lib/python3.6/site-packages/rclpy/executors.py", line 605, in spin_once
    raise handler.exception()
  File "/home/sloretz/ws/ros2/install/rclpy/lib/python3.6/site-packages/rclpy/task.py", line 206, in __call__
    self._handler.send(None)
  File "/home/sloretz/ws/ros2/install/rclpy/lib/python3.6/site-packages/rclpy/executors.py", line 351, in handler
    await call_coroutine(entity, arg)
  File "/home/sloretz/ws/ros2/install/rclpy/lib/python3.6/site-packages/rclpy/executors.py", line 317, in _execute_waitable
    await waitable.execute(data)
  File "/home/sloretz/ws/ros2/install/rclpy/lib/python3.6/site-packages/rclpy/action/client.py", line 306, in execute
    await await_or_execute(self._feedback_callbacks[goal_uuid], feedback_msg)
  File "/home/sloretz/ws/ros2/install/rclpy/lib/python3.6/site-packages/rclpy/executors.py", line 119, in await_or_execute
    return callback(*args)
  File "/home/sloretz/ws/ros2/build/examples_rclpy_minimal_action_client/examples_rclpy_minimal_action_client/client.py", line 41, in feedback_callback
    self.get_logger().info('Received feedback: {0}'.format(feedback.sequence))
AttributeError: 'Fibonacci_FeedbackMessage' object has no attribute 'sequence'
```

jacobperron, 2019-04-15:
This affects all the examples. I'm guessing it is a regression following https://github.com/ros2/rosidl_python/pull/35, but I'll look into it.

dirk-thomas, 2019-04-15:
> I'm guessing it is a regression following ros2/rosidl_python#35, but I'll look into it.

The reference PR only changes the type of attributes but doesn't add/remove any attributes.

PR #236 (2019-04-16): Fix rclpy action client examples
Insert missing nested field when accessing feedback and result.

Fixes #235
closed 2019-04-16 by jacobperron