ros2/examples #252: Issue building example

AlexandreBorowczyk, 2019-09-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Installing ROS2 via Debian Packages
- Version or commit hash:
  - dashing (010ed35)
- DDS implementation:
  -  Fast-RTPS,
- Client library (if applicable):
  - N/A

Hi,

I've followed the installation procedure then the Using colcon to build packages tutorial.
However at build step (colcon build --symlink-install) I'm getting the following:

`Starting >>> examples_rclcpp_minimal_action_client
Starting >>> examples_rclcpp_minimal_action_server
Starting >>> examples_rclcpp_minimal_client
Starting >>> examples_rclcpp_minimal_composition
Starting >>> examples_rclcpp_minimal_publisher
Starting >>> examples_rclcpp_minimal_service
Starting >>> examples_rclcpp_minimal_subscriber
Starting >>> examples_rclcpp_minimal_timer
--- stderr: examples_rclcpp_minimal_client                                                                                                      
Traceback (most recent call last):
  File "/opt/ros/dashing/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 21, in <module>
    from catkin_pkg.package import parse_package_string
ModuleNotFoundError: No module named 'catkin_pkg'
CMake Error at /opt/ros/dashing/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:94 (message):
  execute_process(/home/linuxbrew/.linuxbrew/bin/python3
  /opt/ros/dashing/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py
  /home/alexandreborowczyk/ros2_example_ws/src/examples/rclcpp/minimal_client/package.xml
  /home/alexandreborowczyk/ros2_example_ws/build/examples_rclcpp_minimal_client/ament_cmake_core/package.cmake)
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/dashing/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:49 (_ament_package_xml)
  /opt/ros/dashing/share/ament_cmake_core/cmake/core/ament_package.cmake:63 (ament_package_xml)
  CMakeLists.txt:23 (ament_package)

---
Failed   <<< examples_rclcpp_minimal_client	[ Exited with code 1 ]
--- stderr: examples_rclcpp_minimal_timer
Traceback (most recent call last):
  File "/opt/ros/dashing/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 21, in <module>
    from catkin_pkg.package import parse_package_string
ModuleNotFoundError: No module named 'catkin_pkg'
CMake Error at /opt/ros/dashing/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:94 (message):
  execute_process(/home/linuxbrew/.linuxbrew/bin/python3
  /opt/ros/dashing/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py
  /home/alexandreborowczyk/ros2_example_ws/src/examples/rclcpp/minimal_timer/package.xml
  /home/alexandreborowczyk/ros2_example_ws/build/examples_rclcpp_minimal_timer/ament_cmake_core/package.cmake)
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/dashing/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:49 (_ament_package_xml)
  /opt/ros/dashing/share/ament_cmake_core/cmake/core/ament_package.cmake:63 (ament_package_xml)
  CMakeLists.txt:28 (ament_package)

---
Aborted  <<< examples_rclcpp_minimal_timer
--- stderr: examples_rclcpp_minimal_publisher
Traceback (most recent call last):
  File "/opt/ros/dashing/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 21, in <module>
    from catkin_pkg.package import parse_package_string
ModuleNotFoundError: No module named 'catkin_pkg'
CMake Error at /opt/ros/dashing/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:94 (message):
[... 84 lines cut ...]
--- stderr: examples_rclcpp_minimal_action_client                                     
Traceback (most recent call last):
  File "/opt/ros/dashing/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 21, in <module>
    from catkin_pkg.package import parse_package_string
ModuleNotFoundError: No module named 'catkin_pkg'
CMake Error at /opt/ros/dashing/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:94 (message):
  execute_process(/home/linuxbrew/.linuxbrew/bin/python3
  /opt/ros/dashing/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py
  /home/alexandreborowczyk/ros2_example_ws/src/examples/rclcpp/minimal_action_client/package.xml
  /home/alexandreborowczyk/ros2_example_ws/build/examples_rclcpp_minimal_action_client/ament_cmake_core/package.cmake)
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/dashing/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:49 (_ament_package_xml)
  /opt/ros/dashing/share/ament_cmake_core/cmake/core/ament_package.cmake:63 (ament_package_xml)
  CMakeLists.txt:49 (ament_package)

---
Aborted  <<< examples_rclcpp_minimal_action_client

Summary: 0 packages finished [0.99s]
  1 package failed: examples_rclcpp_minimal_client
  7 packages aborted: examples_rclcpp_minimal_action_client examples_rclcpp_minimal_action_server examples_rclcpp_minimal_composition examples_rclcpp_minimal_publisher examples_rclcpp_minimal_service examples_rclcpp_minimal_subscriber examples_rclcpp_minimal_timer
  8 packages had stderr output: examples_rclcpp_minimal_action_client examples_rclcpp_minimal_action_server examples_rclcpp_minimal_client examples_rclcpp_minimal_composition examples_rclcpp_minimal_publisher examples_rclcpp_minimal_service examples_rclcpp_minimal_subscriber examples_rclcpp_minimal_timer
  7 packages not processed
`

dirk-thomas, 2019-09-17:
As the error message says: `catkin_pkg` is not found. You need to install `python3-catkin-pkg-modules` as described in the install instructions.

closed 2019-09-17 by dirk-thomas

tealeeseng, 2020-03-07:
The hint can be found from this line in the log: execute_process(/home/linuxbrew/.linuxbrew/bin/python3

Is that the right python3? you may not want to go into a particular virtual env.

colcon build --cmake-clean-cache # reset python3 env
colcon build --symlink-install # rebuild

This helps for my case.