ros2/examples #260: minimal_composition build error

TimCraig, 2020-02-27:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - Everything was updated 2/26/20
- Operating System:
  Ubuntu 18.04
- Installation type:
  - binaries
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp, ament_cmake

#### Steps to reproduce issue
When doing a colcon build on the minimal_composition example, I get a build error that seems to point to ament_cmake not finding a package catkin_pkg.

Here is the full error report:

--- stderr: minimal_composition                          
Traceback (most recent call last):
  File "/opt/ros/eloquent/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 21, in <module>
    from catkin_pkg.package import parse_package_string
ModuleNotFoundError: No module named 'catkin_pkg'
CMake Error at /opt/ros/eloquent/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:94 (message):
  execute_process(/home/brutusdev/anaconda3/bin/python3
  /opt/ros/eloquent/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py
  /home/brutusdev/ros2/druai_ws/src/ROS2-Workspace/minimal_composition/package.xml
  /home/brutusdev/ros2/druai_ws/build/minimal_composition/ament_cmake_core/package.cmake)
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/eloquent/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:49 (_ament_package_xml)
  /opt/ros/eloquent/share/ament_cmake_core/cmake/core/ament_package.cmake:63 (ament_package_xml)
  CMakeLists.txt:61 (ament_package)

---
Failed   <<< minimal_composition	[ Exited with code 1 ]

```

```

dirk-thomas, 2020-02-27:
You are using `anaconda`and likely haven't installed the Python package `catkin_pkg` with ` conda`.

TimCraig, 2020-02-27:
@dirk-thomas, installing it with Anaconda handled it.  Thanks.

closed 2020-02-27 by TimCraig