ros2/examples #266: Example using spin_some might throw exception

hdino, 2020-04-22:
The examples in this repository are besides the [ROS2 tutorials](https://index.ros.org/doc/ros2/Tutorials/#tutorials) the first place to learn ROS2. Therefore, many public repositories, for instance of universities or research institutes, contain code that uses exactly the same procedures as shown here.

The [`minimal_publisher` example](https://github.com/ros2/examples/blob/master/rclcpp/topics/minimal_publisher/not_composable.cpp#L40) shows how to use `rclcpp::spin_some`. However, as pointed out in ros2/rclcpp#1066, `rclcpp::spin_some` will throw an exception if the program received a SIGINT and processed the shutdown after checking `rclcpp::ok()`, but before constructing the `SingleThreadedExecutor` in `rclcpp::spin_some`. This behaviour can easily be triggered by:
```c++
while (rclcpp::ok())
{
    std::this_thread::sleep_for(std::chrono::seconds(1));
    rclcpp::spin_some(ros_node);
}
```
@fujitatomoya created a [working example](https://github.com/fujitatomoya/examples/blob/topic-20200417-exception-after-sigin-1066/rclcpp/minimal_publisher/not_composable.cpp#L40-L41).

In ros2/rclcpp#1066 we already discussed some possible solutions, but we thought it's better to discuss this issue here. What are your opinions?

fujitatomoya, 2020-04-22:
sugest to put spin_some in a try-catch block in example code, i guess.

fujitatomoya, 2020-04-24:
@hdino 

are you gonna make a PR on this?

hidmic, 2020-05-07:
@hdino friendly ping.

fujitatomoya, 2020-05-14:
@hdino @hidmic 

could you take a look at https://github.com/ros2/examples/pull/270?

closed 2020-05-28 by jacobperron

luym11, 2021-06-24:
I also recently have the issue of https://github.com/ros2/rclcpp/issues/1066, and the fix I found out is something similar to https://github.com/luym11/examples/commit/26dfef3209d985c0bcc5e5cff435f43e84a0d658. 
Is this the correct way to do this? I thought that ROS should already have a signal handler such that I do not need to override it like this. 
Thanks!