ros2/examples #273: Build problem when running colcon for the first time

luccosta, 2020-05-24:
## System info

- Operating System: Ubuntu 18.04
- Installation type: From source
- Version or commit hash: 5d26b3b222d74852cca0d285fd31fe2c786cf181 / Release 0.9.0
- Client library: rclcpp

#### Steps to reproduce issue

I'm using ADE environment, and used the commands:

```  
mkdir -p ~/ros2_example_ws/src
cd ~/ros2_example_ws/
git clone https://github.com/ros2/examles src/examples 
source /opt/ros/dashing/setup.bash
colcon build --symlink-install
```
And had the following result:
```
/home/lucas/ros2_example_ws/src/examples/rclcpp/services/minimal_client/main.cpp:40:13: error: ‘rclcpp::FutureReturnCode’ has not been declared
     rclcpp::FutureReturnCode::SUCCESS)
```

luccosta, 2020-05-24:
I switched to release 0.8.0 and it worked.

clalancette, 2020-05-26:
Yes, using version 0.9.0 with dashing isn't going to work; the code is too new.  In general, you have to stick with the tags/branches associated with the releases for things to work.  In this case, that would be what is listed in https://github.com/ros2/ros2/blob/dashing/ros2.repos .

I'm going to close this out for now, but feel free to keep commenting or reopen if you think there is another bug here.

closed 2020-05-26 by clalancette