ros2/examples #274: Build error : ‘rclcpp::FutureReturnCode’ has not been declared

doosan-robotics, 2020-05-27:
## Bug report
Build error :‘rclcpp::FutureReturnCode’ has not been declared rclcpp::FutureReturnCode::SUCCESS)

The source codes  must be changed...  
rclcpp::FutureReturnCode::SUCCESS -> rclcpp::executor::FutureReturnCode::SUCCESS

Please check that! 

**Required Info:**

- Operating System:
  - Ubuntu 18.04, ROS2 eloquent
- Installation type:
  - binaries
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

christophebedard, 2020-05-27:
Assuming you're compiling the `examples` packages from source, and since you're using Eloquent, make sure you're using the `eloquent` branch. See https://github.com/ros2/examples/issues/273#issuecomment-633994583

clalancette, 2020-05-27:
Yes, what @christophebedard .  Also please fill out the issue form in the entirety; it gives us the information we need to determine whether this is the problem that you are running into.  Finally, for questions like this, please consider searching on https://answers.ros.org first, and if you can't find it, post your question there.

I'm going to close this out.

closed 2020-05-27 by clalancette

doosan-robotics, 2020-05-28:
The eloquent branch source builds fine without modification.
However, the example sources for the master and eloquent branches are slightly different. 
In my eloquent environment, I want to build the master source and needed the above modifications. Thank you.

newyeo, 2021-09-01:
This issue still exist when compile the example code from https://docs.ros.org/en/foxy/Tutorials/Writing-A-Simple-Cpp-Service-And-Client.html using eloquent. It can be solved by adding namespace. That means replace the `rclcpp::FutureReturnCode::SUCCESS` with `rclcpp::executor::FutureReturnCode::SUCCESS`.

aprotyas, 2021-09-01:
> This issue still exist when compile the example code from https://docs.ros.org/en/foxy/Tutorials/Writing-A-Simple-Cpp-Service-And-Client.html using eloquent. It can be solved by adding namespace. [...]

You're looking at the wrong distribution. Please refer to the example code for eloquent instead: https://docs.ros.org/en/eloquent/Tutorials/Writing-A-Simple-Cpp-Service-And-Client.html

newyeo, 2021-09-09:
Ah, I went through the wrong version.  Thanks very much for your reminding!