ros2/examples #278: Warning when using colcon test

snehaldb90, 2020-06-16:
## Bug report

**Required Info:**

- Operating System:
  - MacOS 10.14.5 (Mojave)
- Installation type:
  - Binaries
- Version or commit hash:
  - Foxy
- DDS implementation:
  - CycloneDDS

#### Steps to reproduce issue
- Build test workspace
  - source ros2_foxy/ros2-osx/setup.bash (underlay)
  - mkdir -p ros2_foxy/ros2-osx/dev_ws/src/ros2_example_ws/src
  - Clone this repo https://github.com/ros2/examples src/examples into ros2_example_ws/src
  - Navigate to examples/ directory and git checkout $ROS_DISTRO
  - Navigate to ros2_example_ws directory and colcon build --symlink-install

- Run Test
  - colcon test

#### Expected behavior
All 16 packages should pass.

#### Actual behavior
--- stderr: examples_rclpy_minimal_action_client                                                                                                                                                           
=============================== warnings summary ===============================
/usr/local/lib/python3.8/site-packages/_pytest/junitxml.py:417
  Warning: The 'junit_family' default value will change to 'xunit2' in pytest 6.0.
  Add 'junit_family=xunit1' to your pytest.ini file to keep the current format in future versions of pytest and silence this warning.

-- Docs: https://docs.pytest.org/en/latest/warnings.html

Finished <<< examples_rclpy_minimal_action_client [5.48s]
--- stderr: examples_rclpy_executors

=============================== warnings summary ===============================
/usr/local/lib/python3.8/site-packages/_pytest/junitxml.py:417
  Warning: The 'junit_family' default value will change to 'xunit2' in pytest 6.0.
  Add 'junit_family=xunit1' to your pytest.ini file to keep the current format in future versions of pytest and silence this warning.

-- Docs: https://docs.pytest.org/en/latest/warnings.html

Finished <<< examples_rclpy_executors [5.55s]
--- stderr: examples_rclpy_minimal_action_server
=============================== warnings summary ===============================
/usr/local/lib/python3.8/site-packages/_pytest/junitxml.py:417
  Warning: The 'junit_family' default value will change to 'xunit2' in pytest 6.0.
  Add 'junit_family=xunit1' to your pytest.ini file to keep the current format in future versions of pytest and silence this warning.

-- Docs: https://docs.pytest.org/en/latest/warnings.html

Finished <<< examples_rclpy_minimal_action_server [5.42s]
--- stderr: examples_rclpy_minimal_client

=============================== warnings summary ===============================
/usr/local/lib/python3.8/site-packages/_pytest/junitxml.py:417
  Warning: The 'junit_family' default value will change to 'xunit2' in pytest 6.0.
  Add 'junit_family=xunit1' to your pytest.ini file to keep the current format in future versions of pytest and silence this warning.

-- Docs: https://docs.pytest.org/en/latest/warnings.html

Finished <<< examples_rclpy_minimal_client [5.34s]
--- stderr: examples_rclpy_minimal_publisher                                                                                                                    

=============================== warnings summary ===============================
/usr/local/lib/python3.8/site-packages/_pytest/junitxml.py:417
  Warning: The 'junit_family' default value will change to 'xunit2' in pytest 6.0.
  Add 'junit_family=xunit1' to your pytest.ini file to keep the current format in future versions of pytest and silence this warning.

-- Docs: https://docs.pytest.org/en/latest/warnings.html

Finished <<< examples_rclpy_minimal_publisher [5.53s]
--- stderr: examples_rclpy_minimal_service                                                                            

=============================== warnings summary ===============================
/usr/local/lib/python3.8/site-packages/_pytest/junitxml.py:417
  Warning: The 'junit_family' default value will change to 'xunit2' in pytest 6.0.
  Add 'junit_family=xunit1' to your pytest.ini file to keep the current format in future versions of pytest and silence this warning.

-- Docs: https://docs.pytest.org/en/latest/warnings.html

Finished <<< examples_rclpy_minimal_service [6.07s]
--- stderr: examples_rclpy_minimal_subscriber                     

=============================== warnings summary ===============================
/usr/local/lib/python3.8/site-packages/_pytest/junitxml.py:417
  Warning: The 'junit_family' default value will change to 'xunit2' in pytest 6.0.
  Add 'junit_family=xunit1' to your pytest.ini file to keep the current format in future versions of pytest and silence this warning.

-- Docs: https://docs.pytest.org/en/latest/warnings.html

Finished <<< examples_rclpy_minimal_subscriber [6.67s]

Summary: 16 packages finished [9.07s]
  7 packages had stderr output: examples_rclpy_executors examples_rclpy_minimal_action_client examples_rclpy_minimal_action_server examples_rclpy_minimal_client examples_rclpy_minimal_publisher examples_rclpy_minimal_service examples_rclpy_minimal_subscriber

#### Additional information

All remaining packages passed

clalancette, 2020-06-16:
See https://github.com/ros2/examples/pull/279, which should fix this.

closed 2020-06-16 by clalancette

snehaldb90, 2020-06-16:
I still get the same warning.

clalancette, 2020-06-16:
> I still get the same warning.

Right, that will be because you are probably on the foxy branch, and I merged this for master.  I'll open a backport PR as well.