ros2/examples #282: rclpy package tests fail for Foxy

shoeffner, 2020-08-02:
## Bug report

**Required Info:**

- Operating System:
  -  macOS 10.15.5
- Installation type:
  -  Binaries
- Version or commit hash:
  -  Foxy Patch 1 (7c75c1c046fcde4cdc8e3a06a027a255d8e1a566), Foxy branch of examples (fe8a47b785c2749242916a7babcc4982f0c9ad14)
- DDS implementation:
  -  I do not know, I get a Connext warning though 
```[connext_cmake_module] Warning: The location at which Connext was found when the workspace was built [[/Applications/rti_connext_dds-5.3.1]] does not point to a valid directory, and the NDDSHOME environment variable has not been set. Support for Connext will not be available.```
- Client library (if applicable):
  -  n/a

#### Steps to reproduce issue

```bash
colcon build --symlink-install
colcon test
```

#### Expected behavior

All tests pass.

#### Actual behavior

5 packages failed: examples_rclpy_executors examples_rclpy_minimal_client examples_rclpy_minimal_publisher examples_rclpy_minimal_service examples_rclpy_minimal_subscriber

```
Starting >>> examples_rclcpp_minimal_action_client
Starting >>> examples_rclcpp_minimal_action_server
Starting >>> examples_rclcpp_minimal_client
Starting >>> examples_rclcpp_minimal_composition
Starting >>> examples_rclcpp_minimal_publisher                                       
Starting >>> examples_rclcpp_minimal_service
Starting >>> examples_rclcpp_minimal_subscriber
Starting >>> examples_rclcpp_minimal_timer
Finished <<< examples_rclcpp_minimal_action_client [0.29s]                           
Starting >>> examples_rclcpp_multithreaded_executor
Finished <<< examples_rclcpp_minimal_action_server [0.32s]                                                                   
Finished <<< examples_rclcpp_minimal_client [0.29s]
Finished <<< examples_rclcpp_minimal_publisher [0.24s]
Finished <<< examples_rclcpp_minimal_service [0.21s]
Finished <<< examples_rclcpp_minimal_composition [0.28s]
Finished <<< examples_rclcpp_minimal_subscriber [0.19s]
Starting >>> examples_rclpy_executors
Starting >>> examples_rclpy_minimal_action_client                                                                            
Starting >>> examples_rclpy_minimal_action_server
Starting >>> examples_rclpy_minimal_client                                                                                   
Starting >>> examples_rclpy_minimal_publisher
Starting >>> examples_rclpy_minimal_service                                                                                  
Finished <<< examples_rclcpp_minimal_timer [0.46s]
Starting >>> examples_rclpy_minimal_subscriber                                                                          
Finished <<< examples_rclcpp_multithreaded_executor [1.73s]
Failed   <<< examples_rclpy_executors [2.35s, exited with code 2]                                                     
Finished <<< examples_rclpy_minimal_action_client [2.30s]
Finished <<< examples_rclpy_minimal_action_server [2.21s]
Failed   <<< examples_rclpy_minimal_client [2.56s, exited with code 2]                                                             
Failed   <<< examples_rclpy_minimal_publisher [2.87s, exited with code 2]                   
Failed   <<< examples_rclpy_minimal_service [3.14s, exited with code 2]                                                          
Failed   <<< examples_rclpy_minimal_subscriber [2.08s, exited with code 2]   

Summary: 11 packages finished [4.62s]
  5 packages failed: examples_rclpy_executors examples_rclpy_minimal_client examples_rclpy_minimal_publisher examples_rclpy_minimal_service examples_rclpy_minimal_subscriber
```

#### Additional information
[... 607 lines cut ...]
[3.931s] Level 1:colcon.colcon_core.task.python.test:test.step() by extension 'pytest'
[3.938s] DEBUG:colcon.colcon_core.event_handler.log_command:Invoking command in '/Users/shoeffner/Projects/test/ros_test_ws/src/examples/rclpy/topics/minimal_subscriber': AMENT_PREFIX_PATH=/Users/shoeffner/Projects/test/ros_test_ws/src/examples/install/examples_rclpy_minimal_subscriber:${AMENT_PREFIX_PATH} PYTEST_ADDOPTS=--tb=short --junit-xml=/Users/shoeffner/Projects/test/ros_test_ws/src/examples/build/examples_rclpy_minimal_subscriber/pytest.xml --junit-prefix=examples_rclpy_minimal_subscriber -o cache_dir=/Users/shoeffner/Projects/test/ros_test_ws/src/examples/build/examples_rclpy_minimal_subscriber/.pytest_cache PYTHONPATH=/Users/shoeffner/Projects/test/ros_test_ws/src/examples/build/examples_rclpy_minimal_subscriber:/Users/shoeffner/Projects/test/ros_test_ws/src/examples/install/examples_rclpy_minimal_subscriber/lib/python3.8/site-packages:${PYTHONPATH} SHLVL=2 _=/usr/bin/env /usr/local/opt/python@3.8/bin/python3.8 -m pytest
[3.945s] DEBUG:colcon.colcon_core.event_handler.log_command:Invoked command in '/Users/shoeffner/Projects/test/ros_test_ws/src/examples/rclpy/actions/minimal_action_server' returned '5': AMENT_PREFIX_PATH=/Users/shoeffner/Projects/test/ros_test_ws/src/examples/install/examples_rclpy_minimal_action_server:${AMENT_PREFIX_PATH} PYTEST_ADDOPTS=--tb=short --junit-xml=/Users/shoeffner/Projects/test/ros_test_ws/src/examples/build/examples_rclpy_minimal_action_server/pytest.xml --junit-prefix=examples_rclpy_minimal_action_server -o cache_dir=/Users/shoeffner/Projects/test/ros_test_ws/src/examples/build/examples_rclpy_minimal_action_server/.pytest_cache PYTHONPATH=/Users/shoeffner/Projects/test/ros_test_ws/src/examples/build/examples_rclpy_minimal_action_server:/Users/shoeffner/Projects/test/ros_test_ws/src/examples/install/examples_rclpy_minimal_action_server/lib/python3.8/site-packages:${PYTHONPATH} SHLVL=2 _=/usr/bin/env /usr/local/opt/python@3.8/bin/python3.8 -m pytest
Finished <<< examples_rclpy_minimal_action_server [2.51s]
[3.965s] DEBUG:colcon.colcon_core.event_handler.log_command:Invoked command in '/Users/shoeffner/Projects/test/ros_test_ws/src/examples/rclpy/services/minimal_client' returned '2': AMENT_PREFIX_PATH=/Users/shoeffner/Projects/test/ros_test_ws/src/examples/install/examples_rclpy_minimal_client:${AMENT_PREFIX_PATH} PYTEST_ADDOPTS=--tb=short --junit-xml=/Users/shoeffner/Projects/test/ros_test_ws/src/examples/build/examples_rclpy_minimal_client/pytest.xml --junit-prefix=examples_rclpy_minimal_client -o cache_dir=/Users/shoeffner/Projects/test/ros_test_ws/src/examples/build/examples_rclpy_minimal_client/.pytest_cache PYTHONPATH=/Users/shoeffner/Projects/test/ros_test_ws/src/examples/build/examples_rclpy_minimal_client:/Users/shoeffner/Projects/test/ros_test_ws/src/examples/install/examples_rclpy_minimal_client/lib/python3.8/site-packages:${PYTHONPATH} SHLVL=2 _=/usr/bin/env /usr/local/opt/python@3.8/bin/python3.8 -m pytest
Failed   <<< examples_rclpy_minimal_client [2.48s, exited with code 2]
[4.294s] DEBUG:colcon.colcon_core.event_handler.log_command:Invoked command in '/Users/shoeffner/Projects/test/ros_test_ws/src/examples/rclpy/topics/minimal_publisher' returned '2': AMENT_PREFIX_PATH=/Users/shoeffner/Projects/test/ros_test_ws/src/examples/install/examples_rclpy_minimal_publisher:${AMENT_PREFIX_PATH} PYTEST_ADDOPTS=--tb=short --junit-xml=/Users/shoeffner/Projects/test/ros_test_ws/src/examples/build/examples_rclpy_minimal_publisher/pytest.xml --junit-prefix=examples_rclpy_minimal_publisher -o cache_dir=/Users/shoeffner/Projects/test/ros_test_ws/src/examples/build/examples_rclpy_minimal_publisher/.pytest_cache PYTHONPATH=/Users/shoeffner/Projects/test/ros_test_ws/src/examples/build/examples_rclpy_minimal_publisher:/Users/shoeffner/Projects/test/ros_test_ws/src/examples/install/examples_rclpy_minimal_publisher/lib/python3.8/site-packages:${PYTHONPATH} SHLVL=2 _=/usr/bin/env /usr/local/opt/python@3.8/bin/python3.8 -m pytest
Failed   <<< examples_rclpy_minimal_publisher [2.76s, exited with code 2]
[4.593s] DEBUG:colcon.colcon_core.event_handler.log_command:Invoked command in '/Users/shoeffner/Projects/test/ros_test_ws/src/examples/rclpy/services/minimal_service' returned '2': AMENT_PREFIX_PATH=/Users/shoeffner/Projects/test/ros_test_ws/src/examples/install/examples_rclpy_minimal_service:${AMENT_PREFIX_PATH} PYTEST_ADDOPTS=--tb=short --junit-xml=/Users/shoeffner/Projects/test/ros_test_ws/src/examples/build/examples_rclpy_minimal_service/pytest.xml --junit-prefix=examples_rclpy_minimal_service -o cache_dir=/Users/shoeffner/Projects/test/ros_test_ws/src/examples/build/examples_rclpy_minimal_service/.pytest_cache PYTHONPATH=/Users/shoeffner/Projects/test/ros_test_ws/src/examples/build/examples_rclpy_minimal_service:/Users/shoeffner/Projects/test/ros_test_ws/src/ex [... cut ...]
[4.923s] INFO:colcon.colcon_core.plugin_system:Skipping extension 'colcon_notification.desktop_notification.notify2': 'notify2' not found
[4.923s] INFO:colcon.colcon_core.plugin_system:Skipping extension 'colcon_notification.desktop_notification.notify_send': Not used on non-Linux systems
[4.923s] INFO:colcon.colcon_core.plugin_system:Skipping extension 'colcon_notification.desktop_notification.win32': Not used on non-Windows systems
[4.923s] INFO:colcon.colcon_notification.desktop_notification:Sending desktop notification using 'terminal_notifier'
[5.022s] DEBUG:colcon.colcon_core.event_reactor:joined thread
```

athackst, 2020-08-18:
Have you tried running `py3clean .` on the directory?  I think I saw something similar when I was switching between versions.

shoeffner, 2020-08-28:
Unfortunately, no, and I instead switched to melodic.
Feel free to close!

closed 2020-09-17 by clalancette