ros2/examples #286: Linting errors in examples

athackst, 2020-08-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04/ Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - foxy, eloquent, dashing
- DDS implementation:
  - n/a
- Client library (if applicable):
  - n/a

#### Steps to reproduce issue

The following linters show errors in this package

<details>
  <summary>ament_cpplint</summary>

```
/home/runner/work/vscode_ros2_workspace/vscode_ros2_workspace/src/examples/rclcpp/multithreaded_executor/multithreaded_executor.cpp:0:  No copyright message found.  You should have a line: "Copyright [year] <Copyright Owner>"  [legal/copyright] [5]
Category 'legal/copyright' errors found: 1
Total errors found: 1
```
</details>

<details>
  <summary>ament_flake8</summary>

```
src/examples/rclpy/actions/minimal_action_server/examples_rclpy_minimal_action_server/server.py:45 in public method `goal_callback`: D401: First line should be in imperative mood (perhaps 'Accept', not 'Accepts')
src/examples/rclpy/actions/minimal_action_server/examples_rclpy_minimal_action_server/server.py:51 in public method `cancel_callback`: D401: First line should be in imperative mood (perhaps 'Accept', not 'Accepts')
src/examples/rclpy/actions/minimal_action_server/examples_rclpy_minimal_action_server/server.py:56 in public method `execute_callback`: D401: First line should be in imperative mood (perhaps 'Execute', not 'Executes')
checking src/examples/rclpy/actions/minimal_action_server/examples_rclpy_minimal_action_server/server_defer.py
src/examples/rclpy/actions/minimal_action_server/examples_rclpy_minimal_action_server/server_defer.py:48 in public method `goal_callback`: D401: First line should be in imperative mood (perhaps 'Accept', not 'Accepts')
src/examples/rclpy/actions/minimal_action_server/examples_rclpy_minimal_action_server/server_defer.py:53 in public method `handle_accepted_callback`: D401: First line should be in imperative mood (perhaps 'Provide', not 'Provides')
src/examples/rclpy/actions/minimal_action_server/examples_rclpy_minimal_action_server/server_defer.py:59 in public method `cancel_callback`: D401: First line should be in imperative mood (perhaps 'Accept', not 'Accepts')
src/examples/rclpy/actions/minimal_action_server/examples_rclpy_minimal_action_server/server_defer.py:70 in public method `execute_callback`: D401: First line should be in imperative mood (perhaps 'Execute', not 'Executes')
checking src/examples/rclpy/actions/minimal_action_server/examples_rclpy_minimal_action_server/server_not_composable.py
src/examples/rclpy/actions/minimal_action_server/examples_rclpy_minimal_action_server/server_not_composable.py:35 in public function `execute_callback`: D401: First line should be in imperative mood (perhaps 'Execute', not 'Executes')
checking src/examples/rclpy/actions/minimal_action_server/examples_rclpy_minimal_action_server/server_queue_goals.py
src/examples/rclpy/actions/minimal_action_server/examples_rclpy_minimal_action_server/server_queue_goals.py:42 in public method `goal_callback`: D401: First line should be in imperative mood (perhaps 'Accept', not 'Accepts')
src/examples/rclpy/actions/minimal_action_server/examples_rclpy_minimal_action_server/server_queue_goals.py:47 in public method `cancel_callback`: D401: First line should be in imperative mood (perhaps 'Accept', not 'Accepts')
src/examples/rclpy/actions/minimal_action_server/examples_rclpy_minimal_action_server/server_queue_goals.py:52 in public method `execute_callback`: D401: First line should be in imperative mood (perhaps 'Execute', not 'Executes')
checking src/examples/rclpy/actions/minimal_action_server/examples_rclpy_minimal_action_server/server_single_goal.py
src/examples/rclpy/actions/minimal_action_server/examples_rclpy_minimal_action_server/server_single_goal.py:49 in public method `goal_callback`: D401: First line should be in imperative mood (perhaps 'Accept', not 'Accepts')
src/examples/rclpy/actions/minimal_action_server/examples_rclpy_minimal_action_server/server_single_goal.py:65 in public method `cancel_callback`: D401: First line should be in imperative mood (perhaps 'Accept', not 'Accepts')
src/examples/rclpy/actions/minimal_action_server/examples_rclpy_minimal_action_server/server_single_goal.py:70 in public method `execute_callback`: D401: First line should be in imperative mood (perhaps 'Execute', not 'Executes')
```
</details>

<details>
  <summary>ament_pep257</summary>

```
src/examples/rclpy/actions/minimal_action_client/examples_rclpy_minimal_action_client/client_cancel.py:15:1: F401 'action_msgs.msg.GoalStatus' imported but unused
from action_msgs.msg import GoalStatus
^

src/examples/rclpy/actions/minimal_action_client/examples_rclpy_minimal_action_client/client_cancel.py:20:1: F401 'rclpy.callback_groups.ReentrantCallbackGroup' imported but unused
from rclpy.callback_groups import ReentrantCallbackGroup
^

src/examples/rclpy/actions/minimal_action_client/examples_rclpy_minimal_action_client/client_not_composable.py:21:1: F401 'rclpy.node.Node' imported but unused
from rclpy.node import Node
^

[... 64 lines cut ...]
4     F401 'action_msgs.msg.GoalStatus' imported but unused

55 files checked
18 errors

'D'-type errors: 14
'F'-type errors: 4
```
</details>

<details>
  <summary>ament_lint_cmake</summary>

```
src/examples/rclcpp/minimal_composition/CMakeLists.txt:30: Mismatching spaces inside () after command [whitespace/mismatch]
```
</details>

The fixes for these are pretty straight forward and I've implemented them here:

https://github.com/athackst/examples/tree/foxy
https://github.com/athackst/examples/tree/eloquent
https://github.com/athackst/examples/tree/dashing

Is this something of use to this repo?

clalancette, 2020-08-11:
In general, yes.  What I'm confused about is why these problems don't show up in CI or my local testing.  @athackst did you have to enable more things in the CMakeLists.txt to see these errors?

If you do open a PR, please open one targeting the latest (`master` branch), and then we can consider whether it is worthwhile to backport from there.  Thanks.

athackst, 2020-08-12:
Ah.  If you want to add them as unit tests then yes, you'll have to add more things to the CMakelists.txt file.  To see the errors you just have to run the commands on the repo.  I found them when I was setting up a [vscode/ros2 template repo](https://github.com/athackst/vscode_ros2_workspace) which includes linting in a [GitHub action](https://github.com/athackst/vscode_ros2_workspace/blob/foxy/.github/workflows/ros.yaml).  You may want to consider doing something similar.

If you run the linters on the folder you should see the same errors.

It looked like the minimal_action_client and minimal_action_server were missing a unit test folder

clalancette, 2020-08-12:
Ah, I see what happened here.  https://github.com/ros2/examples/commit/df1f37c833e63392bd544d1cf39ab86b7581b1b4 fixed up all of the problems you see on master; we just haven't backported that to any of the active distributions.

@athackst If it makes your life easier, please feel free to open up backport PRs for that commit.  If it doesn't make too much difference to you one way or another, please close this out.  Thanks for the report.

athackst, 2020-08-13:
I think that df1f37c only fixes the CPP packages.  I've opened a PR to address the python packages.  I would like to backport it, if possible.

athackst, 2020-08-18:
Opened a backport for foxy.  Is that what you need?  I did have to sign off again on the cherry-picks, so it might be easier on your end.  I can go ahead and open the others if that looks good to you.

jacobperron, 2020-09-08:
@athackst Thanks for the PRs; all are merged :bow:

closed 2020-09-08 by jacobperron