ros2/examples #293: Bug when build with colcon

swordmanager, 2020-10-16:
Hi,

I've followed the installation procedure then the Using colcon to build packages tutorial.
However at build step (colcon build --symlink-install) I'm getting the following:

`Starting >>> turtlebot3_msgs
Starting >>> dynamixel_sdk
Starting >>> hls_lfcd_lds_driver
Starting >>> turtlebot3_description
Starting >>> turtlebot3_cartographer
Starting >>> turtlebot3_navigation2
Starting >>> turtlebot3_teleop
Finished <<< turtlebot3_navigation2 [0.84s]                          
Finished <<< turtlebot3_cartographer [0.87s]                         
Finished <<< turtlebot3_description [0.88s]
Finished <<< dynamixel_sdk [0.91s]
Finished <<< turtlebot3_teleop [0.94s]
--- stderr: turtlebot3_msgs                                          
CMake Error at /opt/ros/dashing/share/rosidl_generator_cpp/cmake/rosidl_generator_cpp_generate_interfaces.cmake:108 (find_package):
  By not providing "Findament_cmake_cppcheck.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ament_cmake_cppcheck", but CMake did not find one.

  Could not find a package configuration file provided by
  "ament_cmake_cppcheck" with any of the following names:

    ament_cmake_cppcheckConfig.cmake
    ament_cmake_cppcheck-config.cmake

  Add the installation prefix of "ament_cmake_cppcheck" to CMAKE_PREFIX_PATH
  or set "ament_cmake_cppcheck_DIR" to a directory containing one of the
  above files.  If "ament_cmake_cppcheck" provides a separate development
  package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  /opt/ros/dashing/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  /opt/ros/dashing/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  CMakeLists.txt:45 (rosidl_generate_interfaces)

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/executor/__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_core/task/__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_ros/task/ament_cmake/build.py", line 66, in build
    if await has_target(args.build_base, 'install'):
  File "/usr/lib/python3/dist-packages/colcon_cmake/task/cmake/__init__.py", line 57, in has_target
    return target in await get_makefile_targets(path)
  File "/usr/lib/python3/dist-packages/colcon_cmake/task/cmake/__init__.py", line 78, in get_makefile_targets
    CMAKE_EXECUTABLE, '--build', path, '--target', 'help'], cwd=path)
  File "/usr/lib/python3/dist-packages/colcon_core/subprocess.py", line 110, in check_output
    'Expected {args} to pass: {stderr_data}'.format_map(locals())
AssertionError: Expected ['/usr/bin/cmake', '--build', '/home/phamtuan/turtlebot3_ws/build/turtlebot3_msgs', '--target', 'help'] to pass: make: *** No rule to make target 'help'.  Stop.

---
Failed   <<< turtlebot3_msgs [1.07s, exited with code 1]
Aborted  <<< hls_lfcd_lds_driver [1.34s]                                   

Summary: 5 packages finished [1.58s]
  1 package failed: turtlebot3_msgs
  1 package aborted: hls_lfcd_lds_driver
  1 package had stderr output: turtlebot3_msgs
  7 packages not processed
Expected ['/usr/bin/cmake', '--build', '/home/phamtuan/turtlebot3_ws/build/turtlebot3_msgs', '--target', 'help'] to pass: make: *** No rule to make target 'help'.  Stop.`

clalancette, 2020-10-16:
This looks like you probably forgot to source `/opt/ros/<distro>/setup.bash` in the terminal that you tried to build in.

In general, please ask questions like this on https://answers.ros.org.  That way the questions and answers are all in one place that is easily searchable.  Please also include exact instructions on what you did (like which ROS distribution you are using).  I'm going to close this out for now.

closed 2020-10-16 by clalancette

dirk-thomas, 2020-10-16:
> `AssertionError: Expected ['/usr/bin/cmake', '--build', '/home/phamtuan/turtlebot3_ws/build/turtlebot3_msgs', '--target', 'help'] to pass: make: *** No rule to make target 'help'. Stop.`

@swordmanager If you update to the just released version of `colcon-ros` (0.3.21) the above follow-up error message will go away.