ros2/examples #294: rclcpp_minimal_action_client not building

TimCraig, 2020-10-31:
## Bug report`

**Required Info:**

- Operating System:
  Ubuntu-MATE 20.04
- Installation type:
  binaries

#### Steps to reproduce issue
I've had a problem getting the examples to build for a while.  I first thought something had gotten out of sync so I got around to creating a new virtualbox with Windows10 host and Ubuntu-MATE 20.04 client and pristine install of the Foxy binaries.  rclcpp_minimal_action_client fails to build.  Below is the output from the compiler.

Starting >>> examples_rclcpp_minimal_action_client
--- stderr: examples_rclcpp_minimal_action_client                             
/home/brutus/ros2/examples_ws/src/examples/rclcpp/actions/minimal_action_client/member_functions.cpp: In member function ‘void MinimalActionClient::send_goal()’:
/home/brutus/ros2/examples_ws/src/examples/rclcpp/actions/minimal_action_client/member_functions.cpp:78:71: error: no match for ‘operator=’ (operand types are ‘rclcpp_action::Client<example_interfaces::action::Fibonacci>::GoalResponseCallback’ {aka ‘std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> > >)>’} and ‘std::_Bind_helper<false, void (MinimalActionClient::*)(std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> >), MinimalActionClient*, const std::_Placeholder<1>&>::type’ {aka ‘std::_Bind<void (MinimalActionClient::*(MinimalActionClient*, std::_Placeholder<1>))(std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> >)>’})
   78 |       std::bind(&MinimalActionClient::goal_response_callback, this, _1);
      | 
                                                                            ^
In file included from /usr/include/c++/9/functional:59,
                 from /home/brutus/ros2/examples_ws/src/examples/rclcpp/actions/minimal_action_client/member_functions.cpp:17:
/usr/include/c++/9/bits/std_function.h:462:7: note: candidate: ‘std::function<_Res(_ArgTypes ...)>& std::function<_Res(_ArgTypes ...)>::operator=(const std::function<_Res(_ArgTypes ...)>&) [with _Res = void; _ArgTypes = {std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> > >}]’
  462 |       operator=(const function& __x)
      |       ^~~~~~~~
      
/usr/include/c++/9/bits/std_function.h:462:33: note:   no known conversion for argument 1 from ‘std::_Bind_helper<false, void (MinimalActionClient::*)(std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> >), MinimalActionClient*, const std::_Placeholder<1>&>::type’ {aka ‘std::_Bind<void (MinimalActionClient::*(MinimalActionClient*, std::_Placeholder<1>))(std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> >)>’} to ‘const std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> > >)>&’
  462 |       operator=(const function& __x)
      |                 ~~~~~~~~~~~~~~~~^~~
      
/usr/include/c++/9/bits/std_function.h:480:7: note: candidate: ‘std::function<_Res(_ArgTypes ...)>& std::function<_Res(_ArgTypes ...)>::operator=(std::function<_Res(_ArgTypes ...)>&&) [with _Res = void; _ArgTypes = {std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> > >}]’
  480 |       operator=(function&& __x) noexcept
      |       ^~~~~~~~
      
/usr/include/c++/9/bits/std_function.h:480:28: note:   no known conversion for argument 1 from ‘std::_Bind_helper<false, void (MinimalActionClient::*)(std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> >), MinimalActionClient*, const std::_Placeholder<1>&>::type’ {aka ‘std::_Bind<void (MinimalActionClient::*(MinimalActionClient*, std::_Placeholder<1>))(std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> >)>’} to ‘std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> > >)>&&’
  480 |       operator=(function&& __x) noexcept
      |                 ~~~~~~~~~~~^~~
      
/usr/include/c++/9/bits/std_function.h:494:7: note: candidate: ‘std::function<_Res(_ArgTypes ...)>& std::function<_Res(_ArgTypes ...)>::operator=(std::nullptr_t) [with _Res = void; _ArgTypes = {std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> > >}; std::nullptr_t = std::nullptr_t]’
  494 |       operator=(nullptr_t) noexcept
      |       ^~~~~~~~
      
/usr/include/c++/9/bits/std_function.h:494:17: note:   no known conversion for argument 1 from ‘std::_Bind_helper<false, void (MinimalActionClient::*)(std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> >), MinimalActionClient*, const std::_Placeholder<1>&>::type’ {aka ‘std::_Bind<void (MinimalActionClient::*(MinimalActionClient*, std::_Placeholder<1>))(std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> >)>’} to ‘std::nullptr_t’
  494 |       operator=(nullptr_t) noexcept
      |                 ^~~~~~~~~
      
/usr/include/c++/9/bits/std_function.h:523:2: note: candidate: ‘template<class _Functor> std::function<_Res(_ArgTypes ...)>::_Requires<std::function<_Res(_ArgTypes ...)>::_Callable<typename std::decay<_Functor>::type>, std::function<_Res(_ArgTypes ...)>&> std::function<_Res(_ArgTypes ...)>::operator=(_Functor&&) [with _Functor = _Functor; _Res = void; _ArgTypes = {std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> > >}]’
  523 |  operator=(_Functor&& __f)
      |  ^~~~~~~~
      
/usr/include/c++/9/bits/std_function.h:523:2: note:   template argument deduction/substitution failed:
/usr/include/c++/9/bits/std_function.h: In substitution of ‘template<class _Res, class ... _ArgTypes> template<class _Cond, class _Tp> using _Requires = typename std::enable_if<_Cond::value, _Tp>::type [with _Cond = std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> > >)>::_Callable<std::_Bind<void (MinimalActionClient::*(MinimalActionClient*, std::_Placeholder<1>))(std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> >)>, std::__invoke_result<std::_Bind<void (MinimalActionClient::*(MinimalActionClient*, std::_Placeholder<1>))(std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> >)>&, std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> > > > >; _Tp = std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> > >)>&; _Res = void; _ArgTypes = {std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> > >}]’:
/usr/include/c++/9/bits/std_function.h:523:2:   required by substitution of ‘template<class _Functor> std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> > >)>::_Requires<std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> > >)>::_Callable<typename std::decay<_Tp>::type, std::__invoke_result<typename std::decay<_Tp>::type&, std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> > > > >, std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> > >)>&> std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> > >)>::operator=<_Functor>(_Functor&&) [with _Functor = std::_Bind<void (MinimalActionClient::*(MinimalActionClient*, std::_Placeholder<1>))(std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> >)>]’
/home/brutus/ros2/examples_ws/src/examples/rclcpp/actions/minimal_action_client/member_functions.cpp:78:71:   required from here
/usr/include/c++/9/bits/std_function.h:385:8: error: no type named ‘type’ in ‘struct std::enable_if<false, std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<example_interfaces::action::Fibonacci> > >)>&>’
  385 |  using _Requires = typename enable_if<_Cond::value, _Tp>::type;
      |        ^~~~~~~~~
      
[... cut ...]
   78 |       std::bind(&MinimalActionClient::goal_response_callback, this, _1);
      |                                                                       ^
      
make[2]: *** [CMakeFiles/action_client_member_functions.dir/build.make:63: CMakeFiles/action_client_member_functions.dir/member_functions.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:165: CMakeFiles/action_client_member_functions.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< examples_rclcpp_minimal_action_client [3.33s, exited with code 2]

```

```

clalancette, 2020-11-01:
Which branch of the examples repository are you attempting to build?  You have to use the `foxy` branch to build against Foxy.

TimCraig, 2020-11-01:
I'm using what comes with "Download the Zip", I assumed that would be the latest.

TimCraig, 2020-11-02:
Ok, figuring out where to get the Foxy version and using it solved the problem.  Considering the people interested most in the examples are new to ROS2, a note in in the ReadMe might be useful indicating how to get the right version and that it's important.

clalancette, 2020-11-02:
While I wouldn't be opposed to adding a note in the README, this isn't our usual entry point for new users.  We usually recommend people start with the tutorials at https://index.ros.org/doc/ros2/Tutorials/#tutorials , which have instructions on how to install the examples from binaries (and generate your own examples).

Just out of curiosity, how did you end up directly at this repository?

TimCraig, 2020-11-03:
I don't remember exactly but probably one of my forays looking for real ROS2 example code on github.  Just search for ROS2 and filter with recently updated.  Installing examples and running them from binaries seems a waste of time for a developer.  Taking these examples and learning how to modify them to suit my cases seems to me more productive  since documentation is pretty sparse.  I think I first grabbed the examples under Dashing and have checked for updates periodically.

clalancette, 2020-11-03:
>  Installing examples and running them from binaries seems a waste of time for a developer.

Yes, that's why the tutorials start with installing from binaries, and then go on to explain how to write your own code (see https://index.ros.org/doc/ros2/Tutorials/Writing-A-Simple-Cpp-Publisher-And-Subscriber/, for example).

That all being said, since the examples here are one possible entry point, a change to the README.md to explain that you must build the branch corresponding to your distribution would not be a bad idea.  If you'd like to contribute that, I'd be happy to review.

TimCraig, 2020-11-03:
I've written my own pub/sub pair for a real device, service server and client, and action.  What I'm looking for lately are the bells and whistles, QoS, error handling, and advance launch details.  Right now I'm cooling my heels in Standford's Cancer Clinic waiting for a chemo treatment that is going to put me on my ass for about a week so modifying the README.md won't be in the cards so some time.  Any coding probably for that matter.  One of the reasons things are going slowly for me.

kilocharlievictor, 2020-12-28:
Hi
I am trying to follow lecture-2 of Autoware Auto and I am also getting the same error when I try to build the samples. See attached snapshot. Am using ade docker.

-----
source /opt/ros/dashing/setup.bash
mkdir -p ~/ros2_example_ws/src
cd ~/ros2_example_ws
git clone https://github.com/ros2/examples src/examples
cd ~/ros2_example_ws/src/examples/
git checkout dashing
cd ~/ros2_example_ws
colcon build --symlink-install
-------
[image]

sloretz, 2020-12-28:
@kilocharlievictor This doesn't seem related to this issue. Please ask a question on https://answers.ros.org, and when you do please copy/paste the text from the terminal instead of taking a screenshot so others can find it by searching.

MShields1986, 2025-02-06:
Same compilation issue as above for me building inside ros:humble-ros-base Docker image running on a Ubuntu 22.04 host system. Switching over to a lamda style implementation for setting the `send_goal_options.goal_response_callback` appears to fix the compilation issue, however I am seeing other issues at runtime that may or may not be related.

```cpp
send_goal_options.goal_response_callback = [this](const GoalHandleFibonacci::SharedPtr & goal_handle)
    {
        if (!goal_handle) {
            RCLCPP_ERROR(rclcpp::get_logger("rclcpp"), "Goal was rejected by server");
        } else {
            RCLCPP_INFO(rclcpp::get_logger("rclcpp"), "Goal accepted by server, waiting for result");
        }
    };
```

commit 30302016 (2020-12-22): Add list of distros branches to README

Fixes #294