ros2/examples #303: Could not import 'rosidl_typesupport_c' for package

joaquin-gl, 2021-03-10:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Binaries
- Version or commit hash:
  - foxy
- Client library (if applicable):
  - rclpy

I have a set of packages built using `ament_python` in a workspace and have been trying to create a separate package, built using `ament-cmake` to contain custom message and service types. I was getting an error and [posted it here](https://answers.ros.org/question/373256/ros2-could-not-import-rosidl_typesupport_c-for-package/), but have not found an answer. Then, I attempted to use [this custom interface tutorial,](https://docs.ros.org/en/foxy/Tutorials/Custom-ROS2-Interfaces.html) which has steps for editing past rclpy examples to use custom interfaces. I ran into the same problem with the same output.

#### Steps to reproduce issue
Follow the rclpy [Publisher / Subscriber](https://docs.ros.org/en/foxy/Tutorials/Writing-A-Simple-Py-Publisher-And-Subscriber.html) or [Service / Client](https://docs.ros.org/en/foxy/Tutorials/Writing-A-Simple-Py-Service-And-Client.html) tutorials and then the [Custom Interfaces](https://docs.ros.org/en/foxy/Tutorials/Custom-ROS2-Interfaces.html) tutorial, editing those previous tutorial scripts to test the new interfaces.

`colcon build` should run without error. Then try running your new, edited script using `ros2 run py_pubsub talker`etc.

#### Expected behavior
Should see the output at the end of the custom interfaces tutorial

#### Actual behavior
Here are some outputs from using `ros2 run` for different edited scripts.

<details>
  <summary>talker</summary>

```bash
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 46, in import_type_support
    return importlib.import_module(module_name, package=pkg_name)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 973, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'tutorial_interfaces.tutorial_interfaces_s__rosidl_typesupport_c'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/joaquin/mlabs/dev_ws/install/py_pubsub/lib/py_pubsub/talkercustom", line 11, in <module>
    load_entry_point('py-pubsub==0.0.0', 'console_scripts', 'talkercustom')()
  File "/home/joaquin/mlabs/dev_ws/install/py_pubsub/lib/python3.8/site-packages/py_pubsub/publisher_member_function_custom.py", line 27, in main
    minimal_publisher = MinimalPublisher()
  File "/home/joaquin/mlabs/dev_ws/install/py_pubsub/lib/python3.8/site-packages/py_pubsub/publisher_member_function_custom.py", line 11, in __init__
    self.publisher_ = self.create_publisher(Num, 'topic', 10)     # CHANGE
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/node.py", line 1140, in create_publisher
    check_for_type_support(msg_type)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/type_support.py", line 29, in check_for_type_support
    msg_type.__class__.__import_type_support__()
  File "/home/joaquin/mlabs/dev_ws/install/tutorial_interfaces/lib/python3.7/site-packages/tutorial_interfaces/msg/_num.py", line 27, in __import_type_support__
    module = import_type_support('tutorial_interfaces')
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 48, in import_type_support
    raise UnsupportedTypeSupport(pkg_name)
rosidl_generator_py.import_type_support_impl.UnsupportedTypeSupport: Could not import 'rosidl_typesupport_c' for package 'tutorial_interfaces'
```
</details>

<details>
  <summary>listener</summary>

```bash
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 46, in import_type_support
    return importlib.import_module(module_name, package=pkg_name)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
[... 76 lines cut ...]

Traceback (most recent call last):
  File "/home/joaquin/mlabs/dev_ws/install/py_srvcli/lib/py_srvcli/clientcustom", line 11, in <module>
    load_entry_point('py-srvcli==0.0.0', 'console_scripts', 'clientcustom')()
  File "/home/joaquin/mlabs/dev_ws/install/py_srvcli/lib/python3.8/site-packages/py_srvcli/client_member_function_custom.py", line 26, in main
    minimal_client = MinimalClientAsync()
  File "/home/joaquin/mlabs/dev_ws/install/py_srvcli/lib/python3.8/site-packages/py_srvcli/client_member_function_custom.py", line 11, in __init__
    self.cli = self.create_client(AddThreeInts, 'add_three_ints')       # CHANGE
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/node.py", line 1248, in create_client
    check_for_type_support(srv_type)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/type_support.py", line 29, in check_for_type_support
    msg_type.__class__.__import_type_support__()
  File "/home/joaquin/mlabs/dev_ws/install/tutorial_interfaces/lib/python3.7/site-packages/tutorial_interfaces/srv/_add_three_ints.py", line 300, in __import_type_support__
    module = import_type_support('tutorial_interfaces')
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 48, in import_type_support
    raise UnsupportedTypeSupport(pkg_name)
rosidl_generator_py.import_type_support_impl.UnsupportedTypeSupport: Could not import 'rosidl_typesupport_c' for package 'tutorial_interfaces'
```
</details>

As you can see, there is a pattern.

#### Additional information

[I'm not the first person to have this problem](https://answers.ros.org/question/350084/define-custom-messages-in-python-package-ros2/), and I haven't been able to find a solution or a workaround thus far. Should I stick to using pre-built messages? Build all my packages with `ament_cmake`?

mabelzhang, 2021-03-10:
I can't reproduce what you're seeing in Foxy. I followed the two tutorials you linked to, and I got this output with the custom `Num` message type:

```
$ ros2 run py_pubsub talker
[INFO] [1615343321.226082955] [minimal_publisher]: Publishing: "0"
[INFO] [1615343321.689618425] [minimal_publisher]: Publishing: "1"
[INFO] [1615343322.189459894] [minimal_publisher]: Publishing: "2"
[INFO] [1615343322.689497095] [minimal_publisher]: Publishing: "3"
[INFO] [1615343323.188021916] [minimal_publisher]: Publishing: "4"
[INFO] [1615343323.689148715] [minimal_publisher]: Publishing: "5"
[INFO] [1615343324.189707000] [minimal_publisher]: Publishing: "6"
[INFO] [1615343324.689562912] [minimal_publisher]: Publishing: "7"
```
```
$ ros2 run py_pubsub listener
[INFO] [1615343324.227974159] [minimal_subscriber]: I heard: "6"
[INFO] [1615343324.690278600] [minimal_subscriber]: I heard: "7"
[INFO] [1615343325.190287056] [minimal_subscriber]: I heard: "8"
[INFO] [1615343325.690386282] [minimal_subscriber]: I heard: "9"
```

One place I noticed that needs updating in the first tutorial ([Python pub/sub](https://docs.ros.org/en/foxy/Tutorials/Writing-A-Simple-Py-Publisher-And-Subscriber.html)) is where it says to add the `<exec_depend>` tags after `ament_python` buildtool. That is outdated.
Currently, the `ament_python` is added in the `<export>` tag. You wouldn't want to add the `<exec_depend>`s in there. They should go outside the `<export>` tag, as in the example [here](https://github.com/ros2/examples/blob/master/rclpy/topics/minimal_publisher/package.xml).

Where do you have your `<exec_depend>` tags? Might that be where the problem is?

Otherwise I didn't see any discrepancies. I followed the tutorials exactly.
The only other thing I can think of is the setup and dependencies. What do you get for this command
```
$ ros2 pkg prefix rosidl_typesupport_c
```

joaquin-gl, 2021-03-10:
Thanks for getting back to me. I did have the tags inside `<export>` and have now moved them outside, like in that example. Unfortunately, I still get the same output.

```bash
 ros2 pkg prefix rosidl_typesupport_c
/opt/ros/foxy
```

joaquin-gl, 2021-03-10:
Yes, (three separate packages with the service / client example) here is my structure:
[image]

in `tutorial_interfaces`, my 1CMakeLists.txt1 has
```
find_package(rosidl_default_generators REQUIRED)

rosidl_generate_interfaces(${PROJECT_NAME}
  "msg/Num.msg"
  "srv/AddThreeInts.srv"
 )
```
and my `package.xml` has
```
  <buildtool_depend>ament_cmake</buildtool_depend>

  <build_depend>rosidl_default_generators</build_depend>

  <exec_depend>rosidl_default_runtime</exec_depend>

  <test_depend>ament_lint_auto</test_depend>
  <test_depend>ament_lint_common</test_depend>

  <member_of_group>rosidl_interface_packages</member_of_group>
```

Do I need these dependencies in `package.xml` for the other packages too, maybe?

mabelzhang, 2021-03-10:
Not much to add from me. Something else I noticed though is that you have build, install, log inside the src directory. That might make things not work correctly. Make sure you delete those and run `colcon build` from the workspace, not from inside the src directory.

joaquin-gl, 2021-03-11:
I deleted those and all build files and `colcon build` again from dev_ws/ to no change. I also uninstalled ros using `sudo apt remove ros-foxy-* && sudo apt autoremove` and then got a fresh install, but got the same error :cry:

clalancette, 2021-03-11:
Since we can't reproduce it, I think you'll have to provide very specific step-by-step instructions on what you are doing, and with what code.

joaquin-gl, 2021-03-11:
Hi, I've made [this temporary repository](https://github.com/joaquin-gl/rosidl-ws) with the code I'm using and instructions to reproduce. Let me know if you get the same error, or what I missed here. Thanks!

mabelzhang, 2021-03-11:
So, using your repo, everything unchanged and just running starting from `colcon build`, I'm producing this output:
```
$ ros2 run py_pubsub talkercustom 
[INFO] [1615454131.158409537] [minimal_publisher]: Publishing: "0"
[INFO] [1615454131.616306963] [minimal_publisher]: Publishing: "1"
[INFO] [1615454132.118315053] [minimal_publisher]: Publishing: "2"
[INFO] [1615454132.618105805] [minimal_publisher]: Publishing: "3"
[INFO] [1615454133.118239157] [minimal_publisher]: Publishing: "4"
```
```
$ ros2 run py_pubsub listenercustom 
[INFO] [1615454132.156022256] [minimal_subscriber]: I heard: "2"
[INFO] [1615454132.619050327] [minimal_subscriber]: I heard: "3"
[INFO] [1615454133.118897506] [minimal_subscriber]: I heard: "4"
```
```
$ ros2 run py_srvcli servicecustom
[INFO] [1615454118.761641774] [minimal_service]: Incoming request
a: 1 b: 2 c: 3
```
```
$ ros2 run py_srvcli clientcustom 1 2 3
[INFO] [1615454118.771857380] [minimal_client_async]: Result of add_three_ints: for 1 + 2 + 3 = 6
```

To further check that I have the latest binary environment to match the environment reported, I retested using our official [foxy Docker image](https://hub.docker.com/_/Ros):
```
$ docker pull ros:foxy-ros-base
```
and I got the same results as above.

I would recommend pulling the Docker image and comparing your environment with it. The results above suggest that the problem isn't in the code, it's in the configuration.

joaquin-gl, 2021-03-11:
Good news!

I tried the image you sent like so:
```bash
sudo docker pull ros:foxy-ros-base
sudo docker run -v ~/rosidl_ws:/home/joaquin/rosidl_ws --rm -i -t ros:foxy-ros-base bash
```
Then tried `colcon build` from `rosidl_ws` and got the same error. But after deleting the build files I was able to get it to work in the docker container!
```bash
# rm -rf build/ install/ log/
# colcon build
Starting >>> tutorial_interfaces
[0.258s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/joaquin/mlabs/rosidl_ws/install/py_srvcli' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.258s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/joaquin/mlabs/rosidl_ws/install/py_pubsub' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.258s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/joaquin/mlabs/rosidl_ws/install/tutorial_interfaces' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.258s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/joaquin/mlabs/rosidl_ws/install/tutorial_interfaces' in the environment variable CMAKE_PREFIX_PATH doesn't exist
Finished <<< tutorial_interfaces [4.61s]                     
Starting >>> py_pubsub
Starting >>> py_srvcli
Finished <<< py_pubsub [0.66s]                                         
Finished <<< py_srvcli [0.67s]

Summary: 3 packages finished [5.44s]
# . install/setup.bash 
# ros2 run py_pubsub talkercustom 
[INFO] [1615488166.694147981] [minimal_publisher]: Publishing: "0"
```

mabelzhang, 2021-03-11:
Great! Glad you were able to get something other than an error.
Does it work on your local machine if you remove `build` `install` directories as well? Sometimes old build files does lead to build errors.

I don't have a real plan as to how exactly to compare... If it were me, I would just poke around without cursing until something good happens. Probably I would compare the environment variables `$ env` (`PYTHONPATH` and maybe others), compare what packages are installed in `/opt/ros/foxy`, especially the dependency packages in `package.xml` and the dependencies of those packages. It sounds a bit brute force and aimless, but the hope is a clue might pop up while looking and Googling.

Do you have multiple ROS versions installed? That shouldn't be a problem, as long as you source things in the right order (ROS 1 first, then ROS 2). If you do, I would suggest not sourcing ROS 1 at all, and just do a clean ROS 2 sourcing.
The only notes I have about that last error message about `rosidl_typesupport_c`, I was running a mix of ROS 1 and ROS 2 intentionally, and the first error I had before the `During handling of the above exception, another exception occurred` line was different. I got past the `rosidl_typesupport_c` one when I sourced `install/setup.bash`

joaquin-gl, 2021-03-11:
Hmm,
So I tried removing the directories and used `colcon build` outside of the container, and got the same error. When I `colcon build` within the container, the scripts run as desired, whether I'm running them from inside or outside the container.

Here are the env differences which might attribute to the error:
```diff
- (no container)
+ (container)

- LANG=en_US.UTF-8
+ LANG=C.UTF-8
- AMENT_PREFIX_PATH=/opt/ros/foxy
+ AMENT_PREFIX_PATH=/home/joaquin/rosidl_ws/install/py_srvcli:/home/joaquin/rosidl_ws/install/py_pubsub:/home/joaquin/rosidl_ws/install/tutorial_interfaces:/opt/ros/foxy
+ CMAKE_PREFIX_PATH=/home/joaquin/rosidl_ws/install/tutorial_interfaces
+ COLCON_PREFIX_PATH=/home/joaquin/rosidl_ws/install
- PYTHONPATH=/opt/ros/foxy/lib/python3.8/site-packages
+ PYTHONPATH=/home/joaquin/rosidl_ws/install/tutorial_interfaces/lib/python3.8/site-packages:/home/joaquin/rosidl_ws/install/py_srvcli/lib/python3.8/site-packages:/home/joaquin/rosidl_ws/install/py_pubsub/lib/python3.8/site-packages:/home/joaquin/rosidl_ws/install/tutorial_interfaces/lib/python3.7/site-packages:/opt/ros/foxy/lib/python3.8/site-packages
- LD_LIBRARY_PATH=/opt/ros/foxy/opt/yaml_cpp_vendor/lib:/opt/ros/foxy/opt/rviz_ogre_vendor/lib:/opt/ros/foxy/lib/x86_64-linux-gnu:/opt/ros/foxy/lib
+ LD_LIBRARY_PATH=/home/joaquin/rosidl_ws/install/tutorial_interfaces/lib:/opt/ros/foxy/opt/yaml_cpp_vendor/lib:/opt/ros/foxy/lib/x86_64-linux-gnu:/opt/ros/foxy/lib
- PATH=/opt/ros/foxy/bin:/home/joaquin/.bin:/home/joaquin/.pyenv/plugins/pyenv-virtualenv/shims:/home/joaquin/.pyenv/shims:/home/joaquin/.pyenv/bin:/home/joaquin/.bin:/home/joaquin/.pyenv/plugins/pyenv-virtualenv/shims:/home/joaquin/.pyenv/shims:/home/joaquin/.pyenv/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
+ PATH=/opt/ros/foxy/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
```

From what I can tell by `ls /opt.ros/foxy/lib` in both situations, they seem to both have the same set of `rosidl` related packages. I've included the `env` and `lib` outputs for container and non-container [in the repo](https://github.com/joaquin-gl/rosidl-ws) as .txt files, if you wanted to take a closer look.

Thanks for your help thus far.

newcanopies, 2021-04-30:
Hi, thanks for this discussion.

I have seemingly the same issue building control_msgs, on Ubuntu 20.04 ROS2 Foxy, server glx vendor string: SGI

~/ws_ROS2$ `colcon build --symlink-install --packages-select control_msgs`
```
Starting >>> control_msgs
--- stderr: control_msgs                         
CMake Error at /opt/ros/foxy/share/rosidl_typesupport_c/cmake/get_used_typesupports.cmake:35 (message):
  No 'rosidl_typesupport_c' found
Call Stack (most recent call first):
  /opt/ros/foxy/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:8 (get_used_typesupports)
  /opt/ros/foxy/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /opt/ros/foxy/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake:21 (find_package)
  /opt/ros/foxy/share/rosidl_default_generators/cmake/rosidl_default_generatorsConfig.cmake:41 (include)
  CMakeLists.txt:15 (find_package)

---
Failed   <<< control_msgs [3.05s, exited with code 1]

Summary: 0 packages finished [3.39s]
  1 package failed: control_msgs
  1 package had stderr output: control_msgs
```

`rm -rf build install log` is done at every new colcon build attempt (the contents of `~/ws_ROS2/build` shown for colcon sequence)
[image]

newcanopies, 2021-04-30:
what should the `COLCON/AMENT/CMAKE_PREFIX_PATH` be ?

$ env
```
SHELL=/bin/bash
ROS_VERSION=2
GAZEBO_MASTER_URI=http://localhost:11345
COLORTERM=truecolor
NVM_INC=/home/user/.nvm/versions/node/v12.20.1/include/node
TERM_PROGRAM_VERSION=1.55.2
WSL_DISTRO_NAME=Ubuntu-20.04
ROS_PYTHON_VERSION=3
NAME=machine
PWD=/home/user
LOGNAME=user
VSCODE_GIT_ASKPASS_NODE=/home/user/.vscode-server/bin/3c4e3df9e89829dce27b7b5c24508306b151f30d/node
HOME=/home/user
GAZEBO_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:
LANG=en_US.UTF-8
GAZEBO_MODEL_DATABASE_URI=http://models.gazebosim.org
LS_COLORS= . . .
AMENT_PREFIX_PATH=/opt/ros/foxy
GIT_ASKPASS=/home/user/.vscode-server/bin/3c4e3df9e89829dce27b7b5c24508306b151f30d/extensions/git/dist/askpass.sh
GAZEBO_RESOURCE_PATH=/usr/share/gazebo-11:/usr/share/gazebo-11:
NVM_DIR=/home/user/.nvm
LESSCLOSE=/usr/bin/lesspipe %s %s
PYTHONPATH=/opt/ros/foxy/lib/python3.8/site-packages
TERM=xterm-256color
LESSOPEN=| /usr/bin/lesspipe %s
USER=user
VSCODE_GIT_IPC_HANDLE=/tmp/vscode-git-b1d11a6d9d.sock
DISPLAY=xxx.xx.xx.1:0.0
SHLVL=1
NVM_CD_FLAGS=
LD_LIBRARY_PATH=/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:/opt/ros/foxy/opt/yaml_cpp_vendor/lib:/opt/ros/foxy/opt/rviz_ogre_vendor/lib:/opt/ros/foxy/lib/x86_64-linux-gnu:/opt/ros/foxy/lib:/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:/usr/lib/x86_64-linux-gnu/gazebo-11/plugins
XDG_RUNTIME_DIR=/run/user/1000
ROS_LOCALHOST_ONLY=0
WSLENV=VSCODE_WSL_EXT_LOCATION/up
VSCODE_GIT_ASKPASS_MAIN=/home/user/.vscode-server/bin/3c4e3df9e89829dce27b7b5c24508306b151f30d/extensions/git/dist/askpass-main.js
GAZEBO_MODEL_PATH=/usr/share/gazebo-11/models/mise_staark_model_description/models:/usr/share/gazebo-11/models:
XDG_DATA_DIRS=/usr/local/share:/usr/share:/var/lib/snapd/desktop
PATH=/home/user/.vscode-server/bin/3c4e3df9e89829dce27b7b5c24508306b151f30d/bin:/home/user/.local/bin:/opt/ros/foxy/bin:/home/user/.nvm/versions/node/v12.20.1/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/mnt/c/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/mnt/c/Python27/:/mnt/c/Python27/Scripts:
 . . . 
NVM_BIN=/home/user/.nvm/versions/node/v12.20.1/bin
HOSTTYPE=x86_64
ROS_DISTRO=foxy
OGRE_RESOURCE_PATH=/usr/lib/x86_64-linux-gnu/OGRE-1.9.0
TERM_PROGRAM=vscode
VSCODE_IPC_HOOK_CLI=/tmp/vscode-ipc-eca0215f-7319-4401-933b-61239177a1d8.sock
_=/usr/bin/env
```

Bmoradi93, 2021-07-29:
I had the same issue:
My solution: One of may message dependencies(e.g. nav_msgs) was not included in my CMakeList.txt. Once it was added, the problem got resolved.
If you have custom messages in your package, that could be the reason.

jacoverster, 2021-08-20:
Hi all, I had the same issue. After inspecting the build log I noticed these lines:

```
[4.935844] (tutorial_interfaces) StderrLine: {'line': b'CMake Error at /opt/ros/foxy/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:64 (message):\n'}
[4.936924] (tutorial_interfaces) StderrLine: {'line': b'  rosidl_generate_interfaces() must be called before ament_package()\n'}
[4.937657] (tutorial_interfaces) StderrLine: {'line': b'Call Stack (most recent call first):\n'}
[4.938331] (tutorial_interfaces) StderrLine: {'line': b'  CMakeLists.txt:39 (rosidl_generate_interfaces)\n'}
```

I modified CMakeLists.txt to and it resolved it:

```
...
if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  # the following line skips the linter which checks for copyrights
  # uncomment the line when a copyright and license is not present in all source files
  #set(ament_cmake_copyright_FOUND TRUE)
  # the following line skips cpplint (only works in a git repo)
  # uncomment the line when this package is not in a git repo
  #set(ament_cmake_cpplint_FOUND TRUE)
  ament_lint_auto_find_test_dependencies()
endif()

find_package(rosidl_default_generators REQUIRED)

rosidl_generate_interfaces(${PROJECT_NAME}
  "msg/Num.msg"
  "srv/AddThreeInts.srv"
 )

ament_package()
```

Hope this helps.

ChangcongWang, 2022-01-11:
I had the same issue. I follow this [answer](https://answers.ros.org/question/326008/ros2-run-symbol-not-found-on-custom-msg/) add DEPENDENCIES in rosidl_generate_interfaces to solve this problem.

SilenceZHOU12345, 2022-01-29:
Hello, I met the same problem as you. I found that this is because I currently use the anaconda environment instead of the system environment of python3.6. When I use ‘’$ conda deactivate‘’ command, the error disappeared.I hope it can help you.

clalancette, 2022-02-03:
Since I don't think there is actually a bug here, I'm going to go ahead and close this one out.  If you think there is a bug, please feel free to open a new issue with the exact reproducing steps.  Otherwise, please ask questions on https://answers.ros.org.

closed 2022-02-03 by clalancette

mozilla2012, 2022-02-25:
I know this issue was just closed, but it's happening to me too.

I have a hunch it has to do with mixing python versions. As it mentions here: [🌐 Using Python Packages with ROS 2](https://docs.ros.org/en/foxy/How-To-Guides/Using-Python-Packages.html)

> **Note**
>
> A cautionary note, if you intended to use pre-packaged binaries (either `deb` files, or the “fat” binary distributions), the Python interpreter must match what was used to build the original binaries. If you intend to use something like `virtualenv` or `pipenv`, make sure to use the system interpreter. If you use something like `conda`, it is very likely that the interpreter will not match the system interpreter and will be incompatible with ROS2 binaries.

I have noticed I can use the `std_msgs` just fine, but my custom interfaces give the `Could not import 'rosidl_typesupport_c' for package` error. If I look at what Python sees in `sys.path`, I see that my interfaces use Python 3.8 while the std_msgs (and the rest of my ROS binary) were compiled with 3.7.

I haven't been able to roll my build back to 3.7 or get an updated binary using 3.8 yet, but that's my current theory.

lhmwtum, 2022-02-28:
Hey everyone,
I had the same issue with 'rosidl_typesupport_c' and I've solved it after reading the comment of @mozilla2012.
In Ubuntu 20.04 LTS, I use virtualenv to run python 3.7 and 3.8 (which is used by the system interpreter).
After settings up a new python3.8 venv + running `rosdep install`  und `colcon build` with this particular python version, I was able to launch ROS2 without the mentioned problem.

Conclusion: Your python interpreter must match what was used to build the original binaries.

ShotaAk, 2022-05-31:
Hello, 
I also had the same problem in a docker image (`osrf/ros-galactic-desktop`).
I found [this](https://github.com/ros2/rosidl/issues/682#issuecomment-1127989182) comment and solved it by adding an `apt-get dist-upgrade` to my Dockerfile. :+1:

Jake-Carter, 2023-09-25:
For anyone still facing this issue (and since it's the first thing that pops up from Google), see https://github.com/ros2/examples/issues/303#issuecomment-1050741701.

I also faced the error when trying to use Python 3.11 on ROS2 Humble, which uses pre-built Python _3.10_ dist-packages.  Matching my Python version to the ROS dist-packages worked for me too.

You can check exactly which version the ROS dist-packages use with `pip show`.

```shell
~ » pip show rosidl-generator-c                                              
Name: rosidl-generator-c
Version: 3.1.5
Summary: UNKNOWN
Home-page: UNKNOWN
Author:
Author-email:
License: UNKNOWN
Location: /opt/ros/humble/local/lib/python3.10/dist-packages # <-- Note python3.10 here
Requires:
Required-by:
```

I was using a local ROS install, so I...
1. Cleaned my colcon workspace (`rm -r build install`)
2. Set the workspace to use Python 3.10.12 with [pyenv](https://github.com/pyenv/pyenv) (`pyenv local 3.10.12`)
3. Re-built the workspace (`colcon build --symlink-install`)
4. Reactivated the install (`source install/setup.zsh`)

... and that fixed the missing import.

I'm not sure how/why ROS2 injects its dist-packages into the Python environment, but `pip list` always showed the ROS packages in the list even in a freshly created virtual environment.  So I think rosdep always sees dependencies as satisfied even across different Python versions.

ciaran-helgen, 2023-10-12:
I had this issue also, but in my case it was due to having a Conda environment enabled. Disabling the conda environment with `conda deactivate` before cleaning the workspace with `rm -rf build install` and building the workspace again with `colcon build` fixed the issue for me. For people more fluent with Python virtual environments there may be a way to get these working together, but I have no advice to give on that

hdy003, 2024-03-22:
> I had this issue also, but in my case it was due to having a Conda environment enabled. Disabling the conda environment with `conda deactivate` before cleaning the workspace with `rm -rf build [...]

谢谢你，super man!

kamronkhakimov, 2024-04-09:
I solved the issue by using a different version of python, the one that is also used in the ros2 code. The exact version is in the build folder of your ros2 workspace. Just run ``grep python . -r`` and you'll find the version somewhere in there. 

I also used ``virtualenv`` instead of changing my python version globally.  

You also might need to install some of the modules along the way, like empy 3.3.4, lark, catkin_pkg, since some older python versions might not include those packages, that newer versions have builtin.

naijekux, 2024-04-12:
I also encoutered the issue and have solved it thanks to the kamronkhakimov' answer. 

To run `printenv | grep -i ROS` to check the python version the ros2 workspace is using, like the line `PYTHONPATH=/opt/ros/humble/lib/python3.10/site-packages:` means that the python version by me is python-3.10.

Then to make sure that your python version conicides with the version you just found out, that means python-3.10 for me.

YKS44, 2024-06-25:
Hi, I'm running into the same issue but my os is Windows 10. I looked into my ROS2 folder and saw that it was using Python3.8, which is also the version of Python that I'm using. I followed the same tutorial and I essentially get the same error.
`Traceback (most recent call last):
  File "C:\dev\ros2_test\install\lib\publisher_py\talker-script.py", line 33, in <module>
    sys.exit(load_entry_point('publisher-py==0.0.0', 'console_scripts', 'talker')())
  File "C:\dev\ros2_test\install\Lib\site-packages\publisher_py\publisher_member_function.py", line 27, in main
    minimal_publisher = MinimalPublisher()
  File "C:\dev\ros2_test\install\Lib\site-packages\publisher_py\publisher_member_function.py", line 11, in __init__
    self.publisher_ = self.create_publisher(Num, 'topic', 10)
  File "C:\Users\PMF\Documents\ROS\ros2-windows\Lib\site-packages\rclpy\node.py", line 1548, in create_publisher
    check_is_valid_msg_type(msg_type)
  File "C:\Users\PMF\Documents\ROS\ros2-windows\Lib\site-packages\rclpy\type_support.py", line 35, in check_is_valid_msg_type
    check_for_type_support(msg_type)
  File "C:\Users\PMF\Documents\ROS\ros2-windows\Lib\site-packages\rclpy\type_support.py", line 29, in check_for_type_support
    msg_or_srv_type.__class__.__import_type_support__()
  File "C:\dev\ros2_test\install\Lib\site-packages\tutorial_interfaces\msg\_num.py", line 36, in __import_type_support__
    module = import_type_support('tutorial_interfaces')
  File "C:\Users\PMF\Documents\ROS\ros2-windows\Lib\site-packages\rosidl_generator_py\import_type_support_impl.py", line 48, in import_type_support
    raise UnsupportedTypeSupport(pkg_name)
rosidl_generator_py.import_type_support_impl.UnsupportedTypeSupport: Could not import 'rosidl_typesupport_c' for package 'tutorial_interfaces'
[ros2run]: Process exited with failure 1`

I quadruple checked all my code and it's identical to the one from the tutorial, but I still can't get rid of the error. Any help would be greatly appreciated.

haohao1331, 2024-07-10:
So I ran into this issue but resolved it through another method. 

For my system, I installed conda first before ros2, and was using the python interpreter associated with conda (python3.12), but not in a virtual environment. I realized that when installing ros2 by following the tutorial [here](https://docs.ros.org/en/foxy/Installation/Ubuntu-Install-Debians.html), it also installed a python interpreter on my system at /usr/bin/python3 which is python3.8, and this has been the python interpreter that ros2 used throughout. However, it seems like the problem for me was that cmake (which is required for custom messages) was using the conda python3.12 while the rest of ros2 was using python3.8 which is what's causing the error. I resolved my problem by following [this](https://robotics.stackexchange.com/questions/105587/ros2-on-windows-10-colcon-build-uses-wrong-python-version-no-module-named) thread, which forces cmake to use the same interpreter as ros2.

boris-il-forte, 2024-10-15:
I had similar issues with ROS humble. It turned out that I had very old python3.6 packages and my system was compiling the rosidl_typesupport modules for python 3.6, and therefore where not loaded by my python interpreter (3.10).

NB: my system was recently upgraded from Ubuntu 20.04

manish-nayak, 2024-11-02:
Faced the same issue while implementing ROS custom message tutorial. Found out that ros install uses a different version. Fixing the CMake to use the specific Python installation as mentioned above finally solved the issue for me after a long session of debugging.

nkhumise-rea, 2024-12-11:
Could there problem be caused by something else besides python packages? I'm having the same issues with ROS humble but I verified that CMake is using python3.10 as my system. I am not using a virtual environment nor conda. My system is Ubuntu 20.04. So far, none of the solution above have worked.