ros2/examples #312: publisher_member_function_with_unique_network_flow_endpoints leads to coredump

fujitatomoya, 2021-04-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - source build
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/2c3dcb37040365ee5f09eede264cd05910134076
- DDS implementation:
  - cyclonedds
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
root@134f29e8f25f:~/ros2_ws/colcon_ws# ros2 run examples_rclcpp_minimal_publisher publisher_member_function_with_unique_network_flow_endpoints
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  error obtaining network flows of publisher: rmw_publisher_get_network_flow_endpoints not implemented, at /root/ros2_ws/colcon_ws/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/rmw_get_network_flow_endpoints.cpp:31, at /root/ros2_ws/colcon_ws/src/ros2/rcl/rcl/src/rcl/network_flow_endpoints.c:83, also error cleaning up network flow array: network_flow_endpoint_array->allocator is null, at /root/ros2_ws/colcon_ws/src/ros2/rmw/rmw/src/network_flow_endpoint_array.c:81: error not set
```

#### Expected behavior

example should not get crashed even if rmw implementation does not support `rmw_publisher_get_network_flow_endpoints`

#### Actual behavior

publisher_member_function_with_unique_network_flow_endpoints coredump

#### Additional information

- rmw_fastdds_cpp supports `rmw_publisher_get_network_flow_endpoints` so this problem cannot be observed.

fujitatomoya, 2021-04-27:
@anamud friendly ping, since you were original author.

anamud, 2021-04-27:
@fujitatomoya Thanks for spotting the issue and submitting a good solution. I have reviewed the patch and suggested a minor change.

endre90, 2021-05-01:
Hi, 

I can't find much info on this type of pubs/subs, so I thought I might just ask the question here. Could somebody elaborate on what does having unique network flow endpoints mean in DDS and what are the benefits compared to nodes that don't use this option? Either that or maybe point me to a link where this is explained?

Thanks in advance,
Endre

anamud, 2021-05-01:
@endre90 Unique network flows enable fine-grained QoS assignment in IP-based networks such as 5G. Please have a look at the design [doc](https://github.com/ros2/design/pull/304) for unique network flows to understand more.

endre90, 2021-05-01:
Interesting.

Thanks @anamud for the thorough explanation in `unique_network_flows.md`. 
There might come a time when something like this could come in handy.

Best Regards,
Endre

clalancette, 2021-05-20:
@fujitatomoya Now that we've merged #313, can we close this?

closed 2021-05-20 by fujitatomoya