ros2/examples #314: ros2 testing examples

simutisernestas, 2021-04-27:
Are there any plans on adding node testing examples?

clalancette, 2021-04-27:
> Are there any plans on adding node testing examples?

Can you explain in more detail what you are looking for?

simutisernestas, 2021-04-27:
Nothing too specific would be great to have guidelines and best practices guide for testing ros2 applications. For instance, it could be unit, integration tests, and examples of using `colcon test` and overall ros2 testing tools like launch_testing package, etc. The problem I'm currently facing is just a lack of informational resources in order to properly test ros2 application.

clalancette, 2021-04-27:
OK, that makes sense.  I'm going to move this over to the ros2_documentation repository, as that seems to be the place where we would do documentation of this sort.