ros2/examples #318: foxy colcon build failed

HorsonLiu, 2021-07-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu20.04
- Installation type:
  - git cone
- Version or commit hash:
  - HEAD -b foxy

#### Steps to reproduce issue
```
ubuntu@ubuntu:~/dev_ws$ git clone https://github.com/ros2/examples src/examples  -b foxy
Cloning into 'src/examples'...
remote: Enumerating objects: 5764, done.
remote: Counting objects: 100% (591/591), done.
remote: Compressing objects: 100% (157/157), done.
remote: Total 5764 (delta 431), reused 567 (delta 420), pack-reused 5173
Receiving objects: 100% (5764/5764), 931.69 KiB | 183.00 KiB/s, done.
Resolving deltas: 100% (3985/3985), done.
ubuntu@ubuntu:~/dev_ws$ . /opt/ros/foxy/setup.sh 
ubuntu@ubuntu:~/dev_ws$ 
ubuntu@ubuntu:~/dev_ws$ colcon build
Starting >>> examples_rclcpp_minimal_action_client
Starting >>> examples_rclcpp_minimal_action_server
Starting >>> examples_rclcpp_minimal_client
Starting >>> examples_rclcpp_minimal_composition
--- stderr: examples_rclcpp_minimal_action_client                                                                                               
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findexample_interfaces.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "example_interfaces", but CMake did not find one.

  Could not find a package configuration file provided by
  "example_interfaces" with any of the following names:

    example_interfacesConfig.cmake
    example_interfaces-config.cmake

  Add the installation prefix of "example_interfaces" to CMAKE_PREFIX_PATH or
  set "example_interfaces_DIR" to a directory containing one of the above
  files.  If "example_interfaces" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< examples_rclcpp_minimal_action_client [1.72s, exited with code 1]
Aborted  <<< examples_rclcpp_minimal_client [1.67s]
Aborted  <<< examples_rclcpp_minimal_action_server [1.69s]
Aborted  <<< examples_rclcpp_minimal_composition [35.9s]                                   

Summary: 0 packages finished [36.7s]
  1 package failed: examples_rclcpp_minimal_action_client
  3 packages aborted: examples_rclcpp_minimal_action_server examples_rclcpp_minimal_client examples_rclcpp_minimal_composition
  3 packages had stderr output: examples_rclcpp_minimal_action_client examples_rclcpp_minimal_action_server examples_rclcpp_minimal_client
  12 packages not processed

```

clalancette, 2021-07-13:
I'm guessing you don't have `ros-foxy-example-interfaces` installed; that should fix your issue, I think.

In the future, please open these kinds of questions over at https://answers.ros.org.  That way we have all of the questions and answers in a centralized place.

closed 2021-07-13 by clalancette

HorsonLiu, 2021-07-13:
it works! thank you.