ros2/examples #325: failure in the `colcon test`

bhomaidan1990, 2021-10-06:
Hi, following [ros2-foxy tutorial](https://docs.ros.org/en/foxy/Tutorials/Colcon-Tutorial.html#create-a-workspace) I have a failure in the `colcon test`.

## Bug report

**Required Info:**

- Operating System:
  Ubuntu 20.04
- Installation type:
  from source
- Version or commit hash:
  3eb5abc
- DDS implementation:
  Fast-RTPS
- Client library (if applicable):
 `examples_rclpy_minimal_client`

#### Steps to reproduce issue

```
Summary: 16 packages finished [4.51s]
  1 package had test failures: examples_rclpy_minimal_client
```

#### Additional information

`cat log/latest_test/examples_rclpy_minimal_client/stdout_stderr.log`

**Output:**

```
============================= test session starts ==============================
platform linux -- Python 3.8.10, pytest-6.2.5, py-1.10.0, pluggy-0.13.0
cachedir: /home/user/ros2_ws_all/ros2_example_ws/build/examples_rclpy_minimal_client/.pytest_cache
rootdir: /home/user/ros2_ws_all/ros2_example_ws/src/examples/rclpy, configfile: pytest.ini
plugins: ament-copyright-0.9.6, ament-mypy-0.9.6, ament-flake8-0.9.6, ament-xmllint-0.9.6, ament-lint-0.9.6, ament-pep257-0.9.6, launch-testing-ros-0.11.3, launch-testing-0.10.6, colcon-core-0.6.1, repeat-0.9.1, rerunfailures-10.2, flake8-1.0.7, mock-1.10.4, cov-2.8.1
collecting ... 
collected 3 items                                                              

test/test_copyright.py .                                                 [ 33%]
test/test_flake8.py F                                                    [ 66%]
test/test_pep257.py .                                                    [100%]

=================================== FAILURES ===================================
_________________________________ test_flake8 __________________________________
test/test_flake8.py:23: in test_flake8
    assert rc == 0, \
E   AssertionError: Found 4 code style errors / warnings:
E     ./examples_rclpy_minimal_client/client.py:36:1: B902 blind except Exception: statement
E     ./examples_rclpy_minimal_client/client_async_callback.py:42:1: B902 blind except Exception: statement
E     ./examples_rclpy_minimal_client/client_async.py:39:1: B902 blind except Exception: statement
E     ./examples_rclpy_minimal_client/client_async_member_function.py:47:1: B902 blind except Exception: statement
E   assert 1 == 0
----------------------------- Captured stdout call -----------------------------

./examples_rclpy_minimal_client/client.py:36:1: B902 blind except Exception: statement
    except Exception as e:
^

./examples_rclpy_minimal_client/client_async_callback.py:42:1: B902 blind except Exception: statement
            except Exception as e:
^

./examples_rclpy_minimal_client/client_async.py:39:1: B902 blind except Exception: statement
            except Exception as e:
^

./examples_rclpy_minimal_client/client_async_member_function.py:47:1: B902 blind except Exception: statement
            except Exception as e:
^

4     B902 blind except Exception: statement

9 files checked
4 errors

'B'-type errors: 4

Checked files:

* ./setup.py
* ./examples_rclpy_minimal_client/__init__.py
* ./examples_rclpy_minimal_client/client.py
* ./examples_rclpy_minimal_client/client_async_callback.py
* ./examples_rclpy_minimal_client/client_async.py
* ./examples_rclpy_minimal_client/client_async_member_function.py
* ./test/test_flake8.py
* ./test/test_copyright.py
* ./test/test_pep257.py
- generated xml file: /home/belal/ros2_ws_all/ros2_example_ws/build/examples_rclpy_minimal_client/pytest.xml -
=========================== short test summary info ============================
FAILED test/test_flake8.py::test_flake8 - AssertionError: Found 4 code style ...
========================= 1 failed, 2 passed in 0.29s ==========================

```

aprotyas, 2021-10-06:
I can't reproduce the test results. What version of Python and flake8 is installed in your computer? For reference, I have 3.8.10 and 3.7.9 respectively.

bhomaidan1990, 2021-10-06:
@aprotyas
- Python `3.8.10`
- flake8 `3.9.2`

clalancette, 2021-10-06:
More importantly, what's your version of `flake8-blind-except`?

In all likelihood, you are using a version newer than 0.1.1, which we know causes problems.  Because of that, in our CI we specifically pin to an earlier version: https://github.com/ros2/ci/blob/eb8d0ad3e9b74a71561c178bb5d8487f77206d35/ros2_batch_job/__main__.py#L55

We would love to move to a newer version of that plugin, but the honest truth is that there are some places where we want to use a blind Exception.  One thing we could consider doing is to mark those lines with `NOLINT` or whatever the flake8 equivalent is, and then unpin `flake8-blind-except`.  @bhomaidan1990 If you are willing to do some of that, we'd be happy to review it.