ros2/examples #344: examples_rclcpp_minimal_subscriber not building

arielc-brillianetor, 2022-05-30:
Hi,
Following the ROS2 humble tutorial for windows, all of the packages successfully build (if you build them one by one) except for the **examples_rclcpp_minimal_subscriber**. I get an error that the lib file is not found. I opened the folder and and .exe exists but no .lib file. 
Is this a problem with the CMakeList.txt file?
Thanks a lot for your help,
Ariel

C:\ros_ws\ros2_example_ws>colcon build --symlink-install --merge-install --event-handlers console_cohesion+  --packages-select examples_rclcpp_minimal_subscriber
Starting >>> examples_rclcpp_minimal_subscriber
--- output: examples_rclcpp_minimal_subscriber
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19044.
-- Found ament_cmake: 1.4.0 (C:/ros2_humble/share/ament_cmake/cmake)
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found rclcpp: 16.2.0 (C:/ros2_humble/share/rclcpp/cmake)
-- Found rosidl_generator_c: 3.2.0 (C:/ros2_humble/share/rosidl_generator_c/cmake)
-- Found rosidl_adapter: 3.2.0 (C:/ros2_humble/share/rosidl_adapter/cmake)
-- Found rosidl_generator_cpp: 3.2.0 (C:/ros2_humble/share/rosidl_generator_cpp/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
-- Found rmw_implementation_cmake: 6.3.0 (C:/ros2_humble/share/rmw_implementation_cmake/cmake)
-- Found rmw_fastrtps_cpp: 6.3.0 (C:/ros2_humble/share/rmw_fastrtps_cpp/cmake)
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- Found rclcpp_components: 16.0.1 (C:/ros_ws/dev_ws/install/share/rclcpp_components/cmake)
-- Found std_msgs: 4.5.0 (C:/ros2_humble/share/std_msgs/cmake)
-- Found ament_lint_auto: 0.13.1 (C:/ros2_humble/share/ament_lint_auto/cmake)
-- Added test 'copyright' to check source files copyright and LICENSE
-- Added test 'cppcheck' to perform static code analysis on C / C++ code
-- Configured cppcheck include dirs:
-- Configured cppcheck exclude dirs and/or files:
-- Added test 'cpplint' to check C / C++ code against the Google style
-- Configured cpplint exclude dirs and/or files:
-- Added test 'lint_cmake' to check CMake code style
-- Added test 'uncrustify' to check C / C++ code style
-- Configured uncrustify additional arguments:
-- Added test 'xmllint' to check XML markup files
-- Configuring done
-- Generating done
-- Build files have been written to: C:/ros_ws/ros2_example_ws/build/examples_rclcpp_minimal_subscriber
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Checking Build System
  Building Custom Rule C:/ros_ws/ros2_example_ws/src/examples/rclcpp/topics/minimal_subscriber/CMakeLists.txt
  static_wait_set_subscriber.vcxproj -> C:\ros_ws\ros2_example_ws\build\examples_rclcpp_minimal_subscriber\Release\static_wait_set_subscriber.exe
  Building Custom Rule C:/ros_ws/ros2_example_ws/src/examples/rclcpp/topics/minimal_subscriber/CMakeLists.txt
  subscriber_content_filtering.vcxproj -> C:\ros_ws\ros2_example_ws\build\examples_rclcpp_minimal_subscriber\Release\subscriber_content_filtering.exe
  Building Custom Rule C:/ros_ws/ros2_example_ws/src/examples/rclcpp/topics/minimal_subscriber/CMakeLists.txt
  subscriber_lambda.vcxproj -> C:\ros_ws\ros2_example_ws\build\examples_rclcpp_minimal_subscriber\Release\subscriber_lambda.exe
  Building Custom Rule C:/ros_ws/ros2_example_ws/src/examples/rclcpp/topics/minimal_subscriber/CMakeLists.txt
  subscriber_member_function.vcxproj -> C:\ros_ws\ros2_example_ws\build\examples_rclcpp_minimal_subscriber\Release\subscriber_member_function.exe
  Building Custom Rule C:/ros_ws/ros2_example_ws/src/examples/rclcpp/topics/minimal_subscriber/CMakeLists.txt
  subscriber_member_function_with_topic_statistics.vcxproj -> C:\ros_ws\ros2_example_ws\build\examples_rclcpp_minimal_subscriber\Release\subscriber_member_function_with_topic_statistics.exe
  Building Custom Rule C:/ros_ws/ros2_example_ws/src/examples/rclcpp/topics/minimal_subscriber/CMakeLists.txt
  subscriber_member_function_with_type_adapter.vcxproj -> C:\ros_ws\ros2_example_ws\build\examples_rclcpp_minimal_subscriber\Release\subscriber_member_function_with_type_adapter.exe
  Building Custom Rule C:/ros_ws/ros2_example_ws/src/examples/rclcpp/topics/minimal_subscriber/CMakeLists.txt
  subscriber_member_function_with_unique_network_flow_endpoints.vcxproj -> C:\ros_ws\ros2_example_ws\build\examples_rclcpp_minimal_subscriber\Release\subscriber_member_function_with_unique_network_flow_endpoints.exe
  Building Custom Rule C:/ros_ws/ros2_example_ws/src/examples/rclcpp/topics/minimal_subscriber/CMakeLists.txt
  subscriber_not_composable.vcxproj -> C:\ros_ws\ros2_example_ws\build\examples_rclcpp_minimal_subscriber\Release\subscriber_not_composable.exe
  Building Custom Rule C:/ros_ws/ros2_example_ws/src/examples/rclcpp/topics/minimal_subscriber/CMakeLists.txt
  time_triggered_wait_set_subscriber.vcxproj -> C:\ros_ws\ros2_example_ws\build\examples_rclcpp_minimal_subscriber\Release\time_triggered_wait_set_subscriber.exe
  Building Custom Rule C:/ros_ws/ros2_example_ws/src/examples/rclcpp/topics/minimal_subscriber/CMakeLists.txt
  wait_set_subscriber.vcxproj -> C:\ros_ws\ros2_example_ws\build\examples_rclcpp_minimal_subscriber\Release\wait_set_subscriber.exe
  Building Custom Rule C:/ros_ws/ros2_example_ws/src/examples/rclcpp/topics/minimal_subscriber/CMakeLists.txt
  wait_set_subscriber_library.vcxproj -> C:\ros_ws\ros2_example_ws\build\examples_rclcpp_minimal_subscriber\Release\wait_set_subscriber_library.dll
  Building Custom Rule C:/ros_ws/ros2_example_ws/src/examples/rclcpp/topics/minimal_subscriber/CMakeLists.txt
-- Execute custom install script
-- Symlinking: C:/ros_ws/ros2_example_ws/install/lib/examples_rclcpp_minimal_subscriber/wait_set_subscriber.exe
-- Symlinking: C:/ros_ws/ros2_example_ws/install/lib/examples_rclcpp_minimal_subscriber/static_wait_set_subscriber.exe
-- Symlinking: C:/ros_ws/ros2_example_ws/install/lib/examples_rclcpp_minimal_subscriber/time_triggered_wait_set_subscriber.exe
-- Symlinking: C:/ros_ws/ros2_example_ws/install/bin/wait_set_subscriber_library.dll
CMake Error at ament_cmake_symlink_install/ament_cmake_symlink_install.cmake:267 (message):
  ament_cmake_symlink_install_targets() can't find
  'C:/ros_ws/ros2_example_ws/build/examples_rclcpp_minimal_subscriber/Release/wait_set_subscriber_library.lib'
Call Stack (most recent call first):
  ament_cmake_symlink_install_targets_3_Release.cmake:1 (ament_cmake_symlink_install_targets)
  ament_cmake_symlink_install/ament_cmake_symlink_install.cmake:323 (include)
  cmake_install.cmake:36 (include)

---
--- stderr: examples_rclcpp_minimal_subscriber
CMake Error at ament_cmake_symlink_install/ament_cmake_symlink_install.cmake:267 (message):
  ament_cmake_symlink_install_targets() can't find
  'C:/ros_ws/ros2_example_ws/build/examples_rclcpp_minimal_subscriber/Release/wait_set_subscriber_library.lib'
Call Stack (most recent call first):
  ament_cmake_symlink_install_targets_3_Release.cmake:1 (ament_cmake_symlink_install_targets)
  ament_cmake_symlink_install/ament_cmake_symlink_install.cmake:323 (include)
  cmake_install.cmake:36 (include)

---
Failed   <<< examples_rclcpp_minimal_subscriber [5.50s, exited with code 1]

Summary: 0 packages finished [5.80s]
  1 package failed: examples_rclcpp_minimal_subscriber
  1 package had stderr output: examples_rclcpp_minimal_subscriber

akshatsingh16, 2022-08-25:
Hey have you find the solution ? cause i am also stuck on this same problem

akshatsingh16, 2022-08-25:
HI,
if you havent found the solution follow this https://www.reddit.com/r/ROS/comments/wxkfes/colcon_build_failed_in_example_failed_examples/

songyuc, 2023-08-14:
Hello,
I am inquiring if you would consider solving the issue on Windows.
Thank you for your time and attention.