ros2/examples #372: Update python nodes SIGINT handling

thomasmoore-torc, 2024-02-29:
## Bug report

`rclpy` was updated to automatically handle SIGINT as part of this PR which was incorporated into the Humble release: https://github.com/ros2/rclpy/pull/844

The python nodes in the demos repository were subsequently updated via the following PRs, but those changes were never incorporated into the examples repository.

* https://github.com/ros2/demos/pull/539
* https://github.com/ros2/demos/pull/555
* https://github.com/ros2/demos/pull/581

For consistency, and to ensure that an exception is not thrown when SIGINT is passed to the example python nodes, these same changes should be incorporated into this repository.

fujitatomoya, 2024-03-01:
@thomasmoore-torc thanks for creating issue.

> https://github.com/ros2/demos/pull/539

i think https://github.com/ros2/examples/pull/330 aligns with?

> https://github.com/ros2/demos/pull/555
> https://github.com/ros2/demos/pull/581

those seem like missing?

thomasmoore-torc, 2024-03-01:
@fujitatomoya - it looks like #330 only updated the [rclpy/executors](https://github.com/ros2/examples/tree/rolling/rclpy/executors) examples but not any of the other `rclpy` examples (e.g. [rclpy/topics/minimal_publisher/examples_rclpy_minimal_publisher/publisher_local_function.py](https://github.com/ros2/examples/blob/rolling/rclpy/topics/minimal_publisher/examples_rclpy_minimal_publisher/publisher_local_function.py)).

PR #379 (2024-05-09): Update the shutdown handling in all of the Python examples.
In particular, we should deal with KeyboardInterrupt (by just doing 'pass'), as well as ExternalShutdownException (which can come from the executor).  In all cases, we should call 'rclpy.try_shutdown()' to cleanup at the end.

Note well that I removed some of the *destroy() calls in the cleanup.  That's because they weren't correct in all cases, and to fix them up properly would really require us to have a nested set of try..except statements.  Given that these are examples, having that complex set of exception handling didn't seem li [... cut]
closed 2024-05-09 by clalancette