ros2/examples #378: `examples_rclpy_minimal_service` generates stack trace when interrupted by user

fujitatomoya, 2024-05-03:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 24.04
- Installation type:
  - source build
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/4d07f58591920794676261a14c2ba9d0a7bf2746
- DDS implementation:
  - Any
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

```bash
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_service service_member_function
[INFO] [1714698951.435342024] [minimal_service]: Incoming request
a: 41 b: 1
[INFO] [1714698955.953502812] [minimal_service]: Incoming request
a: 41 b: 1
[INFO] [1714698959.289201885] [minimal_service]: Incoming request
a: 41 b: 1
[INFO] [1714698963.647180164] [minimal_service]: Incoming request
a: 41 b: 1
^CTraceback (most recent call last):
  File "/opt/ros/jazzy/lib/examples_rclpy_minimal_service/service_member_function", line 33, in <module>
    sys.exit(load_entry_point('examples-rclpy-minimal-service==0.19.3', 'console_scripts', 'service_member_function')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/examples_rclpy_minimal_service/service_member_function.py", line 39, in main
    rclpy.spin(minimal_service)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/__init__.py", line 240, in spin
    executor.spin_once()
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 827, in spin_once
    self._spin_once_impl(timeout_sec)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 811, in _spin_once_impl
    handler, entity, node = self.wait_for_ready_callbacks(
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 781, in wait_for_ready_callbacks
    return next(self._cb_iter)
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 685, in _wait_for_ready_callbacks
    wait_set.wait(timeout_nsec)
KeyboardInterrupt
[ros2run]: Interrupt
```

#### Expected behavior

Stack trace should not be printed.

#### Actual behavior

see above.

#### Additional information

this issue is found by https://github.com/osrf/ros2_test_cases/issues/1260

clalancette, 2024-05-03:
This is actually a duplicate of #372 , so closing this out.

closed 2024-05-03 by clalancette

fujitatomoya, 2024-05-03:
@clalancette thanks for checking this.