ros2/examples #390: Single threaded executor is used for callbacks of reentrant callback group

sayali-purdue, 2024-07-25:
## Bug report

**Required Info:**

- Operating System:
  - N/A
- Installation type:
  - Static analysis of source code
- Version or commit hash:
  - Latest
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
Static review/inspection of source code

#### Expected behavior

#### Actual behavior

#### Additional information

In file [client_async_callback.py](rclpy/services/minimal_client/examples_rclpy_minimal_client/client_async_callback.py), single-threaded executor is used to process callbacks of reentrant callback group. As discussed in ROS documentation, callbacks from reentrant callback group can execute in parallel. However due to single thread, the callbacks of this group can not execute in parallel. Note that the timer callback also belongs to this reentrant callback group. Using a multi-threaded executor would be more efficient for this use case.

I would appreciate to hear your thoughts on this.

fujitatomoya, 2024-07-26:
> In file [client_async_callback.py](https://github.com/ros2/examples/issues/rclpy/services/minimal_client/examples_rclpy_minimal_client/client_async_callback.py),

this is 404, i think you meant https://github.com/ros2/examples/blob/rolling/rclpy/services/minimal_client/examples_rclpy_minimal_client/client_async_callback.py

i believe this example intentionally uses `SingleThreadedExecutor` to show how to call the service async in the callback of timer, using `async & await`.

sloretz, 2024-08-08:
> However due to single thread, the callbacks of this group can not execute in parallel. Note that the timer callback also belongs to this reentrant callback group. Using a multi-threaded executor would be more efficient for this use case.

The single threaded executor is intentional; this example shows how to use python coroutines. `call_service` is a coroutine that `awaits` the result of the service call. The single threaded executor is able to execute other callbacks while `call_service` `await`s the result.

If `call_service` was a normal function, then a multithreaded executor would be required. A normal function would block one thread while another thread processed the service server's response.

closed 2024-08-08 by sloretz