ros2/examples #395: error: no matching function for call to ‘rclcpp::TimerBase::execute_callback(bool&)’

CLOWN2255, 2024-10-09:
[image]
I tried to change one_off_timer ->executioner_callback (data); After deleting the data in, the compilation passed.
It's strange that everyone has encountered this problem before, why hasn't anyone reported it?

christophebedard, 2024-10-09:
`rclcpp::TimerBase::execute_callback()` indeed has no parameter in Humble: https://github.com/ros2/rclcpp/blob/28de27e4ff2bdec3b268001302bc9a631f56d5b0/rclcpp/include/rclcpp/timer.hpp#L108.

However, since Rolling (or potentially an earlier distro), it takes a `const std::shared_ptr<void> & data`: https://github.com/ros2/rclcpp/blob/41d3a27437973e85cf8fbb5612beb82fb6811e98/rclcpp/include/rclcpp/timer.hpp#L124.

Were you copying code meant to be run with a more recent distro and building it against Humble?

christophebedard, 2024-10-09:
Ah, yes, I'm assuming the line in your screenshot is: https://github.com/ros2/examples/blob/8aeda25df28f723298c3db0c329df965484f44cb/rclcpp/wait_set/src/wait_set_topics_and_timer.cpp#L76. Just make sure you're using the `humble` branch of this repo.

clalancette, 2024-10-09:
Please open questions like this on https://robotics.stackexchange.com/questions/tagged/ros, which is our central Question and Answer site.  You'll get a better answer there, and it will be searchable for the future.

Make sure to include a lot of information on what platform you are using, which ROS distribution you are using, and the exact steps you took.

closed 2024-10-09 by clalancette

CLOWN2255, 2024-10-09:
> Please open questions like this on https://robotics.stackexchange.com/questions/tagged/ros, which is our central Question and Answer site. You'll get a better answer there, and it will be searchable [...]

OK👌