ros2/examples #396: [FEA] add type negotiation(REP-2009) example.

ZhenshengLee, 2024-10-30:
## Feature request

#### Feature description

Type adaptation (REP-2007) and Type negotiation (REP-2009) has been introduced in ros2 humble version. There is type adaption example in topic example https://github.com/ros2/examples/blob/rolling/rclcpp/topics/minimal_publisher/member_function_with_type_adapter.cpp

But, there is no type negotiation example. So I suggest to add a good example to this repo.

#### Implementation considerations

NITROS is NVIDIA’s implementation of type adaption and negotiation. REP-2007 and REP-2009 are somehow for NVIDIA.

there is an sample from osrf for REP-2009, but I am not sure if it's best suited for official ros2 example, https://github.com/osrf/negotiated

clalancette, 2024-10-30:
I don't think that REP-2009 example is appropriate for this repository.  It isn't part of the ROS 2 core, and these are all core examples.  That said, I think it would be useful to have an example, so we could release that repository you released to if you would find that useful.

ZhenshengLee, 2024-10-30:
> It isn't part of the ROS 2 core

Thank you for your reply!

It makes sense if the REP-2009 feature implementation wasn't merged to rclcpp+rmw stack.

closing this.

closed 2024-10-30 by ZhenshengLee