ros2/examples #402: "colcon build" never finishes.

Shekyaga, 2024-12-30:
## Bug report

**Required Info:**

- Operating System:
  - Docker Container Ubuntu 22.04
- Version or commit hash:
  - osrf/ros:humble-desktop
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

After I install ros2 humble desktop using dep package and check talker-listener, I wanted to create a workspace as it described in tutorial https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Colcon-Tutorial.html.
```
mkdir -p ~/ros2_ws/src
cd ~/ros2_ws
git clone https://github.com/ros2/examples src/examples -b humble
colcon build --symlink-install
```

#### Expected behavior
It should install all the example package as shown in the tutorial.
#### Actual behavior
It cannot finish installing packages and the terminal freezes during building. The installation is also freeze the whole system so that I cannot halt or cancel process after 20 seconds build started.
#### Additional information
When I tried the same process for Ubuntu 22.04 Desktop, It ended up the same. Only can recover desktop as I reboot. When I checked the github repo, last time updated of rclcpp and  rclpy file was last week. Maybe it can result from the update somehow.

fujitatomoya, 2024-12-30:
> freeze the whole system so that I cannot halt or cancel process after 20 seconds build started.

`docker stop` does help? which sends the sigkill to the container process after grace period. (if i remember correctly 10 seconds)

> Only can recover desktop as I reboot.

you mean window manager is not responding at all? can you use the shell with other terminals and try to kill the build process id?

Shekyaga, 2025-01-02:
Neither `docker stop` is working. I tried to kill wsl on task manager and I could not. When I restart the system, it returns to normal. I finally solved it by downloading all packages one by one. I think it is basically because system tries to build all packages at once and it loads the CPU alot. Btw, sorry for delay in replying. It was due to Christmas.

fujitatomoya, 2025-01-02:
> I think it is basically because system tries to build all packages at once and it loads the CPU alot.

that is what we do pretty much every day. using `--parallel-workers` or `--executor sequential` could help your situation?

sounds like this is problem for kernel, user space should not be able to do this.

Shekyaga, 2025-01-03:
Thank you very much. I will use those options.

christophebedard, 2025-01-16:
Since a workaround was provided, I will close this issue.

I personally often use `--executor sequential`, especially when I'm building multiple workspaces at once and have a bunch of applications open. It helps avoid crashing/freezing my computer.

closed 2025-01-16 by christophebedard